espressif/esp-mdf #3: MESH ID

akshar001, 2018-05-30:
Finally you have done mesh, And been digging it till three days and funcionality is great..
There is one thing i need to ask is the mesh network will run without a router suppose there is no router avaible than mesh will work?
i realise there is special thing like mesh_id for it so the question is it's encrypted mesh_id and what if the same two same mesh_id network established around each other

akshar001, 2018-06-04:
Hello Sam, @Sam-espressif i can understand that there is no feature in "no router in mesh" but you have already built a very great support with mesh than why we should be reliable in router, i know espressif must have something differnet in mind but my question is if we already are connecting all of the nodes in one network than why shoudn't we allow them to talk to each other with or without router,
And that's really what makes mesh beautiful?

Sam-espressif, 2018-06-05:
Hi akshar001,
For current step, the router is required for the config of mesh network, but after the mesh network is established, two devices in the same mesh network, can communicate with each other as long as they have each other’s mac address. We have a demo which will be released soon, that a light intensity sensor to control the light automatically. I believe this is what you want.

For the next step, we will focus on the other scenarios, including totally without router, which means no router needed for the config of network, or the local control via smartphone. But this will take time, so if you want this feature, I’m sorry that you’ll have to wait for some more time.

akshar001, 2018-06-27:
@Sam-espressif ?

closed 2018-11-27 by akshar001