espressif/esp-mdf #5: esp-mesh 项目协助

Glen-SmarthomeG4, 2018-07-30:
尊敬的乐鑫技术大咖你好，感谢百忙之中处理我们的疑问
我们目前正在探索你们的esp-mdf软件包,可以肯定是它在建立ESP-MESH 时功能更加完善强大，毕竟它是在esp-idf之上进行的封装，但是通过初步的了解，感觉esp-mdf项目文件的的布局显得过于干瘪，
代码量比较大，而且作为一个新推行的技术，说明性的文件太少太不详细，将严重影响它的使用和普及，我相信这也不符合乐鑫的初衷。我在esp-mdf\components 目录底下找到了
大量的代码，却鲜有说明他们如何使用，使用某种功能时会涉及到哪些组件，而在esp-mdf/examples 之下的例子也没有详细的评测步骤！
然而相比之下esp-idf/examples/mesh 之下的实例却容易理解很多。然而你们的文件似乎指引着如果使用乐鑫的MESH 功能应该使用esp-mdf软件包
1> 我们的项目目的是使用乐鑫的MESH功能来实现智能家居控制系统的数据传输，我们有自己的服务器和交互协议
2> 我们的困惑是实现这一项目目的应该用esp-idf还是esp-mdf? 实现步骤是怎样的？应该研读那些文件和代码？这些文件的研读顺序是怎样的？（目前已经基于esp-idf 做过一些简单的测试）

请给予迅速支持，我们的项目正在进行之中,我们已经向我们的国外客户推介了你们的技术方案.

dongdong004, 2018-08-26:
```
 If you are building a commercial product with this and want support for that, please contact sales@espressif.com.

ESP-LBB, 2019-03-14:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

closed 2019-03-14 by ESP-LBB