espressif/esp-mdf #6: invalid static ip

germacovitch, 2018-08-27:
Hi
I am build a test application with ESP32 MDF in order to get familiar with the API. A configure a single node with a fixed WiFi network. The device can connect to wifi, but a always get the error:"invalid static ip"

Here goes my log:

I (15327) wifi: n:1 1, o:1 0, ap:1 1, sta:1 0, prof:1
I (16317) wifi: state: init -> auth (b0)
I (16322) wifi: state: auth -> assoc (0)
I (16328) wifi: state: assoc -> run (10)
I (16364) wifi: connected with Skynet, channel 1
I (16365) wifi: pm start, type: 0

E (16366) event: invalid static ip <---------------------------
W (16367) wifi: [beacon]new interval:300ms
I (16370) mdf_wifi_mesh: [esp_mesh_event_cb, 70]:esp_mesh_event_cb event.id: 7
I (16377) mdf_wifi_mesh: [esp_mesh_event_cb, 74]:wifi connected
I (16386) mdf_wifi_mesh: [esp_mesh_event_cb, 70]:esp_mesh_event_cb event.id: 14
I (18242) mdf_wifi_mesh: [esp_mesh_event_cb, 70]:esp_mesh_event_cb event.id: 17
I (18242) mdf_wifi_mesh: [esp_mesh_event_cb, 131]:wifi get ip
I (18246) event: sta ip: 192.168.0.8, mask: 255.255.255.0, gw: 192.168.0.1

Is there a way to eliminate the error?
BR Gilberto

Sam-espressif, 2018-08-31:
1. The error log will not affect the node working, it is because the dhcp procedure was delay started by mesh.
2. In mesh, all nodes' dhcp was disabled duration initialization, because the IP was no need in mesh network, and when the root was connected to router, it will enable the dhcp client in station interface to get ip to communication with extern network.

ESP-LBB, 2019-03-14:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

Robokishan, 2019-04-16:
hello after esp_mesh_waive_root  the one esp32 which got the major vote tries to connect to wifi but got the error

E (592027) event: invalid static ip
W (592027) wifi: [beacon]interval:300ms
I (592029) mesh: <nvs>write layer:1
I (592029) [mwifi, 66]: Parent is connected
I (592034) [mwifi_examples, 347]: event_loop_cb, event: 7
I (592039) [mwifi_examples, 355]: Parent is connected on station interface
I (592047) [mwifi_examples, 347]: event_loop_cb, event: 11
I (592053) [mwifi_examples, 347]: event_loop_cb, event: 14

and then connect to existing network is there any thing you can do i mean how can i properly use 

esp_mesh_waive_root??

zhanzhaocheng, 2019-04-16:
This tip (`E (592027) event: invalid static ip` )  is normal, not a mistake, you don't need to care about it.

Robokishan, 2019-04-16:
but the problem is esp_mesh_waive_root is not working

Robokishan, 2019-04-16:
is there anyone to help me out here?
just want to switch root node is there any tutorial ?

ESP-LBB, 2019-04-18:
Hi Robokishan，

Sorry for the late reply, out team is testing this function, and will update as soon as we got the result.

Robokishan, 2019-04-18:
should i post some logs ?

ESP-LBB, 2019-04-18:
Closing this as we will discuss further in #48

closed 2019-04-18 by ESP-LBB

Robokishan, 2019-04-18:
ok

kishorsherolla1506, 2021-05-18:
iam new to mesh network , trying to send the data over mesh , on power on and reset  some time it is sending the data , after some time it is not send the data if again i restart  the board is working good

kishorsherolla1506, 2021-05-18:
waht is the that problem , can any one help  me out ... thanks

zhanzhaocheng, 2021-05-18:
@kishorsherolla1506 Can you provide the logs of the receiving and sending equipment so that we can further analyze