espressif/esp-mdf #10: wifi mesh

chenqy2018, 2018-11-18:
What is the difference between MDF and mesh of example in IDF?
Is the mesh same between 8266 and 32，can they communicate with each other in mesh？

ESP-LBB, 2018-11-20:
Hello, thanks for you attention on MDF first.

Mesh in IDF only focus on how to generate and maintain the mesh network, and the data transfer inside the network. MDF is based on mesh, but we add several modules for easy use and test, for example, net config, OTA, Debug module, and will add more examples later, so that users can develop their own product quickly.

The mesh of 8266 and 32 are not compatible, and we don't support mesh on ESP8266 as its restricted RAM. We will focus on ESP32 Mesh and create more scenarios.

ESP-LBB, 2019-03-19:
Closing as this question appears to be answered.

closed 2019-03-19 by ESP-LBB