espressif/esp-mdf #11: Mesh  with arduino

andrelmbraga, 2018-11-28:
Hello Guys,

I'm new with esp-mdf and I would like to know if is possible to use it on arduino IDE.

Thanks,

me-no-dev, 2018-11-29:
MDF will be made available to Arduino as soon as the heavy active development is done. Currently MDF changes too much and too often to be possible to make it available to Arduino users. The basic mesh functionality on which MDF is build is already in Arduino.

vangalvin, 2019-08-06:
By any chance is ESP-IDF and ESP-MESH getting any closer for deployment in the Arduino IDE?

adonizer, 2019-08-17:
I am also waiting for the Arduino IDE implementation of the ESP-MDF or vice versa.

jwktje, 2019-10-21:
Any update on this?

jwktje, 2019-12-08:
Anything we can do to help further this along?

asetyde, 2019-12-12:
It can be useful who as our pass to mesh on Arduino , because OTA is difficult with standard IDF-arduino that use only wifi to upgrade , mdd has a better system to broadcast Ota ?

me-no-dev, 2019-12-12:
it's just not that easy to add MDF to Arduino right now. That will become easier soon and it will happen. I do not know how much of MDF will end up in there though... Also some xDFs are linked to particular IDF version and if that version does not match, then there will be issues.

asetyde, 2019-12-12:
thanks for reply !