espressif/esp-mdf #16: ldgen fails due to bug in included IDF version

Dee-Fuse, 2019-01-10:
## Environment

- MDF version:                         7fe8393
- Development Env:		[Eclipse]
- Operating System:		[Windows]

## Problem Description

MDF get_startet project does not compile with current MDF / IDF provided.
This issue has been fixed with espressif/esp-idf@d3938c8a6659ff179968d26404512b771a01b8a3
See:  espressif/esp-idf#2812

Is it possible to update to this IDF commit without breaking MDF?
Replacing only the ``/esp-mdf/esp-idf/make/ldgen.mk`` file resolves this issue for now.

### Expected Behavior
Eclipse can Build example program

### Actual Behavior
Generating esp32.common.ld fails due to wrong paths.

### Steps to repropduce

1. setup Toolchain 
2. clone latest MDF
3. setup Eclipse Project 
4. hit "Build"

## Debug Logs

```
Generating esp32.common.ld
python /home/admin/esp/esp-mdf/esp-idf/tools/ldgen/ldgen.py --input          /home/admin/esp/esp-mdf/esp-idf/components/esp32/ld/esp32.common.ld.in --config        /c/Users/admin/Documents/ConstellationLightDisplays/mdfMesh/sdkconfig --fragments      /home/admin/esp/esp-mdf/esp-idf/components/app_trace/linker.lf           /home/admin/esp/esp-mdf/esp-idf/components/esp32/ld/esp32_fragments.lf /home/admin/esp/esp-mdf/esp-idf/components/esp32/linker.lf       /home/admin/esp/esp-mdf/esp-idf/components/esp_ringbuf/linker.lf     /home/admin/esp/esp-mdf/esp-idf/components/freertos/linker.lf /home/admin/esp/esp-mdf/esp-idf/components/heap/linker.lf                            /home/admin/esp/esp-mdf/esp-idf/components/soc/linker.lf /home/admin/esp/esp-mdf/esp-idf/components/spi_flash/linker.lf          /home/admin/esp/esp-mdf/esp-idf/components/xtensa-debug-module/linker.lf --output         /c/Users/admin/Documents/ConstellationLightDisplays/mdfMesh/build/esp32/esp32.common.ld --sections      /c/Users/admin/Documents/ConstellationLightDisplays/mdfMesh/build/ldgen.section_infos --kconfig       /home/admin/esp/esp-mdf/esp-idf/Kconfig --env           "COMPONENT_KCONFIGS=/home/admin/esp/esp-mdf/esp-idf/components/app_trace/Kconfig   /home/admin/esp/esp-mdf/esp-idf/components/aws_iot/Kconfig   /home/admin/esp/esp-mdf/esp-idf/components/bt/Kconfig    /home/admin/esp/esp-mdf/esp-idf/components/driver/Kconfig  /home/admin/esp/esp-mdf/esp-idf/components/esp32/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/esp_adc_cal/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/esp_event/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/esp_http_client/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/esp_http_server/Kconfig     /home/admin/esp/esp-mdf/esp-idf/components/ethernet/Kconfig  /home/admin/esp/esp-mdf/esp-idf/components/fatfs/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/freemodbus/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/freertos/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/heap/Kconfig    /home/admin/esp/esp-mdf/esp-idf/components/libsodium/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/log/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/lwip/Kconfig  /home/admin/esp/esp-mdf/esp-idf/components/mbedtls/Kconfig C:\\msys32\\home\\admin\\esp\\esp-mdf/components/mcommon/Kconfig C:\\msys32\\home\\admin\\esp\\esp-mdf/components/mconfig/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/mdns/Kconfig C:\\msys32\\home\\admin\\esp\\esp-mdf/components/mespnow/Kconfig    /home/admin/esp/esp-mdf/esp-idf/components/mqtt/Kconfig C:\\msys32\\home\\admin\\esp\\esp-mdf/components/mupgrade/Kconfig C:\\msys32\\home\\admin\\esp\\esp-mdf/components/mwifi/Kconfig   /home/admin/esp/esp-mdf/esp-idf/components/nvs_flash/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/openssl/Kconfig    /home/admin/esp/esp-mdf/esp-idf/components/pthread/Kconfig    /home/admin/esp/esp-mdf/esp-idf/components/spi_flash/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/spiffs/Kconfig  /home/admin/esp/esp-mdf/esp-idf/components/tcpip_adapter/Kconfig  /home/admin/esp/esp-mdf/esp-idf/components/unity/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/vfs/Kconfig /home/admin/esp/esp-mdf/esp-idf/components/wear_levelling/Kconfig   " --env           "COMPONENT_KCONFIGS_PROJBUILD= /home/admin/esp/esp-mdf/esp-idf/components/app_update/Kconfig.projbuild   /home/admin/esp/esp-mdf/esp-idf/components/bootloader/Kconfig.projbuild                /home/admin/esp/esp-mdf/esp-idf/components/esptool_py/Kconfig.projbuild             /c/Users/admin/Documents/ConstellationLightDisplays/mdfMesh/main/Kconfig.projbuild                /home/admin/esp/esp-mdf/esp-idf/components/partition_table/Kconfig.projbuild                 " --env           "IDF_CMAKE=n"
Traceback (most recent call last):
  File "C:/msys32/home/admin/esp/esp-mdf/esp-idf/tools/ldgen/ldgen.py", line 114, in <module>
    main()
  File "C:/msys32/home/admin/esp/esp-mdf/esp-idf/tools/ldgen/ldgen.py", line 98, in main
    sdkconfig = SDKConfig(kconfig_file, config_file, args.env)
  File "C:/msys32/home/admin/esp/esp-mdf/esp-idf/tools/ldgen/sdkconfig.py", line 54, in __init__
    self.config = kconfiglib.Kconfig(kconfig_file.name)
  File "C:\msys32\home\admin\esp\esp-mdf\esp-idf\tools\kconfig_new/kconfiglib.py", line 653, in __init__
    self.top_node)  # prev_node
  File "C:\msys32\home\admin\esp\esp-mdf\esp-idf\tools\kconfig_new/kconfiglib.py", line 1546, in _parse_block
    self._enter_file(sourced_file)
  File "C:\msys32\home\admin\esp\esp-mdf\esp-idf\tools\kconfig_new/kconfiglib.py", line 1130, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: C:/msys32/home/admin/esp/esp-mdf/esp-idf/Kconfig:59: Could not open '/home/admin/esp/esp-mdf/esp-idf/components/app_update/Kconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
make: *** [/home/admin/esp/esp-mdf/esp-idf/components/esp32/Makefile.projbuild:51: /c/Users/admin/Documents/ConstellationLightDisplays/mdfMesh/build/esp32/esp32.common.ld] Error 1

```

zhanzhaochen, 2019-01-10:
Thank you very much for your feedback, I will fix it as soon as possible with esp-idf.
Since updating esp-idf I need to verify that other features of esp-mdf are working properly, the fix for this issue will be completed next week.

zhanzhaochen, 2019-01-23:
I have already updated esp-idf

Dee-Fuse, 2019-01-23:
Is now fixed! Thank you!

closed 2019-01-23 by Dee-Fuse