espressif/esp-mdf #18: Possible typo for mutex handle in `mwifi.c`

alafi, 2019-01-22:
In the following line (I might be mistaken but) I think it is a typo since `SemaphoreHandle_t` is already a pointer and there is no need to define a pointer to pointer here.
https://github.com/espressif/esp-mdf/blob/a1e731196dccb9bda50011cd2d73b29786b8c178/components/mwifi/mwifi.c#L424

I don't believe this is affecting the execution in any way but I am just highlighting it to be considered for fixing (if it is really an issue).

zhanzhaochen, 2019-01-23:
Thank you very much for your review of the code, this is caused by my mistake, I have fixed it, I will merge it on the master branch as soon as possible.

zhanzhaochen, 2019-01-25:
I have updated the changes to the master branch, please help verify it.

alafi, 2019-01-25:
Correction confirmed. Thanks.
Closing this one.

closed 2019-01-25 by alafi