espressif/esp-mdf #19: Part of the packet is lost when specifying wait_ticks other than portMAX_DELAY for mwifi_root_read()

alafi, 2019-01-22:
First of all I have tested and traced this issue for `mwifi_root_read()` but I believe it goes to `mwifi_read()` function as well.

To reproduce the issue, let the node send a large buffer that will span over multiple payloads (e.g. 4000 bytes) to the root, and let the root read the buffer with timeout other than portMAX_DELAY (e.g. 20 seconds).

The root will start showing the warning message: 
`Part of the packet is lost, expect_seq: 0, recv_seq: 2`

To be honest, I couldn't wait to receive a fix and as per my efforts to trace the issue, it seems like it is related to the following code block:

https://github.com/espressif/esp-mdf/blob/a1e731196dccb9bda50011cd2d73b29786b8c178/components/mwifi/mwifi.c#L972-L975

The calculation of the remaining waiting time does not reset the variable `start_ticks` to current tick count (`xTaskGetTickCount()`) after updating `wait_ticks` at the beginning of each iteration.

zhanzhaochen, 2019-01-23:
`wait_ticks` refers to the total time of all packets received and the gap between the two packets, so the start_ticks will not be updated after receiving a packet of data.

You can increase the waiting time. If this does not meet your needs, I would like to describe your usage scenario in order to better solve it.

alafi, 2019-01-24:
The scenario is pretty explained in the issue description. You may reproduce it with mwifi example:
1. Let the node send packets with size of 4000 bytes every 15 seconds.
2. Let the root read from node with timeout of 20 seconds.

That's it.

To explain it with the implementation in mwifi.c, consider the following flow:

- The root will wait for 15 seconds before receiving the first payload.
- On the second iteration of the loop, `wait_ticks` will be 5 seconds (15 - 0 = 15 ==> `wait_ticks` = 20 - 15 ==> `wait_ticks` = 5) which is OK.
- On the third iteration the issue happens, when only few micro seconds had passed `wait_ticks` will be 0 (15.000034 - 0 = 15.000034 ==> 15.000034 > `wait_ticks` ==> `wait_ticks`=0) which is wrong and will cause the loop to exit with timeout.

The solution is to update the value of `start_ticks` at the end or beginning of the loop.

zhanzhaochen, 2019-01-25:
As you said, due to the `wait_ticks` error, I have modified it as follows and have updated it to the master branch. Please help verify it.

https://github.com/espressif/esp-mdf/blob/b264ce4e3798facb700986427bd46c751236d35f/components/mwifi/mwifi.c#L973-L980

alafi, 2019-01-25:
I have test it and the issue is fixed. Thank you.
I am closing this one.

closed 2019-01-25 by alafi