espressif/esp-mdf #20: mupgrade example doesn't work when http server uses Transfer-Encoding:chunked

raulMrello, 2019-01-29:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Core (if using chip or module): [ESP-WROOM32]
- MDF version: f028311
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

Trying to update firmware via ```mupgrade```, from an http server with Transfer-Encoding ```chunked``` results in this error:
```cpp
D (28670) HTTP_CLIENT: http_on_chunk_complete
D (28674) HTTP_CLIENT: content_length = -1
E (28685) [mupgrade_root, 43]: <MDF_ERR_INVALID_ARG> !(size > 0)
```

### Expected Behavior

It should download binary file from http server normally, handling ```chunked``` Transfer-Encoding.

### Actual Behavior

### Steps to repropduce

Fetching headers from http server with instruction:
```cpp
total_size = esp_http_client_fetch_headers(client);
```
results in: ```total_size = 0```

Then at ```mupgrade_root.c``` line 43 fails checking size value:
```cpp
mdf_err_t mupgrade_firmware_init(const char *name, size_t size)
{
    MDF_PARAM_CHECK(name);
    MDF_PARAM_CHECK(size > 0);       //<---- THIS CHECKING FAILS
```

### Code to reproduce this issue

You can reproduce this issue, hosting the binary file in a google.sites repository like this:
```cpp
https://sites.google.com/site/username/hello-world.bin
```

## Debug Logs

```
D (28571) HTTP_CLIENT: on_message_begin
D (28572) HTTP_CLIENT: HEADER=Content-Type:text/html; charset=UTF-8
D (28572) HTTP_CLIENT: HEADER=X-Frame-Options:SAMEORIGIN
D (28576) HTTP_CLIENT: HEADER=X-Robots-Tag:noarchive
D (28581) HTTP_CLIENT: HEADER=Cache-Control:no-cache, no-store, max-age=0, must-revalidate
D (28589) HTTP_CLIENT: HEADER=Pragma:no-cache
D (28594) HTTP_CLIENT: HEADER=Expires:Mon, 01 Jan 1990 00:00:00 GMT
D (28600) HTTP_CLIENT: HEADER=Date:Tue, 29 Jan 2019 09:31:34 GMT
D (28606) HTTP_CLIENT: HEADER=Last-Modified:Thu, 24 Jan 2019 12:46:13 GMT
D (28613) HTTP_CLIENT: HEADER=Location:https://sites.google.com/site/username/hello-world.bin
D (28623) HTTP_CLIENT: HEADER=X-Content-Type-Options:nosniff
D (28629) HTTP_CLIENT: HEADER=X-XSS-Protection:1; mode=block
D (28635) HTTP_CLIENT: HEADER=Server:GSE
D (28639) HTTP_CLIENT: HEADER=Alt-Svc:quic=
D (28643) HTTP_CLIENT: HEADER=Accept-Ranges:none
D (28648) HTTP_CLIENT: HEADER=Vary:Accept-Encoding
D (28653) HTTP_CLIENT: HEADER=Transfer-Encoding:chunked
D (28658) HTTP_CLIENT: http_on_headers_complete, status=302, offset=621, nread=621
D (28666) HTTP_CLIENT: http_on_body 259
D (28670) HTTP_CLIENT: http_on_chunk_complete
D (28674) HTTP_CLIENT: content_length = -1
E (28685) [mupgrade_root, 43]: <MDF_ERR_INVALID_ARG> !(size > 0)
```

zhanzhaocheng, 2019-02-17:
1. Since `https://sites.google.com/site/username/hello-world.bin` is not a download link, it needs to log in to get the firmware. I use esp32 to connect to it and get the following result:

```html
<HEAD>
<TITLE>Moved Temporarily</TITLE>
</HEAD>
<BODY BGCOLOR="#FFFFFF" TEXT="#000000">
<H1>Moved Temporarily</H1>
The document has moved <A HREF="https://sites.google.com/site/sites/system/errors/WebspaceNotFound?path=%2Fusername%2Fhello-world.bin">here</A>.
</BODY>
</HTML>
```

You can test it using the project on github, such as: https://raw.githubusercontent.com/zhanzhaocheng/test/master/hello-world.bin

2. I have added support for `Transfer-Encoding:chunked` in mupgrade example, https://github.com/espressif/esp-mdf/commit/188bbaff0938e90f33ca83c499569e27602e8108

ESP-LBB, 2019-03-29:
Closing as this question appears to be answered.

closed 2019-03-29 by ESP-LBB