espressif/esp-mdf #21: esp_http_client_cleanup in C++ compilation, raises an exception and leaves the opened socket hung

raulMrello, 2019-02-01:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Core (if using chip or module): [ESP-WROOM32]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): f028311
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

Building MUPGRADE example at ```example/function_demo/mupgrade``` using C++ compilation, has different behaviour than compiling using ANSI C.

In this case, functions ```esp_http_client_cleanup``` raises an exception and leaves the opened socket in an unknown state. 

Due to this issue, http server gets blocked.

### Expected Behavior

This function should close the opened socket and release all allocated resources properly.

### Actual Behavior

#### Steps to repropduce

1. Change ```mupgrade_example.c``` to be compiled as C++ code. So rename it as ```mupgrade_example.cpp``` 
2. Append ```extern "C"``` block on ```app_main```
```cpp
extern "C" {
void app_main(){
....
}
}
```
3. Build the example ```make all``` and generates ```mupgrade.bin```
4. Rename ```upgrade.bin``` as ```hello-world.bin``` and uploads to the http server. Doing that we will update the board with the same firmware, and then it will try to update itself continuosly.
4. Flash the board ```make flash monitor```
5. Once started, it connects to http server, downloads ```hello-world.bin```
6. But after executing ```esp_http_client_cleanup``` an exception is raised and opened socket with the server gets hung.

## Debug Logs

Next logs shows the http connection establishment and first packets download:
```
I (17611) mesh: [scan]new scanning time:600ms
W (17611) wifi: [beacon]new interval:300ms
I (17614) mesh: <nvs>write layer:1
I (17614) [mwifi, 65]: Parent is connected
I (17617) [mupgrade_example, 186]: MDF_EVENT_PARENT_CONNECTED
I (17625) [mupgrade_example, 38]: ROOT read task is running
I (17626) [mupgrade_example, 69]: NODE read task is running
I (18325) event: sta ip: 192.168.254.34, mask: 255.255.255.0, gw: 192.168.254.231
I (18326) [mupgrade_example, 198]: MDF_EVENT_MWIFI_ROOT_GOT_IP
I (18329) [mupgrade_example, 112]: Setting remote link: http://192.168.254.65:8000/hello-world.bin
I (18340) [mupgrade_example, 118]: Open HTTP connection: http://192.168.254.65:8000/hello-world.bin
I (18595) [mupgrade_root, 54]: Running partition, label: ota_1, type: 0x0, subtype: 0x11, address: 0x150000
I (18598) [mupgrade_root, 56]: Update partition, label: ota_0, type: 0x0, subtype: 0x10, address: 0x10000
I (19901) [mupgrade_example, 140]: Writing to flash 1456 bytes
I (19913) [mupgrade_example, 140]: Writing to flash 1456 bytes
..... continue downloading packets.....
```

Next logs shows the end of the downloading process. And later, the socket closing, where the exception is raised:

```cpp
I (31364) [mupgrade_example, 140]: Writing to flash 1456 bytes
I (31377) [mupgrade_example, 140]: Writing to flash 1456 bytes
I (31385) [mupgrade_example, 140]: Writing to flash 48 bytes
I (31387) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2b674 (177780) map
[... 179 lines cut ...]
D (20670) esp_core_dump: Dump task 3ffcc51c
D (20674) esp_core_dump: Dump task 3ffcd090
D (20678) esp_core_dump: Dump task 3ffce374
D (20681) esp_core_dump: Dump task 3ffbb55c
D (20686) esp_core_dump: Dump task 3ffbb6cc
D (20691) esp_core_dump: Dump task 3ffbdfd4
D (20694) esp_core_dump: Dump task 3ffc1d34
D (20698) esp_core_dump: Dump task 3ffafd0c
D (20701) esp_core_dump: Dump task 3ffc05fc
D (20705) esp_core_dump: Dump task 3ffc95e0
D (20709) esp_core_dump: Dump task 3ffba5b4
D (20712) esp_core_dump: Data from flash:
D (20713) esp_core_dump: 429c
D (20716) esp_core_dump: 1
D (20719) esp_core_dump: 11
D (20722) esp_core_dump: 16c
D (20724) esp_core_dump: Dump data CRC = 0xf6ad19e8
I (20729) esp_core_dump: Core dump has been saved to flash.
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee

```

orbisdeveloper, 2019-02-04:
Hi, updating ESP-IDF from ```esp-idf @ a62cbfe``` to ```esp-idf @ master``` the problem get solved:

### Actions to solve the problem using ```git bash```

```cpp
RaulM@RaulM MINGW64 ~/workspace/esp-mdf (master)
$ cd esp-idf/

RaulM@RaulM MINGW64 ~/workspace/esp-mdf/esp-idf ((a62cbfec9...))
$ git checkout master
Previous HEAD position was a62cbfec9... Merge branch 'bugfix/fix_psram_eid' into 'master'
Switched to branch 'master'
M       components/esp32/lib
M       components/esptool_py/esptool
Your branch is up-to-date with 'origin/master'.

RaulM@RaulM MINGW64 ~/workspace/esp-mdf/esp-idf (master)
$ git submodule update --recursive --init
Submodule path 'components/esp32/lib': checked out 'fa2100a8078ee779ffe7663ba3ccf3abf8b551ce'
Submodule path 'components/esptool_py/esptool': checked out '9ad444a6e06e58833d5e6044c1d5f3eb3dd56023'

RaulM@RaulM MINGW64 ~/workspace/esp-mdf/esp-idf (master)
$
```

Nevertheless, I still leave the issue opened for your evaluation.

raulMrello, 2019-02-04:
Hi, updating ESP-IDF from ```esp-idf @ a62cbfe``` to ```esp-idf @ master``` the problem get solved:

### Actions to solve the problem using ```git bash```

```cpp
RaulM@RaulM MINGW64 ~/workspace/esp-mdf (master)
$ cd esp-idf/

RaulM@RaulM MINGW64 ~/workspace/esp-mdf/esp-idf ((a62cbfec9...))
$ git checkout master
Previous HEAD position was a62cbfec9... Merge branch 'bugfix/fix_psram_eid' into 'master'
Switched to branch 'master'
M       components/esp32/lib
M       components/esptool_py/esptool
Your branch is up-to-date with 'origin/master'.

RaulM@RaulM MINGW64 ~/workspace/esp-mdf/esp-idf (master)
$ git submodule update --recursive --init
Submodule path 'components/esp32/lib': checked out 'fa2100a8078ee779ffe7663ba3ccf3abf8b551ce'
Submodule path 'components/esptool_py/esptool': checked out '9ad444a6e06e58833d5e6044c1d5f3eb3dd56023'

RaulM@RaulM MINGW64 ~/workspace/esp-mdf/esp-idf (master)
$
```

Nevertheless, I still leave the issue opened for your evaluation.

zhanzhaocheng, 2019-02-17:
Thank you very much, your positioning of the problem, I need to verify it further, if there is no problem, I will update esp-idf as soon as possible

closed 2019-06-04 by zhanzhaocheng