espressif/esp-mdf #22: WindowOverflow and ESP_CORE_DUMP Error

fjalonzo, 2019-02-03:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Core (if using chip or module): [ESP-WROOM32]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): f028311
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[VS Code]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

0x40080000: _WindowOverflow4 at C:/msys32/home/FrancisJeralyn/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1685

0x4008c4ac: coex_bt_high_prio at /home/qgu/git_tree/chip7.1_rtc/board_code/app_test/pp/rtc/bt_bb.c:402

E (254) esp_core_dump: Failed to mmap core dump data `(260)!`

[image]

[image]

This is the "**get_started**" example program.

igrr, 2019-02-04:
FWIW, `WindowOverflow` and similar lines are not an error: IDF monitor decodes addresses found in serial output, converting them to function names. See https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses.
With regards to the error coming from core dump — it is benign, no real issue, but should indeed be fixed in IDF.

ESP-LBB, 2019-03-29:
Closing as this question appears to be answered.

closed 2019-03-29 by ESP-LBB