espressif/esp-mdf #24: mwifi example: Server gets broken messages

DevAlone, 2019-02-05:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Core (if using chip or module): ESP-WROOM32
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 
f028311
- Development Env:		Make
- Operating System:		Arch Linux
- Power Supply:         USB

## Problem Description

I've just changed the period of delay in node_write_task to 500ms and sometimes my server which is running on laptop gets wrong messages. 
It gets this:

`{"addr":"84:0d:8e:0d:33:60","data":{"seq":9,"layer":2,"status":0}}}`

instead of this:

`{"addr":"84:0d:8e:0d:33:60","data":{"seq":9,"layer":2,"status":0}}`

extra closing bracket is added for some reason. Mesh network consists of 3 devices placed pretty close to each other(but the problem happened even on the network with 2 nodes placed at around 10 meters from each other).
Sometimes wrong message goes away and comes back when I move nodes.

### Expected Behavior

MDF does not produce wrong messages or drops them by checksum. 

### Actual Behavior

Wrong messages are happening.

### Steps to repropduce

1. change period of delay in node_write_task in mwifi example
2. upload program
3. test

DevAlone, 2019-02-09:
The problem was with overflowing the buffer. 

Steps to reproduce:
1 connect two devices with mwifi example to network
2 after 10 packets sent(seq is 10 on both devices) reboot any device

Created a pull request https://github.com/espressif/esp-mdf/pull/25

zhanzhaocheng, 2019-02-17:
This is due to the fact that the bufffer in `tcp_client_write_task` is not cleared. It has been fixed. See [L49](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mwifi/main/mwifi_example.c#L49)

DevAlone, 2019-02-17:
@zhanzhaocheng, you don't need memset, it's enough to set just one zero terminator as in my pull request.

zhanzhaocheng, 2019-02-18:
Setting a zero-termination  is really a good idea, but memset might be better understood and consistent with the [tcp_client_read_task](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mwifi/main/mwifi_example.c#L125) method.

It can also be solved as follows:
```c
asprintf(&json_data, "{\"addr\":\"" MACSTR "\",\"data\":%.*s}", MAC2STR(src_addr), size, data);
```

ESP-LBB, 2019-04-11:
Closing as this question appears to be answered.

closed 2019-04-11 by ESP-LBB