espressif/esp-mdf #28: Checking for errors in mwifi init

rikkertkoppes, 2019-02-25:
When I initialize mwifi with a wrong configuration, in my case a wrong wifi password, the system aborts with an error message and reboots. This results in a boot loop as the wifi password was previously stored in flash.

I believe the problem is here: <https://github.com/espressif/esp-mdf/blob/38b7f07ab761aff6b4fca977facceea3c1b23d25/components/mwifi/mwifi.c#L295> 

Since `esp_mesh_start()` is wrapped in an `ESP_ERROR_CHECK`, there seems to be nothing I can do to capture the error. I think it would be wise to return the error from `mwifi_start()` so that I can handle that error properly.

Is there something I can do to fix this?

zhanzhaocheng, 2019-02-26:
Please provide your log and *.elf files in the build directory.
For an analysis of logs when errors occur, see:https://docs.espressif.com/projects/esp-mdf/en/latest/api-guides/error-handling.html#fatal-errors

rikkertkoppes, 2019-02-26:
I fully understand why the error occurs (wrong password), which is also what the log says:

```
W (699) wifi: [beacon]interval:100ms
I (702) wifi: mode : sta (24:0a:c4:20:58:68) + softAP (24:0a:c4:20:58:69)
I (711) wifi: Init max length of beacon: 752/752
I (713) wifi: Init max length of beacon: 752/752
W (720) wifi: [beacon]interval:100ms
I (721) mesh: [CONFIG]invalid router password, len:7, password:1234235
ESP_ERROR_CHECK failed: esp_err_t 0x4008 (ESP_ERR_MESH_ARGUMENT) at 0x4008e1e8
0x4008e1e8: _esp_error_check_failed at /home/rikkert/esp/esp-mdf/esp-idf/components/esp32/panic.c:685

file: "/home/rikkert/esp/esp-mdf/components/mwifi/mwifi.c" line 293
func: mwifi_start
expression: esp_mesh_set_config(&mesh_config)

Backtrace: 0x4008dd40:0x3ffb3050 0x4008e1eb:0x3ffb3070 0x400d8988:0x3ffb3090 0x400f63b8:0x3ffb31a0 0x400f6457:0x3ffb32a0 0x400d4c19:0x3ffb3340 0x400d212e:0x3ffb3370 0x400903f5:0x3ffb3390
0x4008dd40: invoke_abort at /home/rikkert/esp/esp-mdf/esp-idf/components/esp32/panic.c:680

0x4008e1eb: _esp_error_check_failed at /home/rikkert/esp/esp-mdf/esp-idf/components/esp32/panic.c:686

0x400d8988: mwifi_start at /home/rikkert/esp/esp-mdf/components/mwifi/mwifi.c:391

0x400f63b8: mesh_init_with_ap at /home/rikkert/workspace/camarillooko/firmware/mdf-events/components/Tasks/meshTask.c:211

0x400f6457: setupMesh at /home/rikkert/workspace/camarillooko/firmware/mdf-events/components/Tasks/meshTask.c:229

0x400d4c19: app_main at /home/rikkert/workspace/camarillooko/firmware/mdf-events/main/main.cpp:295

0x400d212e: main_task at /home/rikkert/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:514

0x400903f5: vPortTaskWrapper at /home/rikkert/esp/esp-mdf/esp-idf/components/freertos/port.c:403
```

The problem is that an `ESP_ERROR_CHECK` (and hence an abort() on error) is hardcoded in `mwifi_start()` and I don't see a way to catch that error

zhanzhaocheng, 2019-02-26:
I understand what you mean, I will update it. The mwifi module will return more errors to the previous layer. 
I will notify you after the update.

ESP-LBB, 2019-04-11:
Closing as this question appears to be answered.

closed 2019-04-11 by ESP-LBB