espressif/esp-mdf #30: mdf报错是什么原因

chenguang3312, 2019-03-02:
我使用的是window7系统，按照说明书下载并配置了msys2环境，在git clone时，因为中国国内原因导致下载不了，所以直接下载github上的zip压缩包 解压运行。下载到的zip压缩包里 ``esp-idf文件夹是空的，所以我在github上也下载了3.1.2版本的esp-idf，通过配置后可以正常编译烧写esp-idf内的例子。但是到esp-mdf下编译get-start例子时，报错如下：
# make menuconfig
fatal: 不是一个 git 仓库（或者任何父目录）：.git
C:/msys32/home/Administrator/esp/esp-mdf/project.mk:9: C:/msys32/home/Administrator/esp/esp-mdf/esp-idf/make/project.mk: No such file or directory
make: *** 没有规则可制作目标“C:/msys32/home/Administrator/esp/esp-mdf/esp-idf/make/project.mk”。 停止。
`

zhanzhaocheng, 2019-03-04:
1. 原因:
这是由于你 esp-idf 未下载完全造成的. esp-idf 是 esp-mdf 的子工程您无法直接通过压缩包的方法进行下载

2. 解决:
* 您可以通过手机热点进行下载
* 我将 esp-mdf 打包放置在了[百度云盘](https://pan.baidu.com/s/1EzRU5jHvyjytJCNmYyGnIg) (链接: https://pan.baidu.com/s/1EzRU5jHvyjytJCNmYyGnIg 提取码: estp), 不过还是推荐您直接 `git clone` 的方式进行下载,以为后期的版本维护

chenguang3312, 2019-03-11:
好的 多谢

ESP-LBB, 2019-03-29:
Closing as this question appears to be answered.

closed 2019-03-29 by ESP-LBB