espressif/esp-mdf #31: 节点间通讯问题

dongjieqing, 2019-03-08:
利用mwifi包通讯，各节点与服务器通讯正常，为了实现节点间点对点通讯，将发送地址和接收地址改为同一个地址，例如，一个模块动MAC地址是{0x24,0x0a,0xc4,0xc1,0x1e,0x08}，另一个是{0x9c,0xa6,0x15,0x1d,0x3d,0x41}，两个模块分别烧录不同MAC地址，进行通讯验证，结果模块收不到另一个模块发出动数据，请问是什么原因？

static void node_read_task(void *arg)
{
    mdf_err_t ret = MDF_OK;
    cJSON *pJson = NULL;
    cJSON *pSub  = NULL;
    char *data    = MDF_MALLOC(MWIFI_PAYLOAD_LEN);
    size_t size   = MWIFI_PAYLOAD_LEN;
    mwifi_data_type_t data_type      = {0x0};
    //uint8_t src_addr[MWIFI_ADDR_LEN] = {0x0};
     //uint8_t src_addr[MWIFI_ADDR_LEN]={0x24,0x0a,0xc4,0xc0,0xfc,0x70};
     // uint8_t src_addr[MWIFI_ADDR_LEN]={0x24,0x0a,0xc4,0xc1,0x1e,0x08};
    uint8_t src_addr[MWIFI_ADDR_LEN]={0x9c,0xa6,0x15,0x1d,0x3d,0x41};
    MDF_LOGI("Note read task is running");
    for (;;) {
        if (!mwifi_is_connected()) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }
        size = MWIFI_PAYLOAD_LEN;
        memset(data, 0, MWIFI_PAYLOAD_LEN);
        //src_addr = {0x24,0x0a,0xc4,0xc0,0xfc,0x70};
        ret = mwifi_read(src_addr, &data_type, data, &size, portMAX_DELAY);
        printf("倒计%s秒\n",data);
       // uart_write_bytes(UART_NUM_1, (const char *) data , size);
        MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_read", mdf_err_to_name(ret));
        MDF_LOGD("Node receive1: " MACSTR ", size: %d, data: %s", MAC2STR(src_addr), size, data);

        pJson = cJSON_Parse(data);
        MDF_ERROR_CONTINUE(!pJson, "cJSON_Parse, data format error, data: %s", data);

        pSub = cJSON_GetObjectItem(pJson, "status");

        if (!pSub) {
            MDF_LOGW("cJSON_GetObjectItem, Destination address not set");
            cJSON_Delete(pJson);
            continue;
        }

        gpio_set_level(CONFIG_LED_GPIO_NUM, pSub->valueint);

        cJSON_Delete(pJson);
    }

    MDF_LOGW("Note read task is exit");

    MDF_FREE(data);
    vTaskDelete(NULL);
}

static void node_write_task(void *arg)
{
    mdf_err_t ret = MDF_OK;
    int count     = 0;
    size_t size   = 0;
    char *data    = NULL;
    mwifi_data_type_t data_type      = {0x0};

    //const uint8_t *dest_addrs={0x24,0x0a,0xc4,0xc1,0x1e,0x08};
    //const uint8_t *dest_addrs[MWIFI_ADDR_LEN]= {0x24,0x0a,0xc4,0xc0,0xfc,0x70};
    // const uint8_t *dest_addrs[MWIFI_ADDR_LEN]= {0x24,0x0a,0xc4,0xc1,0x1e,0x08};
    const uint8_t *dest_addrs[MWIFI_ADDR_LEN]= {0x9c,0xa6,0x15,0x1d,0x3d,0x41};

    MDF_LOGI("NODE task is running1");

    for (;;) {
        if (!mwifi_is_connected()) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }

        size = asprintf(&data, "{\"seq1\":%d,\"layer\":%d,\"status\":%d}",
                        count++, esp_mesh_get_layer(), gpio_get_level(CONFIG_LED_GPIO_NUM));

        MDF_LOGD("Node send2, size: %d, data: %s", size, data);
        ret = mwifi_write(dest_addrs, &data_type, data, size, true);
        MDF_FREE(data);
        MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_writedongjie", mdf_err_to_name(ret));

        vTaskDelay(1000/ portTICK_RATE_MS);
    }

    MDF_LOGW("NODE task is exit");

    vTaskDelete(NULL);
}

zhanzhaocheng, 2019-03-08:
你的变量定义是错误 `dest_addrs` 你写成了二级指针

请将:  `const uint8_t *dest_addrs[MWIFI_ADDR_LEN]= {0x9c,0xa6,0x15,0x1d,0x3d,0x41};`
修改为: `const uint8_t dest_addrs[MWIFI_ADDR_LEN]= {0x9c,0xa6,0x15,0x1d,0x3d,0x41};`

> 注: 如果还是不能成功希望你提供一下设备运行的日志

dongjieqing, 2019-03-09:
可以了，谢谢！
倒计{"seq1":189,"layer":1,"status":0}秒
D (64377) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":189,"layer":1,"status":0}
D (64471) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":55,"layer":2,"status":0}
倒计{"seq1":190,"layer":1,"status":0}秒
D (65380) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":190,"layer":1,"status":0}
D (65474) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":56,"layer":2,"status":0}
倒计{"seq1":191,"layer":1,"status":0}秒
D (66383) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":191,"layer":1,"status":0}
D (66482) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":57,"layer":2,"status":0}
倒计{"seq1":192,"layer":1,"status":0}秒
D (67389) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":192,"layer":1,"status":0}
D (67485) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":58,"layer":2,"status":0}
倒计{"seq1":193,"layer":1,"status":0}秒
D (68392) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":193,"layer":1,"status":0}
D (68489) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":59,"layer":2,"status":0}
倒计{"seq1":194,"layer":1,"status":0}秒
D (69395) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":194,"layer":1,"status":0}
D (69492) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":60,"layer":2,"status":0}
倒计{"seq1":195,"layer":1,"status":0}秒
D (70398) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":195,"layer":1,"status":0}
D (70507) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":61,"layer":2,"status":0}
倒计{"seq1":196,"layer":1,"status":0}秒
D (71401) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":196,"layer":1,"status":0}
D (71511) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":62,"layer":2,"status":0}
倒计{"seq1":197,"layer":1,"status":0}秒
D (72404) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":197,"layer":1,"status":0}
D (72516) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":63,"layer":2,"status":0}
倒计{"seq1":198,"layer":1,"status":0}秒
D (73409) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":198,"layer":1,"status":0}
D (73519) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":64,"layer":2,"status":0}
倒计{"seq1":199,"layer":1,"status":0}秒
D (74415) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":199,"layer":1,"status":0}
D (74531) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":65,"layer":2,"status":0}
倒计{"seq1":200,"layer":1,"status":0}秒
D (75419) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":200,"layer":1,"status":0}
D (75534) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":66,"layer":2,"status":0}
倒计{"seq1":201,"layer":1,"status":0}秒
D (76422) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":201,"layer":1,"status":0}
D (76541) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":67,"layer":2,"status":0}
倒计{"seq1":202,"layer":1,"status":0}秒
D (77425) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":202,"layer":1,"status":0}
D (77548) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":68,"layer":2,"status":0}
倒计{"seq1":203,"layer":1,"status":0}秒
D (78428) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":203,"layer":1,"status":0}
D (78560) [mwifi_examples, 276]: Node send2, size: 32, data: {"seq1":69,"layer":2,"status":0}
倒计{"seq1":204,"layer":1,"status":0}秒
D (79431) [mwifi_examples, 228]: Node receive1: 24:0a:c4:c1:1e:08, size: 33, data: {"seq1":204,"layer":1,"status":0}

zhanzhaocheng, 2019-03-10:
如果问题已经解决,请修改问题的描述并关闭它,以便其他开发者也能避免此类问题,谢谢

closed 2019-03-11 by dongjieqing