espressif/esp-mdf #32: 获得uart数据并用wifi发送到服务器问题

dongjieqing, 2019-03-09:
我有一个数组，用来保存串口发送过来动数据char buffer_data[255];我要将该数据用mwifi_write函数发送出去，请问下该怎么写？

zhanzhaocheng, 2019-03-10:
你想问的问题是否是如何将 `串口输入` 与 `ESP-MESH` 发送两者结合吗?

1. 串口的使用示例参见: [外设的使用示例](https://github.com/espressif/esp-idf/tree/8a69ffc36f166c964d7c30188a865847bf199561/examples/peripherals) 
2. ESP-MESH 的发送的使用示例参见: [mwifi 模块使用示例](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi)
3. 两者结合的使用修改如下: 
    - 跑通 `uart_echo` 的示例,并代码移入 mwifi 的示例中
   - 定义简单的通信协议如: 前 6 Byte 为目的地址,之后跟 4 字节的长度, 最后为有数据 
    - 解析串口输入, 并将 `uart_write_bytes` 修改成 `mwifi_write`
4. 如果无法正常运行请提供你的示例代码,详细描述你的需求和遇到的问题

dongjieqing, 2019-03-11:
您好，我就是这样做的，收到数据后用uart_write_bytes发给串口没有问题，但是用mwifi_write有问题。
static void rx_task()
{
    uint8_t* data = (uint8_t*) malloc(RX_BUF_SIZE+1);
    int rxBytes=0;
    mdf_err_t ret = MDF_OK;
    int count = 0;
    size_t size   = 0;
    char* buffer_data=(char*)malloc(sizeof(char)*255);
    char* buffer_data1=(char*)malloc(sizeof(char)*255);
    int send_length =0;
    int k =0;
    mwifi_data_type_t data_type  = {0x0};
    char* dest_addrs=(char*)malloc(sizeof(char)*6);
    int state = 0;
        while (1) {
            if (!mwifi_is_connected()) {
                vTaskDelay(500 / portTICK_RATE_MS);
                continue;
                }
            int rxBytes = uart_read_bytes(UART_NUM_1, data, BUF_SIZE, 20 / portTICK_RATE_MS);

            int i=0;
            int m=0;
            if(rxBytes > 5) {

                do{
                    i++;
                }while(data[i-1]!=0x70);
                for(int j = 0;j < 255;j++) {
                    buffer_data[j] = data[i+j-1];
                    if (buffer_data[j] == 0x63) {
                        count = j + 1;
                        //Tool_short(buffer_data,buffer_data1);
                       // for(int k=0; k < 6;k++){
                        //   dest_addrs[k]=buffer_data1[k+5];
                        //    if(dest_addrs[k]==0xff) {
                        //        m++;
                        //    }
                       // }
                        //if(m == 6){
                            dest_addrs[MWIFI_ADDR_LEN]=NULL;
                        //}
                        //printf("数据长度%d\n",sizeof(buffer_data)/sizeof((buffer_data)[0]));
                        uart_write_bytes(UART_NUM_1, (const char *) buffer_data, count);
                        mwifi_write(NULL, &data_type, buffer_data,255, true);

                        break;
                    }
                }
            }
        }
    free(data);
}

zhanzhaocheng, 2019-03-11:
1. `但是用mwifi_write有问题` 具体是什么问题,请**详细描述**并提供根节点与子节点的日志.
2. 你的 `uart_write_bytes` 与 `mwifi_write` 的数据长度是否一致的, 并且希望你在调用时加入返回值检查
[image]
3. 修改 mwifi 模块的日志等级: 在 `app_main` 中调用它 `esp_log_level_set("mwifi", ESP_LOG_DEBUG);`

dongjieqing, 2019-03-11:
修改后一样，我用串口发送的数据为70 03 00 13 48 ff 45 ff ff ff ff a5  77 77 77 55 63，hex发送，串口接收没问题，服务器接收为Len:38】:"addr":"24:0a:c4:c0:fc:70","data":p
【2019-03-11 08:36:12收到数据,192.168.0.168:53090 Len:68】:"addr":"24:0a:c4:c0:f4:28","data":"seq":548,"layer":1,"status":0
【2019-03-11 08:36:15收到数据,192.168.0.168:53090 Len:68】:"addr":"24:0a:c4:c0:f4:28","data":"seq":549,"layer":1,"status":0
【2019-03-11 08:36:18收到数据,192.168.0.168:53090 Len:68】:"addr":"24:0a:c4:c0:f4:28","data":"seq":550,"layer":1,"status":0
【2019-03-11 08:36:21收到数据,192.168.0.168:53090 Len:68】:"addr":"24:0a:c4:c0:f4:28","data":"seq":551,"layer":1,"status":0
【2019-03-11 08:36:23收到数据,192.168.0.168:53090 Len:38】:"addr":"24:0a:c4:c0:fc:70","data":p
【2019-03-11 08:36:24收到数据,192.168.0.168:53090 Len:68】:"addr":"24:0a:c4:c0:f4:28","data":"seq":552,"layer":1,"status":0
【2019-03-11 08:36:27收到数据,192.168.0.168:53090 Len:68】:"addr":"24:0a:c4:c0:f4:28","data":"seq":553,"layer":1,"status":0
代码修改后为
static void rx_task()
{
    uint8_t* data = (uint8_t*) malloc(RX_BUF_SIZE+1);
    int rxBytes=0;
    mdf_err_t ret = MDF_OK;
    int count = 0;
    size_t size   = 0;
    uint8_t* buffer_data=(uint8_t*) malloc(RX_BUF_SIZE+1);
    uint8_t* buffer_data1=(uint8_t*) malloc(RX_BUF_SIZE+1);
    int send_length =0;
    int k =0;
    mwifi_data_type_t data_type  = {0x0};
    uint8_t* dest_addrs=(uint8_t*) malloc(RX_BUF_SIZE+1);
    int state = 0;
        while (1) {
            if (!mwifi_is_connected()) {
                vTaskDelay(500 / portTICK_RATE_MS);
                continue;
                }
            int rxBytes = uart_read_bytes(UART_NUM_1, data, BUF_SIZE, 20 / portTICK_RATE_MS);

            int i=0;
            int m=0;
            if(rxBytes > 5) {

                do{
                    i++;
                }while(data[i-1]!=0x70);
                for(int j = 0;j < 255;j++) {
                    buffer_data[j] = data[i+j-1];
                    if (buffer_data[j] == 0x63) {
                        count = j + 1;
                        Tool_short(buffer_data,buffer_data1);
                       for(int k=0; k < 6;k++){
                          dest_addrs[k]=buffer_data1[k+5];
                           if(dest_addrs[k]==0xff) {
                               m++;
                          }
                       }
                        //if(m == 6){
                         //   dest_addrs[MWIFI_ADDR_LEN]=NULL;
                        //}
                        //printf("数据长度%d\n",sizeof(buffer_data)/sizeof((buffer_data)[0]));
                        uart_write_bytes(UART_NUM_1, (const char *) buffer_data, count);
                        //mwifi_write(NULL, &data_type, buffer_data,255, true);
                        ret = mwifi_write(NULL, &data_type, buffer_data, count, true);
                        MDF_FREE(buffer_data);
                        MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_write_uart", mdf_err_to_name(ret));
                        break;
                    }
                }
            }
        }
    free(data);
}
日志打印为：
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
[... 261 lines cut ...]
D (1456) [mwifi, 173]: max_layer             : 16
D (1461) [mwifi, 174]: max_connection        : 6
D (1466) [mwifi, 175]: capacity_num          : 256
D (1470) [mwifi, 177]: ****************  Stability  ****************
D (1477) [mwifi, 178]: assoc_expire_ms       : 30000
D (1482) [mwifi, 180]: beacon_interval_ms    : 100
D (1487) [mwifi, 181]: passive_scan_ms       : 300
D (1492) [mwifi, 182]: monitor_duration_ms   : 60000
D (1497) [mwifi, 183]: cnx_rssi              : -120
D (1502) [mwifi, 184]: select_rssi           : -78
D (1507) [mwifi, 185]: switch_rssi           : -78
D (1511) [mwifi, 187]: **************** Transmission ****************
D (1518) [mwifi, 188]: xon_qsize             : 64
D (1523) [mwifi, 189]: retransmit_enable     : true
D (1528) [mwifi, 191]: **************** Router info  ****************
D (1534) [mwifi, 192]: ssid                  : TP-LINK_3D41
D (1540) [mwifi, 193]: password              : sdl05927680519
D (1546) [mwifi, 194]: bssid                 : 00:00:00:00:00:00
D (1552) [mwifi, 195]: mesh_type             : idle
D (1557) [mwifi, 196]: mesh_id               : 39:43:41:36:31:35
D (1563) [mwifi, 197]: mesh_password         : sdl05927680519
D (1568) [mwifi, 198]: mesh_authmode         : 4
D (1573) [mwifi, 199]: channel               : 0
D (1578) [mwifi, 200]: allow_router_switch   : true
D (1583) [mwifi, 201]: allow_channel_switch  : true

zhanzhaocheng, 2019-03-11:
1. 原因: 内存越界访问了
[image]

2. 问题分析
- 请详细阅读[异常处理](https://docs.espressif.com/projects/esp-mdf/zh_CN/latest/api-guides/error-handling.html)
- buf 的大小是否越界.   确认其大小

[image]

- task 栈空间是否太小.   将 rx_task 的栈空间调整到 8KByte
- 代码解析的问题.  > 简化代码逻辑,逐步增加入代码
    1. 先保证之前的代码正常运行 
    2. 只在 `uart_read` 后加入 `mwifi_write` 只发固定的数据. 
    3. 引入 `uart_read` 的数据解析. 
    4. 将解析后的数据放入 `mwifi_write` 中

dongjieqing, 2019-03-11:
我定义一个常量发送是可以的，后面我接收到数据解析直接发送，和原来的现象一样动。
static void rx_task()
{
    mwifi_data_type_t data_type  = {0x0};
    mdf_err_t ret = MDF_OK;
    uint8_t *data = (uint8_t *) malloc(BUF_SIZE);
    uint8_t *data1 = (uint8_t *) malloc(BUF_SIZE);
    while (1) {
        if (!mwifi_is_connected()) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }
        // Read data from the UART
        int len = uart_read_bytes(UART_NUM_1, data, BUF_SIZE, 20 / portTICK_RATE_MS);
        // Write data back to the UART
        //printf("数据长度%d\n",len);
        if(len > 10) {
            data1=data;
           // uart_write_bytes(UART_NUM_1, (const char *) data, len);
            //uint8_t data1[]="hello";
            printf("数据长度df%d\n",len);
            ret = mwifi_write(NULL, &data_type, (const char *) data1,len, true);
            MDF_FREE(data1);
            MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_write_uart", mdf_err_to_name(ret));
            len=0;
        }
    }
    free(data);
}
I (326) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 0
I (327) wifi: mode : sta (24:0a:c4:c0:fc:70)
I (329) [mwifi, 136]: esp-mdf version: 0c64172
W (331) wifi: [beacon]new interval:100ms
I (336) wifi: mode : sta (24:0a:c4:c0:fc:70) + softAP (24:0a:c4:c0:fc:71)
I (344) wifi: Init max length of beacon: 752/752
I (346) wifi: Init max length of beacon: 752/752
I (353) mesh: <nvs>read layer:0
I (354) mesh: <nvs>read assoc:0
W (356) wifi: [beacon]interval:100ms
I (1365) wifi: Set ps type: 0

I (1370) wifi: mode : sta (24:0a:c4:c0:fc:70)
I (1373) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:1
D (1374) [mwifi, 61]: esp_mesh_event_cb event.id: 0
I (1377) [mwifi, 97]: MESH is started
I (1382) [mwifi_examples, 366]: event_loop_cb, event: 0
I (1387) [mwifi_examples, 370]: MESH is started
D (1393) [mwifi, 161]: **************** Root config ****************
D (1399) [mwifi, 162]: vote_percentage       : 0.90
D (1404) [mwifi, 163]: vote_max_count        : 15
D (1409) [mwifi, 164]: backoff_rssi          : -78
D (1414) [mwifi, 165]: scan_min_count        : 10
D (1418) [mwifi, 166]: attempt_count         : 60
D (1423) [mwifi, 167]: monitor_ie_count      : 10
D (1428) [mwifi, 168]: root_healing_ms       : 6000
D (1433) [mwifi, 169]: root_conflicts_enable : false
D (1438) [mwifi, 170]: fix_root_enable       : false
D (1443) [mwifi, 172]: ****************  Capacity   ****************
D (1449) [mwifi, 173]: max_layer             : 16
D (1454) [mwifi, 174]: max_connection        : 6
D (1459) [mwifi, 175]: capacity_num          : 256
D (1464) [mwifi, 177]: ****************  Stability  ****************
D (1470) [mwifi, 178]: assoc_expire_ms       : 30000
D (1475) [mwifi, 180]: beacon_interval_ms    : 100
D (1480) [mwifi, 181]: passive_scan_ms       : 300
D (1485) [mwifi, 182]: monitor_duration_ms   : 60000
D (1490) [mwifi, 183]: cnx_rssi              : -120
D (1495) [mwifi, 184]: select_rssi           : -78
D (1500) [mwifi, 185]: switch_rssi           : -78
D (1505) [mwifi, 187]: **************** Transmission ****************
[... 43 lines cut ...]
W (9223) wifi: [beacon]new interval:300ms
I (9224) mesh: 1888<arm>parent monitor, my layer:2(cap:16), interval:5941ms, retries:1<normal connected>
D (9229) [mwifi, 61]: esp_mesh_event_cb event.id: 7
I (9235) [mwifi, 66]: Parent is connected
I (9239) [mwifi_examples, 366]: event_loop_cb, event: 7
D (9240) [mwifi, 61]: esp_mesh_event_cb event.id: 11
I (9246) [mwifi_examples, 374]: Parent is connected on station interface
D (9251) [mwifi, 61]: esp_mesh_event_cb event.id: 14
I (9258) [mwifi_examples, 366]: event_loop_cb, event: 11
I (9268) [mwifi_examples, 366]: event_loop_cb, event: 14
W (9623) wifi: [beacon]new interval:1000ms
I (9626) mesh: [scan]new scanning time:1500ms
I (15314) mesh: 4415<active>parent layer:1, channel:7, rssi:-39, assoc:2(cnx rssi threshold:-120)my_assoc:0
I (16818) mesh: 5159<scan>parent layer:1, rssi:-40, assoc:2(cnx rssi threshold:-120)
I (16819) mesh: [SCAN][ch:7]AP:2, other(ID:0, RD:0), MAP:2, idle:0, candidate:2, root:1, topMAP:0[c:2,i:2][9c:a6:15:1d:3d:41]<weak>
I (16827) mesh: 6545[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
I (16834) mesh: 616[monitor]no change, parent:24:0a:c4:c0:f4:29, rssi:-40
I (16840) mesh: 1888<arm>parent monitor, my layer:2(cap:16), interval:11218ms, retries:2<>
数据长度df17
I (28336) mesh: 4415<active>parent layer:1, channel:7, rssi:-38, assoc:2(cnx rssi threshold:-120)my_assoc:0
I (29840) mesh: 5159<scan>parent layer:1, rssi:-43, assoc:2(cnx rssi threshold:-120)
I (29841) mesh: [SCAN][ch:7]AP:2, other(ID:0, RD:0), MAP:2, idle:0, candidate:2, root:1, topMAP:1[c:0,i:2][9c:a6:15:1d:3d:41]<weak>
I (29849) mesh: 6545[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
I (29857) mesh: 616[monitor]no change, parent:24:0a:c4:c0:f4:29, rssi:-43
I (29863) mesh: 1888<arm>parent monitor, my layer:2(cap:16), interval:7649ms, retries:3<>

zhanzhaocheng, 2019-03-11:
你将内存处理错误, 将 data1 指向了 data,  然后又 free(data1), 这就相当于 把 data 释放了, 而你之后又使用了, data 的空间

[image]

dongjieqing, 2019-03-11:
我将MDF_FREE(data1);屏蔽也是错的，请问该怎么写？

dongjieqing, 2019-03-11:
我定义乐一个常量uint8_t buffer_data[17]="123456789abcdefg";对其中的元素一个个赋值
buffer_data[0]='d';
buffer_data[1]='j';
buffer_data[16]='e';
这样赋值没有问题，但是换一种赋值就出问题了。
for (int j = 0; j < len; j++) {
               char me=data[j];
                buffer_data[j] =  me;
                printf("data[j]:= %d \n", me);
            }
不用for循环，一个个动赋值也不对。
static void rx_task()
{
    mwifi_data_type_t data_type  = {0x0};
    mdf_err_t ret = MDF_OK;
    uint8_t *data = (uint8_t *) malloc(BUF_SIZE);
    while (1) {
        if (!mwifi_is_connected()) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }
        // Read data from the UART
        int len = uart_read_bytes(UART_NUM_1, data, BUF_SIZE, 20 / portTICK_RATE_MS);
        // Write data back to the UART
        //printf("数据长度%d\n",len);
        int i=0;
        int m=0;
        if(len > 10) {
            uart_write_bytes(UART_NUM_1, (const char *) data, len);
            //do{
            //    i++;
           // }while(data[i-1]!=0x70);
            uint8_t buffer_data[17]="123456789abcdefg";
            for (int j = 0; j < len; j++) {
               char me=data[j];
                buffer_data[j] =  me;
                printf("data[j]:= %d \n", me);
            }
            //    buffer_data[j] = data[j];
               // if (buffer_data[j] == 0x63) {
               //     count = j + 1;
               // }
           // }
            //buffer_data[0]='d';
            //buffer_data[1]='j';
            //buffer_data[16]='e';
            //strcpy(buffer_data, data)
            //printf("data[0]:= %d \n", data[0]);
            //data1[1]=data[1];
            printf("数据长度df%d\n",len);
            ret = mwifi_write(NULL, &data_type, (const char *) buffer_data,len, true);
            MDF_FREE(*buffer_data);
            MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_write_uart", mdf_err_to_name(ret));
            len=0;
        }
    }
    free(data);
}
搞不懂是什么原因。应该就是这里的问题，您可以用两个模块试试。

zhanzhaocheng, 2019-03-11:
你直接将你修改的整个 c 文件发给, 我来给你调试

dongjieqing, 2019-03-11:
好的
/*
 * ESPRESSIF MIT License
 *
 * Copyright (c) 2018 <ESPRESSIF SYSTEMS (SHANGHAI) PTE LTD>
 *
 * Permission is hereby granted for use on all ESPRESSIF SYSTEMS products, in which case,
 * it is free of charge, to any person obtaining a copy of this software and associated
 * documentation files (the "Software"), to deal in the Software without restriction, including
 * without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the Software is furnished
 * to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all copies or
 * substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 */
//#include <stdio.h>
//#include "freertos/FreeRTOS.h"
//#include "freertos/task.h"
#include "mdf_common.h"
#include "mwifi.h"
#include "driver/uart.h"
#include <string.h>

// #define MEMORY_DEBUG
#define ECHO_TEST_TXD  (GPIO_NUM_4)
#define ECHO_TEST_RXD  (GPIO_NUM_5)
#define ECHO_TEST_RTS  (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS  (UART_PIN_NO_CHANGE)
#define UART_START_CODE 0x70     //串口通讯头码$
#define UART_END_CODE 0x23     //串口通讯结束码#
#define BUF_SIZE (1024)
#define RD_BUF_SIZE (BUF_SIZE)
#define RX_BUF_SIZE (BUF_SIZE)

static int g_sockfd    = -1;
static const char *TAG = "mwifi_examples";

void tcp_client_write_task(void *arg)
{
    mdf_err_t ret = MDF_OK;
    char *data    = MDF_CALLOC(1, MWIFI_PAYLOAD_LEN);
    size_t size   = MWIFI_PAYLOAD_LEN;
    uint8_t src_addr[MWIFI_ADDR_LEN] = {0x0};
    mwifi_data_type_t data_type      = {0x0};

    MDF_LOGI("TCP client write task is running");

    while (mwifi_is_connected()) {
        if (g_sockfd == -1) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }

        size = MWIFI_PAYLOAD_LEN - 1;
        memset(data, 0, MWIFI_PAYLOAD_LEN);
        ret = mwifi_root_read(src_addr, &data_type, data, &size, portMAX_DELAY);
        MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_root_read", mdf_err_to_name(ret));

        char *json_data = NULL;
        int json_size   = asprintf(&json_data, "{\"addr\":\"" MACSTR "\",\"data\":%s}",
                                   MAC2STR(src_addr), data);
[... 512 lines cut ...]
{
    init();
    /**
     * @breif Create handler
     */

    xTaskCreate(rx_task, "uart_rx_task", 2 * 1024,
                NULL, configMAX_PRIORITIES, NULL);

/*
xTaskCreate(node_read_task, "node_read_task", 4 * 1024,
NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);

xTaskCreate(node_write_task, "node_write_task", 4 * 1024,
NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);

xTaskCreate(node_read_task, "node_read_task", 4 * 1024,
NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);

TimerHandle_t timer = xTimerCreate("print_system_info", 10000 / portTICK_RATE_MS,
                   true, NULL, print_system_info_timercb);
xTimerStart(timer, 0);
*/

}

zhanzhaocheng, 2019-03-11:
你直接将 c 当成一个附件传给我, 直接往输入框中, 或者在当成代码插入, 不然会被网页错误解析
[image]

dongjieqing, 2019-03-11:
``

dongjieqing, 2019-03-11:
`
[mwifi_example.c.tar.gz](https://github.com/espressif/esp-mdf/files/2951230/mwifi_example.c.tar.gz)

`

dongjieqing, 2019-03-11:
请问下你们发HEX格式数据是不是必须在最后加上0x0a，我刚不发hex，发字符类型，发现收到的数据中多了一个0x0a，后面我发hex数据的时候多发一个0x0a,服务器收到的数据就正常了。

dongjieqing, 2019-03-11:
请教一下，你们wifi模块是否只能发ASSCII码对应的字符串，我发字符串"p123456789c"没有问题，发它对应的HEX没有问题，发HEX70 03 00 13 48 ff 45 ff ff ff ff a5  77 77 77 55 63就有问题了！

zhanzhaocheng, 2019-03-11:
1. ``就有问题了!``    是指什么问题 **请详细说明**, 是指服务端收不到, 还是打印异常, 如果是打印异常, 是因为我定义与服务器的通信是 字符串的格式, 如果你需要使用二进制请修改如下部分:

2. 运行了你的示例存在, 内存泄露\溢出等, 我重新编写示例如下: 
```c
typedef struct {
    size_t size;
    uint8_t data[BUF_SIZE];
} uart_data_t;

static void rx_task()
{
    mdf_err_t ret               = MDF_OK;
    uint8_t temp                = 0;
    uart_data_t *uart_data      = MDF_CALLOC(1, sizeof(uart_data_t));
    mwifi_data_type_t data_type = {0x0};

    while (1) {
        if (!mwifi_is_connected()) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }

        mdf_err_t ret = uart_read_bytes(UART_NUM_1, &temp, 1, portMAX_DELAY);
        MDF_ERROR_CONTINUE(ret <= 0, "<%s> uart_read_bytes", mdf_err_to_name(ret));

        MDF_LOGW("size: %d, temp: %02x", uart_data->size, temp);
        uart_data->data[uart_data->size] = temp;
        uart_data->size++;

        if (temp == UART_START_CODE) {
            uart_data->data[0] = temp;
            uart_data->size = 1;
        } else if (temp == UART_END_CODE) {
            MDF_LOGI("size: %d, data:", uart_data->size);
            ESP_LOG_BUFFER_HEX(TAG, uart_data->data, uart_data->size);

            ret = mwifi_write(NULL, &data_type, uart_data->data, uart_data->size, true);
            MDF_ERROR_CONTINUE(ret != MDF_OK, "<%s> mwifi_write_uart", mdf_err_to_name(ret));

            uart_data->size = 0;
        } else if (uart_data->size >= BUF_SIZE - 1) {
            uart_data->size = 0;
            MDF_LOGW("Data exceeds maximum length");
        } else {
            MDF_LOGV("0x%02x", temp);
        }
    }

    MDF_FREE(uart_data);
}
```

dongjieqing, 2019-03-11:
好的，谢谢！

closed 2019-03-11 by dongjieqing