espressif/esp-mdf #33: 在mdf一个例子中引入其他功能头文件 一直报错

chenguang3312, 2019-03-11:
我在mwifi例子中想改成手机配网，而不是设置固定wifi和密码。所以我在mwifi中移植了mconfig中的配网方式，我将mconfig所需的头文件，在mwifi的main.c中引用了，但是一直报错，引用的头文件里的函数undefined，所以我去mdf的components目录里找到对于的c文件拷贝到mwif下的main目录了。编译的时候又报错误

`In file included from C:/msys32/home/Administrator/esp/esp-mdf/components/mconfi                                                                                                                g/include/mconfig_blufi.h:32:0,
                 from C:/msys32/home/Administrator/esp/esp-mdf/components/mcommo                                                                                                                n/include/mdf_common.h:66,
                 from C:/msys32/home/Administrator/esp/esp-mdf/components/mconfi                                                                                                                g/include/mconfig_queue.h:28,
                 from C:/msys32/home/Administrator/esp/mwifi/main/mconfig_queue.                                                                                                                c:25:
C:/msys32/home/Administrator/esp/esp-mdf/components/mconfig/include/mconfig_chai                                                                                                                n.h:99:38: error: unknown type name 'mconfig_data_t'
 mdf_err_t mconfig_chain_master(const mconfig_data_t *config, TickType_t duratio                                                                                                                n_ticks);
                                      ^
make[1]: *** [/home/Administrator/esp/esp-mdf/esp-idf/make/component_wrapper.mk:                                                                                                                290：mconfig_queue.o] 错误 1
make: *** [C:/msys32/home/Administrator/esp/esp-mdf/esp-idf/make/project.mk:551                                                                                                                ：component-main-build] 错误 2

`
我想问下，这个错误怎么解决。mdf的一般编程步骤或者工程框架是怎样的，直接在main中include其他功能头文件为啥不行。期待您的解答 谢谢！

zhanzhaocheng, 2019-03-11:
1.  报错的原因:
 - 使用方法错误,你无需将 `mconfig_queue.c` 移到了你的代码中,直接调用其相关的 API 即可, 参见: [mconfig_example](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mconfig)
- 如果你报错如下, 是因为配置 配网需要用到蓝牙, 你需要通过 `make menuconfig` 打开它, 基于你的需求推荐你可以参考 [light 的配置](), 来替换`mwifi` 的默认配置

[image]

[image]

运行如下命令跟换配置

```shell
cd examples/function_demo/mwifi
rm sdkconfig sdkconfig.old -rf
cp ../../development_kit/light/sdkconfig.defaults .
```
2. mdf的一般编程步骤或者工程框架
    1. 阅读相关文档
        - [README.md](https://github.com/espressif/esp-mdf/blob/master/README_cn.md
) 对 ESP-MDF 的框架有详细的说明
       - [ESP-MDF 编程指南](https://docs.espressif.com/projects/esp-mdf/zh_CN/latest/get-started/index.html],
        - 使用示例时,详细阅读示例下的 README.md

    2. 先运行 `development_kit` 示例, 这里面的示例是使用 `esp-mdf` 开发出的一个完整的方案, 包含有智能灯,开关和传感器, 可以通过 APP 进行控制升级\配网等. 对 ESP-MESH 的具体使用场景有一定的了解
    3. 运行 `function_demo` 示例, 这里是各个模块的使用说用, 由于没有应用逻辑你可以更好的了解 API 的使用.
    4. 跟据你的实际使用 选择相应的模块与你的具体相结合
    5. 如果出现问题可以即时很我们反馈

chenguang3312, 2019-03-11:
1.这个sdkconfig的配置文件是执行make menuconfig后自动生成的嘛？ 我是直接把mconfig里用的头文件添加进去的，然后报的有函数undefined，后来就直接把头文件对应的C文件也拷过去的。
2. 那正常开发是不是只要在主文件里引入对应功能头文件，在配置中打开功能就可以了？
3.那个配置文件有没有官方说明文档，我怎么查每个参数对应啥意思。
4.还有点不明白的就是在make menuconfig 中设置的哪些选项，如何自定义。然后它是怎么传给.c文件的。比如mwifi中在menuconfig中设置的路由器ssid 和passwd 怎么在.c文件里接收的。

望解答，谢谢

zhanzhaocheng, 2019-03-11:
1. `sdkconfig` 是你修改的 `make menuconfig` 加上 `sdkconfig.defaults` 自动生成的, 你直接替换 `sdkconfig.defaults`  或者通过 `make menuconfig` 来打开蓝牙了都是可以的

```shell
cd examples/function_demo/mwifi
cp ../../development_kit/light/sdkconfig.defaults .
make defconfig
```
2. 那正常开发只要在主文件里引入对应功能头文件就可以. 但由于打开蓝牙后内存会减少并固件会变大, 因此只有在需要时才打开. 

3.  `make menufig` 是标准的与 linux 基本相同 , 配置文件的参数, 你可以配置时通过 `help` 来查看.  

4.  `make menuconfig` 会读取 "Kconfig.projbuild" 生成配置项, 当修改配置后, 会将配置会生成为sdkconfig文件, 编译时会将`sdkconfig` 文件生成 `sdkconfig.h` 即宏定义. 详见 [构建系统](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/api-guides/build-system.html)

chenguang3312, 2019-03-11:
In file included from C:/msys32/home/Administrator/esp/esp-mdf/components/mconfig/include/mconfig_blufi.h:32:0,
                 from C:/msys32/home/Administrator/esp/esp-mdf/components/mcommon/include/mdf_common.h:66,
                 from C:/msys32/home/Administrator/esp/esp-mdf/components/mconfig/include/mconfig_queue.h:28,
                 from C:/msys32/home/Administrator/esp/esp-mdf/components/mconfig/mconfig_queue.c:25:
C:/msys32/home/Administrator/esp/esp-mdf/components/mconfig/include/mconfig_chain.h:99:38: error: unknown type name 'mconfig_data_t'
 mdf_err_t mconfig_chain_master(const mconfig_data_t *config, TickType_t duration_ticks);
                                      ^
make[1]: *** [/home/Administrator/esp/esp-mdf/esp-idf/make/component_wrapper.mk:290：mconfig_queue.o] 错误 1
make: *** [C:/msys32/home/Administrator/esp/esp-mdf/esp-idf/make/project.mk:551：component-mconfig-build] 错误 2
请问这是啥错误？

zhanzhaocheng, 2019-03-11:
1. 你使用 `mconfig_example` 可以正常运行吗?
2. 你是否已经替换了 sdkconfig.defaults

chenguang3312, 2019-03-11:
使用mconfig正常，替换了sdkconfig.defaults   再编译就错了。

chenguang3312, 2019-03-11:
mconfig里面是mconfig_data_t 而mwifi里是mconfig_data  我把mconfig里的这个结构放在mwifi里就报这错误

zhanzhaocheng, 2019-03-11:
我无法复现你的问题,我将两者结合写了一个示例,你试试是否可行 
[mwifi_example.tar.gz](https://github.com/espressif/esp-mdf/files/2951764/mwifi_example.tar.gz)

chenguang3312, 2019-03-11:
可以了 原来的文件使用eclipse时候污染了  重新弄了下环境可以了   非常感谢

ESP-LBB, 2019-03-29:
Closing as this question appears to be answered.

closed 2019-03-29 by ESP-LBB