espressif/esp-mdf #35: App与Esp32本地http通信问题

MasterWen, 2019-03-13:
App与Esp32本地http通信问题

当App快速操作时，Esp32会打印 httpd: httpd_accept_conn: error in accept (23)
 httpd: httpd_server: error accepting new connection

你们没有提供iOS的SDK，所以是我自己写的demo。主要是想问你们客户端如何操作断开之前的连接

fanbaoying, 2019-03-13:
在发送请求之前，添加

`{ NSDate* lastRequestDate;} `

`
if (lastRequestDate) {
        NSLog(@"两次请求时间间隔：%f",[[NSDate date] timeIntervalSinceDate:lastRequestDate]);
        if ([[NSDate date] timeIntervalSinceDate:lastRequestDate]<0.5) {
            return;//过滤频繁操作
        }
    }
    lastRequestDate=[NSDate date];`
可以解决频繁操作导致的，code=-1005

MasterWen, 2019-03-13:
可是我把间隔控制1到2s还是一样的。你们的苹果App也是这样限制的吗？我用你们App点的很快也不会出现问题

fanbaoying, 2019-03-13:
对，我们的app就是这样使用的

fanbaoying, 2019-03-13:
如果还是解决不了，建议你把请求模块的代码给出来，帮你看看是什么问题

MasterWen, 2019-03-14:
这是我的请求代码
AFHTTPSessionManager *manger=[AFHTTPSessionManager manager];
    manger.responseSerializer=[AFHTTPResponseSerializer serializer];
    manger.requestSerializer=[AFJSONRequestSerializer serializer];
    [manger POST:url parameters:dic headers:@{@"Mesh-Node-Mac":mac,@"Connection":@"close",@"root-response":@"true"} progress:^(NSProgress * _Nonnull uploadProgress) {
        
    } success:^(NSURLSessionDataTask * _Nonnull task, id  _Nullable responseObject) {
        success(task,responseObject);
        
    } failure:^(NSURLSessionDataTask * _Nullable task, NSError * _Nonnull error) {
        NSLog(@"++++%@",error);
    }];

fanbaoying, 2019-03-14:
你是用的AFHTTPSessionManager这是第三方AFNetworking的网络请求，项目中我使用的是原生网络请求
这是我的代码，ipUrl请求地址，requestHeader请求头，bodyContent请求体，success成功返回，failure失败返回。
`- (NSURLSessionTask *)requestWithIpUrl:(NSString *)ipUrl withRequestHeader:(NSDictionary *)requestHeader withBodyContent:(NSDictionary *)bodyContent andSuccess:(void (^)(NSDictionary * _Nonnull))success andFailure:(void (^)(int))failure {

    NSURL *url=[NSURL URLWithString:ipUrl];
    NSMutableURLRequest *request=[NSMutableURLRequest requestWithURL:url];

    //设置提交方式
    [request setHTTPMethod:@"post"];
    [request setValue:@"application/json" forHTTPHeaderField:@"Content-Type"];
    [request setValue:requestHeader[@"meshNodeMac"] forHTTPHeaderField:@"Mesh-Node-Mac"];
    [request setValue:requestHeader[@"meshNodeNum"] forHTTPHeaderField:@"Mesh-Node-Num"];
    if (requestHeader[@"rootResponse"]) {
        [request setValue:requestHeader[@"rootResponse"] forHTTPHeaderField:@"Root-Response"];
    }

    //提交参数
    NSData *data = [NSJSONSerialization dataWithJSONObject:bodyContent options:0 error:nil];
    [request setHTTPBody:data];

    //会话
    NSURLSession *session=[NSURLSession sharedSession];
    NSURLSessionDataTask *task=[session dataTaskWithRequest:request completionHandler:^(NSData * _Nullable data, NSURLResponse * _Nullable response, NSError * _Nullable error) {

        if (error) {
            NSLog(@"localizedDescription-->%@",error.localizedDescription);
            failure ? failure((int)error) : nil;
        }else {
            NSMutableDictionary *jsonMutableDic = [NSMutableDictionary dictionaryWithCapacity:0];
            if ([@"-1" isEqualToString:[NSString stringWithFormat:@"%lld",response.expectedContentLength]]) {
                NSArray *dataArr = [EspDeviceUtil getChunkedResponseListForData:data];
                NSLog(@"%@",dataArr);
            }else {
                NSDictionary *jsonDict = [NSJSONSerialization JSONObjectWithData:data options:NSJSONReadingMutableLeaves error:nil];
                jsonMutableDic[@"result"] = jsonDict;
                success ? success(jsonMutableDic) : nil;
            }
        }
    }];
    //启动任务
    [task resume];
    return task;
}`
你先替换成我的这种网络请求，看是否能解决问题，我尝试使用AFNetworking看能否把你的问题重现。

还有请问你哪里的设备连接的Wi-Fi是否有外网？AFNetworking库如果没有外网，进行网络请求是会连不上服务的。

MasterWen, 2019-03-14:
谢谢，我试了你的方法确实没有什么问题。

ESP-LBB, 2019-03-29:
Closing as this question appears to be answered.

closed 2019-03-29 by ESP-LBB

MasterWen, 2019-04-08:
你好，我想问下，刷新获取多个设备状态时（post请求get_status），设备返回的数据是什么格式的，如何处理。一个设备是json格式。多个设备就不是了

fanbaoying, 2019-04-10:
@MasterWen 你好，当有多个设备时，返回的数据中每个设备格式类似是这样的：
[image]
处理方法：
判断response.expectedContentLength值是否为-1，返回值为-1时是多个设备返回数据，对返回的data数据进行处理：

`+ (NSArray<EspHttpResponse *> *)getChunkedResponseListForData:(NSData *)data {
    // TODO
    if ([EspCommonUtils isNull:data]) {
        return nil;
    }
    NSMutableArray *result = [NSMutableArray array];
    Byte *bytes = (Byte *)[data bytes];
    NSMutableData *temp = [NSMutableData data];
    long totalLen = data.length;
    for (long index = 0; index < totalLen; index++) {
        // Get header data
        Byte headerBytes[1] = { bytes[index] };
        [temp appendBytes:headerBytes length:1];
        if (![EspDeviceUtil headEnd:temp]) {
            continue;
        }
        // Get content length
        NSString *headStr = [[NSString alloc] initWithData:temp encoding:NSUTF8StringEncoding];
        NSArray *headArray = [headStr componentsSeparatedByString:@"\r\n"];
        long contentLength = -1;
        for (NSString *s in headArray) {
            NSArray *kv = [s componentsSeparatedByString:@": "];
            @try {
                NSString *headerName = [kv[0] uppercaseString];
                if ([headerName isEqualToString:[EspHttpHeaderContentLength uppercaseString]]) {
                    contentLength = [kv[1] intValue];
                    break;
                }
            }
            @catch (NSException *e) {
                NSLog(@"%@", e.name);
            }
        }
        // Get content data
        for (long contentCount = 0; contentCount < contentLength; contentCount++) {
            index++;
            if (index >= totalLen) {
                break;
            }
            Byte contentBytes[1] = {bytes[index]};
            [temp appendBytes:contentBytes length:1];
        }
        // Get EspHttpResponse
        EspHttpResponse *response = [EspHttpUtils getResponseWithFixedLengthData:temp];
        if (response != nil) {
            [result addObject:response];
        }
        // Clear temp data
        [temp replaceBytesInRange:NSMakeRange(0, [temp length]) withBytes:NULL length:0];
    }
    return result;
}`
处理后为数组格式，再进行解析
`NSArray *dataArr = [EspDeviceUtil getChunkedResponseListForData:data];`