espressif/esp-mdf #36: Looking for change LED active status

thangta, 2019-03-27:
Hi, 
I'm running  example/development kit/light on my ESP32 custom LED board. My LED is driven by low logic (LED shall light at 0 logic control). ESP light example using active high to turn on LED.  Could you let me know where is place in the code that I can change logic output of light driver.

lqq96, 2019-03-28:
You can try to modify your code based on the tips of the image and then verify that the changes meet your needs.
[image]
[image]
[image]

ESP-LBB, 2019-04-11:
Closing as this question appears to be answered.

closed 2019-04-11 by ESP-LBB