espressif/esp-mdf #39: Error with Mlink when answering an HTTP req in a pvTimer callback

jonathand1, 2019-03-30:
After some investigation, I've found an issue in MLink HTTP server...

the "mlink_connection_add" function starts a timer which fires up another method when timeout is reached.
In this callback, HTTP responses are not working anymore, no matter which API we use :

- sockfd is seen as closed when trying to write though it to answer HTTP req
- when we call _httpd_resp_set_type_ for instance, we get an "_ESP_ERR_HTTPD_INVALID_REQ_" as ret value.

After digging a bit, I've found this is related to the _httpd_valid_req_ function in _httpd_parse.c_. **httpd_os_thread_handle() == hd->hd_td.handle** almost always return false in such case... Looks like callback is done through another "thread" which does not have access to that "handle" and does not have the ability to use the sockfd somehow.

I'm available for any question.

zhanzhaocheng, 2019-04-02:
I am not very clear about your needs, can you describe it again, or provide some routines?

Mlink is our defined communication protocol with mobile phones. It has nothing to do with MESH communication. Unless you need to use our mobile APP, you can not use `mlink_xxx` related functions.

If you want to use http you can refer to: https://github.com/espressif/esp-idf/tree/8a69ffc36f166c964d7c30188a865847bf199561/examples/protocols/http_server

jonathand1, 2019-04-02:
Hi,

Thanks for your answer, sorry if I have not been clear enough. I know that MLink has nothing to do with Mesh. But MLink uses the HTTP server.
And on my ESP32 chips, I've done some experiments reusing mlink_httpd implementation. When you send like a device request (I've used Postman to simulate that REST API call), there is a timer instanciated which answers to the HTTP request sender and close connection in case nothing has been done before the timer timeout.

The issue I experience in this timer timeout callback "mlink_connection_timeout_cb".
In this function, sockfd is seen as closed, and we cannot answer anymore to the HTTP request. So I changed the code a bit on my side to directly send the pointer to `httpd_req_t req` as parameter in the callback... 

That's how I've been able to get more information about the issue (when trying to use http server API to answer the request in the callback func). That led me to this `httpd_valid_req ` function in `httpd_parse.c`. and that `httpd_os_thread_handle() == hd->hd_td.handle` which fails.

jonathand1, 2019-04-11:
I extracted the mlink http deamon in a file I rewrote, but the idea of registering a xTimer when a POST device_request is done is the same.

The only difference I have (but it was also going wrong with the implementation done in mlink) is that instead of using **httpd_default_send** in the timer timeout callback, I am using **httpd_queue_work** giving a function ptr. the func ptr given points out to a function which calls in order :

    MDF_LOGW("ESP httpd set type");
    ret = httpd_resp_set_type(resp->request, HTTPD_TYPE_JSON);
    ESP_ERROR_CHECK(ret);
    MDF_LOGW("ESP httpd set status");
    ret = httpd_resp_set_status(resp->request, resp->status_code);
    ESP_ERROR_CHECK(ret);
    MDF_LOGW("ESP httpd send message");
    ret = httpd_resp_send(resp->request, resp->message, strlen(resp->message));
    ESP_ERROR_CHECK(ret);
    MDF_FREE(response);

**httpd_resp_set_type** fails, that is my issue. Imagine I remove that line, then it will be **httpd_resp_set_status** which will fail with that code ESP_ERR_HTTPD_INVALID_REQ

That is the log I get (some of them are mine (httpd)) :

`I (15523) event: sta ip: 192.168.1.163, mask: 255.255.255.0, gw: 192.168.1.1
I (15523) [initd, 19]: event_loop_cb, event: 17
I (15523) [initd, 26]: I am the root node. I obtained an IP address.
I (15533) [initd, 28]: Starting HTTP server...
I (15533) [httpd, 416]: Starting HTTP server
I (15543) [httpd, 417]: HTTP port         : 80
I (15543) [httpd, 418]: Max URI handlers  : 8
I (15553) [httpd, 419]: Max open sessions : 7
I (15553) [httpd, 420]: Max header length : 2048
I (15563) [httpd, 421]: Max URI length    : 2048
I (15563) [httpd, 422]: Max stack size    : 4096
D (15573) httpd_uri: httpd_register_uri_handler: [0] installed /mesh_info
D (15573) httpd_uri: httpd_find_uri_handler: [0] = /mesh_info
D (15583) httpd_uri: httpd_register_uri_handler: [0] exists /mesh_info
D (15593) httpd_uri: httpd_register_uri_handler: [1] installed /device_request
D (25023) httpd_sess: httpd_sess_new: fd = 57
D (25033) httpd_sess: httpd_sess_process: httpd_req_new
D (25033) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (25033) httpd_txrx: httpd_recv_with_opt: received length = 128
D (25043) httpd_parse: read_block: received HTTP request block size = 128
D (25043) httpd_parse: cb_url: message begin
D (25053) httpd_parse: cb_url: processing url = /device_request
D (25063) httpd_parse: verify_url: received URI = /device_request
D (25063) httpd_parse: cb_header_field: headers begin
D (25073) httpd_txrx: httpd_unrecv: length = 97
D (25073) httpd_parse: pause_parsing: paused
D (25083) httpd_parse: cb_header_field: processing field = Content-Type
D (25083) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (25093) httpd_txrx: httpd_recv_with_opt: pending length = 97
D (25093) httpd_parse: read_block: received HTTP request block size = 97
D (25103) httpd_parse: continue_parsing: skip pre-parsed data of size = 13
D (25113) httpd_parse: continue_parsing: un-paused
D (25113) httpd_parse: cb_header_value: processing value = application/json
D (25123) httpd_parse: cb_header_field: processing field = Mesh-Node-Mac
D (25133) httpd_parse: cb_header_value: processing value = b4e62dc0128f
D (25133) httpd_parse: cb_header_field: processing field = cache-control
D (25143) httpd_parse: cb_header_value: processing value = no-cache
D (25153) httpd_parse: cb_header_field: processing field = Postman-Tok
D (25153) httpd_parse: parse_block: parsed block size = 97
D (25163) httpd_txrx: httpd_recv_with_opt: requested length = 128
D (25163) httpd_txrx: httpd_recv_with_opt: received length = 128
D (25173) httpd_parse: read_block: received HTTP request block size = 128
D (25183) httpd_parse: cb_header_field: processing field = en
D (25183) httpd_parse: cb_header_value: processing value = a329bb58-e86e-490d-8001-e930b5037eb6
D (25193) httpd_parse: cb_header_field: processing field = User-Agent
D (25203) httpd_parse: cb_header_value: processing value = PostmanRuntime/7.6.1
D (25213) httpd_parse: cb_header_field: processing field = Accept
D (25213) httpd_parse: cb_header_value: processing value = */*
D (25223) httpd_parse: cb_header_field: processing field = Host
D (25223) httpd_parse: cb_header_value: processing value = 192.168.1.163
D (25233) httpd_parse: cb_header_field: processing field = accept-encoding
[... 28 lines cut ...]
W (27383) [httpd, 110]: ESP httpd set status
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008c590
0x4008c590: _esp_error_check_failed at /home/jdruet/esp/esp-mdf/esp-idf/components/esp32/panic.c:712

file: "/home/jdruet/esp/ESP32_Network/httpd/httpd.c" line 112
func: esp_httpd_resp
expression: ret

Backtrace: 0x4008c148:0x3ffd9370 0x4008c593:0x3ffd9390 0x4014f82a:0x3ffd93b0 0x40170995:0x3ffd93d0 0x40170bba:0x3ffd9410 0x40170ce5:0x3ffd9450 0x4008e751:0x3ffd9470
0x4008c148: invoke_abort at /home/jdruet/esp/esp-mdf/esp-idf/components/esp32/panic.c:707

0x4008c593: _esp_error_check_failed at /home/jdruet/esp/esp-mdf/esp-idf/components/esp32/panic.c:713

0x4014f82a: esp_httpd_resp at /home/jdruet/esp/ESP32_Network/httpd/httpd.c:166

0x40170995: httpd_process_ctrl_msg at /home/jdruet/esp/esp-mdf/esp-idf/components/esp_http_server/src/httpd_main.c:132 (discriminator 6)

0x40170bba: httpd_server at /home/jdruet/esp/esp-mdf/esp-idf/components/esp_http_server/src/httpd_main.c:172 (discriminator 6)

0x40170ce5: httpd_thread at /home/jdruet/esp/esp-mdf/esp-idf/components/esp_http_server/src/httpd_main.c:215

0x4008e751: vPortTaskWrapper at /home/jdruet/esp/esp-mdf/esp-idf/components/freertos/port.c:403

Rebooting...
`

zhanzhaocheng, 2019-04-16:
From the log, it is caused by the http connection being closed.

[image]

1. Is the http client running in your program, I tested this problem when using both http server and client
2. Do you have the initiative to close the connection?
3. The connection has been replied, and the http server will automatically close the connection that has already been replied to.
4. Can I provide a simple sample program and a version of MDF so that I can quickly reproduce the problem?