espressif/esp-mdf #40: 设置app配网后重启不能正常组网是怎么回事

chenguang3312, 2019-03-31:
仿照light例子 做了一个手机配网，然后将配网信息保存到esp32模组中，但是首次使用app配网是，模组能很快的连接网络设置信道，但是模组重启后，信道一直在扫描。下面是监控信息。
`I (30) boot: ESP-IDF v3.3-beta1-506-gebdcbe8c6-dirty 2nd stage bootloader
I (31) boot: compile time 18:06:07
I (32) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (68) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (75) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  3 factory          factory app      00 00 00010000 00140000
I (90) boot:  4 ota_0            OTA app          00 10 00150000 00140000
I (98) boot:  5 ota_1            OTA app          00 11 00290000 00140000
I (105) boot:  6 coredump         Unknown data     01 03 003d0000 00010000
I (113) boot:  7 reserved         Unknown data     01 fe 003e0000 00020000
I (120) boot: End of partition table
I (125) boot: Defaulting to factory image
I (129) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x38128 (229672) map
I (219) esp_image: segment 1: paddr=0x00048150 vaddr=0x3ffbdb60 size=0x03150 ( 12624) load
I (224) esp_image: segment 2: paddr=0x0004b2a8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/Administrator/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (226) esp_image: segment 3: paddr=0x0004b6b0 vaddr=0x40080400 size=0x04960 ( 18784) load
I (242) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xe90fc (954620) map
0x400d0018: _flash_cache_start at ??:?

I (577) esp_image: segment 5: paddr=0x0013911c vaddr=0x40084d60 size=0x129e0 ( 76256) load
0x40084d60: ld_sniff_cntl_offset_adjust at ??:?

I (623) boot: Loaded app from partition at offset 0x10000
I (623) boot: Disabling RNG early entropy source...
I (623) cpu_start: Pro cpu up.
I (627) cpu_start: Application information:
I (632) cpu_start: Project name:     mwifi
I (637) cpu_start: App version:      1
I (641) cpu_start: Compile time:     Mar 23 2019 18:06:36
I (647) cpu_start: ELF file SHA256:  ef16027bf6cc7d09...
I (653) cpu_start: ESP-IDF:          v3.3-beta1-506-gebdcbe8c6-dirty
I (660) cpu_start: Single core mode
I (665) heap_init: Initializing. RAM available for dynamic allocation:
I (672) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (678) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (684) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (690) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (696) heap_init: At 3FFC9298 len 00016D68 (91 KiB): DRAM
I (702) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (708) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (715) heap_init: At 40097740 len 000088C0 (34 KiB): IRAM
I (721) cpu_start: Pro cpu start user code
I (68) esp_core_dump_flash: Init core dump to flash
I (68) esp_core_dump_flash: Found partition 'coredump' @ 3d0000 65536 bytes
E (70) esp_core_dump_common: Failed to mmap core dump data (260)!
I (77) cpu_start: Starting scheduler on PRO CPU.
I (453) wifi: wifi driver task: 3ffcd414, prio:23, stack:3584, core=0
I (454) wifi: wifi firmware version: 955b7af
I (454) wifi: config NVS flash: enabled
I (456) wifi: config nano formating: disabled
I (460) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (470) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (488) wifi: Init dynamic tx buffer num: 32
I (488) wifi: Init data frame dynamic rx buffer num: 32
I (488) wifi: Init management frame dynamic rx buffer num: 32
I (494) wifi: Init management short buffer num: 32
I (499) wifi: Init static rx buffer size: 1600
I (503) wifi: Init static rx buffer num: 10
I (507) wifi: Init dynamic rx buffer num: 32
I (512) wifi: Set ps type: 0

[... 96 lines cut ...]
    /**
     * @brief Enable ESP-MESH network

     */
    MDF_ERROR_ASSERT(mwifi_init(&init_config));
    MDF_ERROR_ASSERT(mwifi_set_config(&ap_config));
    MDF_ERROR_ASSERT(mwifi_start());
    } else {
         MDF_ERROR_ASSERT(get_network_config(&init_config, &ap_config));
          MDF_LOGI("mconfig, ssid: %s, password: %s, mesh_id: " MACSTR,
                 ap_config.router_ssid, ap_config.router_password,
                 MAC2STR(ap_config.mesh_id));

        MDF_ERROR_ASSERT(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));

    /**
     * @brief Enable ESP-MESH network

     */
  

        MDF_ERROR_ASSERT(mwifi_init(&init_config));
        MDF_ERROR_ASSERT(mwifi_set_config(&ap_config));
        MDF_ERROR_ASSERT(mwifi_start());
    }`

chenguang3312, 2019-03-31:
这是我的工程文件
-`

[mwifi2.zip](https://github.com/espressif/esp-mdf/files/3026479/mwifi2.zip)

`

chenguang3312, 2019-03-31:
解决了，需要禁用信道扫描才能正常联网， 再问下，调用mconfig_chain_slave_channel_switch_disable（）后，进行联网，联网结束后用不用在把通道扫描开启？

zhanzhaocheng, 2019-04-02:
@chenguang3312 配网结束不需要调用 `mconfig_chain_slave_channel_switch_xxx` 调函数的, 具体可以有参考我之前给您发的工程 或 light , 这个函数是为了同时工作多种配网方式, 如 同时使用 MESH 设备间配网和一键配网, 两者都需要同时操作信道

ESP-LBB, 2019-04-11:
Closing as this question appears to be answered.

closed 2019-04-11 by ESP-LBB