espressif/esp-mdf #42: Esp32 与手机通信

MasterWen, 2019-04-08:
你好，我想问下，刷新获取多个设备状态时（post请求get_status），设备返回的数据是什么格式的，如何处理。一个设备是json格式。多个设备就不是了

fanbaoying, 2019-04-11:
@MasterWen 你好，当有多个设备时，返回的数据中每个设备格式类似是这样的：

[image]

处理方法：
判断response.expectedContentLength值是否为-1，返回值为-1时是多个设备返回数据，对返回的data数据进行处理：

`+ (NSArray<EspHttpResponse *> *)getChunkedResponseListForData:(NSData *)data { // TODO if ([EspCommonUtils isNull:data]) { return nil; } NSMutableArray *result = [NSMutableArray array]; Byte *bytes = (Byte *)[data bytes]; NSMutableData *temp = [NSMutableData data]; long totalLen = data.length; for (long index = 0; index < totalLen; index++) { // Get header data Byte headerBytes[1] = { bytes[index] }; [temp appendBytes:headerBytes length:1]; if (![EspDeviceUtil headEnd:temp]) { continue; } // Get content length NSString *headStr = [[NSString alloc] initWithData:temp encoding:NSUTF8StringEncoding]; NSArray *headArray = [headStr componentsSeparatedByString:@"\r\n"]; long contentLength = -1; for (NSString *s in headArray) { NSArray *kv = [s componentsSeparatedByString:@": "]; @try { NSString *headerName = [kv[0] uppercaseString]; if ([headerName isEqualToString:[EspHttpHeaderContentLength uppercaseString]]) { contentLength = [kv[1] intValue]; break; } } @catch (NSException *e) { NSLog(@"%@", e.name); } } // Get content data for (long contentCount = 0; contentCount < contentLength; contentCount++) { index++; if (index >= totalLen) { break; } Byte contentBytes[1] = {bytes[index]}; [temp appendBytes:contentBytes length:1]; } // Get EspHttpResponse EspHttpResponse *response = [EspHttpUtils getResponseWithFixedLengthData:temp]; if (response != nil) { [result addObject:response]; } // Clear temp data [temp replaceBytesInRange:NSMakeRange(0, [temp length]) withBytes:NULL length:0]; } return result; }`
处理后为数组格式，再进行解析
`NSArray *dataArr = [EspDeviceUtil getChunkedResponseListForData:data];`

ESP-LBB, 2019-06-04:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

closed 2019-06-04 by ESP-LBB