espressif/esp-mdf #43: How to setup mesh to repeat wifi

fluentart, 2019-04-08:
Hi, been searching around and it is a bit unclear. Perhaps this is easy, or not supported/possible. 
I have tested mdf wifi and have nodes communicating, can send status 1 to them and so on. But I can not connect with phone/laptop to nodes wifi, only to the router. 

Is there some setting I am missing to make the wifi access point with same SSID/pass available from mesh nodes some distance away from router?

zhanzhaocheng, 2019-04-09:
The ESP-MESH AP is directly connected to devices that do not support non-ESP-MESH. The mobile phone needs to pass through the router to communicate with the ESP-MESH device.
You can also connect directly to ESP32 using BLE

bbinet, 2019-04-12:
I'm also interested in such a feature.
@zhanzhaocheng could we imagine using the unused softAP interface from the ESP-MESH leaf nodes to allow to repeat router wifi?

fluentart, 2019-04-12:
Having the leaf nodes repeat wifi would be ideal! Even if its just exposing a tcp port to send/recieve messages across the mwifi in json format as in the example. Best case would be NAT, but I imagine that would quite complicated to implement.

dongdong004, 2019-04-15:
So we have discussed this issue with our colleagues.

The ESP-MESH protocol design group will not support this in the current plan.

fluentart, 2019-04-15:
Hi all, just an update from our side. We've resolved the immediate need by connecting an additional ESP32 to the leaf edge nodes. This additional ESP provides a wifi AP for any wifi device as well as a TCP server. These TCP packets are then sent over serial to the other ESP (stacked D1 ESP32 nodes) which then sends the packet through the mesh. Not the most elegant solution, but definitely the fastest way to get this working for our tests.

tekpalb, 2019-05-21:
Hi fluentart,
I am having the same problem as your so can you please share the solution codes or exact technic.

thanks,

fluentart, 2019-05-21:
> Hi fluentart,
> I am having the same problem as yours. so, can you please share the solution codes or exact technic.
> 
> thanks,

Hi tek145. Sure here is the two firmware codes:

https://github.com/IoT-nxt/prototype/tree/dev/examples/esp32/esp32hotspot
https://github.com/IoT-nxt/prototype/tree/dev/examples/esp32/esp32mesh

The esp32mesh is the firmware used to form the mesh ofcourse. 
And esp32hotspot is the firmware loaded on the additional esp32 that communicates over serial to the esp32mesh "co-processor" 
Please note that the hotspot node only opens one TCP port and forwards that over serial, then over the mesh. You can use https://github.com/IoT-nxt/prototype/tree/dev/examples/esp32/esp32mesh/tcpserver to run a TCP server at the root of the mesh to get the TCP packets from the edge hotspot.

I hope that helps!

tek145, 2019-05-21:
Thank you fluentart. this is really helpful.
I have one more doubt that can we connect mobile phone to the leaf node for downward internet(wifi) connection using this technic.
regards.

fluentart, 2019-05-21:
> Thank you fluentart. this is really helpful.
> I have one more doubt that can we connect mobile phone to the leaf node for downward internet(wifi) connection using this technic.
> regards.

You are welcome. I think it is possible, just need to be able to forward all IP/NAT/DNS requests instead of just the one TCP port. But I do not know how to code this now.

vangalvin, 2019-08-10:
@fluentart Thank you for your pointer, I have been looking at options myself as to how to create a mesh that does this and you have just answered that question for me. I am prety new to the esp-idf enviroment so it will be an interesting experance :)

I did try to compile and run esp32mesh however ended up getting an error about not being able to find u8g2. For my project I do not need an LCD screen so may just pick thought the code and remove it.

I am just installing prototype as well as I think that will also be handy in saving the sensor data.

jBernavaPrah, 2022-08-07:
Hi! 
Sorry to pop up this issue from 2019, but is regarding my issue with mesh and AP.

I was looking at implementing the mesh network in one of my project but is mandatory that also exists the AP at the same time. 

Is something that now possible?

Thanks!