espressif/esp-mdf #48: How to use esp_mesh_waive_root not working

Robokishan, 2019-04-16:
playing with this esp_mesh_waive_root and i don't know how its working because the esp32 which is nearest to router getting vote 1.00 but its not changing to root . and the esp32 which is far from router but previously was root stays root.
do we have to switch it manually ? or anything else ?

Robokishan, 2019-04-18:
Should i provide some logs ?

ESP-LBB, 2019-04-18:
Yes please. We've done some test and we succeed with this function. We can have a look with your log
or code of this part? We may help provide an simple example about how to run this.

dongdong004, 2019-04-18:
The node becomes new root only when it receives `ROOT_SWITCH_ACK`.

LOG:
```
I (850275) mesh: [VOTE][ch:6]AP:1, other(ID:0, RD:0), idle:0, mAP:1, root:1, [Layer:2][54:a0:50:e0:6a:e8][VOTING]
I (850276) mesh: [VOTE:15]by root, vote[128][30:ae:a4:80:0f:15,-26], self[30:ae:a4:80:0f:14,-25,reason:1,votes:2,node][mine:2,voter:]

I (850290) mesh: <MESH_NEW_ROOT_VOTE_DONE>new voted root, s_vote_scan_times:15

MESH_EVENT_VOTE_STOPPED
I (850300) mesh: [scan]new scanning time:600ms
W (850303) wifi: [beacon]new interval:300ms
I (850309) mesh: 1888<arm>parent monitor, my layer:2(cap:25), interval:10897ms, retries:1<>
I (850316) mesh: 2063<layer:2>xmit MESH_PKT_ROOT_SWITCH_REQ to 30:ae:a4:80:48:41
I (850325) mesh: <MESH_NEW_ROOT_SWITCH_ACK>triggered by received acknowledgment
I (850330) mesh: [WAIVE]new voted, connect to router:lcj-2G, 54:a0:50:e0:6a:e8
I (850337) wifi: state: run -> init (0)
I (850340) wifi: pm stop, total sleep time: 0 us / 16190515 us

I (850346) wifi: new:<6,0>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:6

MESH_EVENT_ROOT_SWITCH_ACK

I (850404) mesh: [wifi]disconnected reason:8(assoc leave), continuous:1/max:12, root, vote(,stopped)<><>
I (850530) wifi: new:<6,1>, old:<6,0>, ap:<6,1>, sta:<6,0>, prof:6
I (850531) wifi: state: init -> auth (b0)
I (851106) wifi: state: auth -> assoc (0)
I (851186) wifi: state: assoc -> run (10)
I (851187) wifi: connected with lcj-2G, channel 6, bssid = 54:a0:50:e0:6a:e8
I (851192) wifi: pm start, type: 0

E (851195) event: invalid static ip 
```

Robokishan, 2020-06-30:
No i drop the idea about esp32 in mesh