espressif/esp-mdf #49: 根节点能否不连接路由器wifi，而是使用eth与外网通信

MGDG, 2019-04-23:
你好，在实际应用中现场只提供eth的接入方式没有wifi，我在尝试让根节点通过eth连接外网而不是wifi，实际测试发现使用eth后虽然可以和外网通信但是子节点无法连接上根节点，根节点一直产生MDF_EVENT_MWIFI_PARENT_DISCONNECTED事件，这应该怎么解决？谢谢。

zhanzhaocheng, 2019-04-24:
在这种场景中, 将连接 eth 的设备配置成`MESH_ROOT`的类型, 其他设备配置成 `MESH_NODE` 类型即可, 参见 [get_started](https://github.com/espressif/esp-mdf/tree/master/examples/get-started) 示例 如果还是无法连接, 请提供日志文件
```c
    mwifi_config_t config   = {
        .mesh_type = MESH_ROOT,
    };
```

MGDG, 2019-04-25:
抱歉是我疏忽了，在头文件的结构体描述里看到这句话，我之前连接了eth但是也配置了router_ssid等信息。

`/**
     * @brief  Information about the router connected to the root.
     *         This item does not need to be set in the routerless scheme.
     */`

closed 2019-04-25 by MGDG