espressif/esp-mdf #50: A panic occurs on the route device side

nopnop2002, 2019-04-24:
I tried esp-mdf/examples/get-started.

A panic occurs on the route device side.

Wha worng??

	I (29) boot: ESP-IDF v3.3-dev-264-g8a69ffc36 2nd stage bootloader
	I (30) boot: compile time 13:15:10
	I (30) boot: Enabling RNG early entropy source...
	I (36) boot: SPI Speed      : 40MHz
	I (40) boot: SPI Mode       : DIO
	I (44) boot: SPI Flash Size : 4MB
	I (48) boot: Partition Table:
	I (51) boot: ## Label            Usage          Type ST Offset   Length
	I (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000
	I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
	I (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000
	I (81) boot:  3 factory          factory app      00 00 00010000 00140000
	I (88) boot:  4 ota_0            OTA app          00 10 00150000 00140000
	I (96) boot:  5 ota_1            OTA app          00 11 00290000 00140000
	I (103) boot:  6 coredump         Unknown data     01 03 003d0000 00010000
	I (111) boot:  7 reserved         Unknown data     01 fe 003e0000 00020000
	I (119) boot: End of partition table
	I (123) boot: Defaulting to factory image
	I (127) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f3a8 (127912) map
	I (181) esp_image: segment 1: paddr=0x0002f3d0 vaddr=0x3ffb0000 size=0x00c40 (  3136) load
	I (183) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x91a84 (596612) map
	0x400d0018: _stext at ??:?

	I (397) esp_image: segment 3: paddr=0x000c1aa4 vaddr=0x3ffb0c40 size=0x01d60 (  7520) load
	I (400) esp_image: segment 4: paddr=0x000c380c vaddr=0x40080000 size=0x00400 (  1024) load
	0x40080000: _WindowOverflow4 at /home/robotics/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1779

	I (404) esp_image: segment 5: paddr=0x000c3c14 vaddr=0x40080400 size=0x10ce0 ( 68832) load
	I (451) boot: Loaded app from partition at offset 0x10000
	I (451) boot: Disabling RNG early entropy source...
	I (452) cpu_start: Pro cpu up.
	I (455) cpu_start: Single core mode
	I (460) heap_init: Initializing. RAM available for dynamic allocation:
	I (467) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (473) heap_init: At 3FFB9528 len 00026AD8 (154 KiB): DRAM
	I (479) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
	I (485) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
	I (491) heap_init: At 400910E0 len 0000EF20 (59 KiB): IRAM
	I (498) cpu_start: Pro cpu start user code
	I (180) cpu_start: Starting scheduler on PRO CPU.
	I (206) wifi: wifi driver task: 3ffc17d8, prio:23, stack:3584, core=0
	I (206) wifi: wifi firmware version: 0bff4fb
	I (207) wifi: config NVS flash: enabled
	I (209) wifi: config nano formating: disabled
	I (213) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (222) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (239) wifi: Init dynamic tx buffer num: 32
	I (240) wifi: Init data frame dynamic rx buffer num: 32
	I (241) wifi: Init management frame dynamic rx buffer num: 32
	I (247) wifi: Init static rx buffer size: 1600
	I (251) wifi: Init static rx buffer num: 10
	I (254) wifi: Init dynamic rx buffer num: 32
	I (259) wifi: Set ps type: 0

	I (321) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 0
	I (322) wifi: mode : sta (30:ae:a4:ca:e1:40)
	I (324) [mwifi, 138]: esp-mdf version: 268f292-dirty
	W (326) wifi: [beacon]interval:100ms
	I (330) wifi: mode : sta (30:ae:a4:ca:e1:40) + softAP (30:ae:a4:ca:e1:41)
	I (338) wifi: Init max length of beacon: 752/752
	I (341) wifi: Init max length of beacon: 752/752
	I (348) mesh: [IO]disable self-organizing<reconnect>
	W (350) wifi: [beacon]interval:100ms
	I (353) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
	I (1356) wifi: Set ps type: 0

	I (1361) mesh: [MANUAL]designated as root and router is not set
	I (1361) [mwifi, 98]: MESH is started
	I (1362) [get_started, 199]: event_loop_cb, event: 0
	I (1366) [get_started, 203]: MESH is started
	I (1372) [get_started, 40]: Root is running
	I (2562) wifi: n:13 2, o:13 0, ap:13 2, sta:0 0, prof:13
	I (2563) wifi: station: 3c:71:bf:94:66:9c join, AID=1, bgn, 40D
	I (2566) [mwifi, 116]: Routing table is changed by adding newly joined children add_num: 1, total_num: 2
	I (2572) [get_started, 199]: event_loop_cb, event: 5
	I (2577) [get_started, 199]: event_loop_cb, event: 3
	I (3059) mesh: 1009[xrsp:1]the asked:7, max window:4, force to increase/decrease(up) xseqno:1 for child 3c:71:bf:94:66:9c, xrsp_seqno:4, heap:177388
	I (3069) mesh: 1163[recv]cidx[0]3c:71:bf:94:66:9c, xseqno duplicate check failure, new xseqno:7, in:1, out:1, pending:0
	I (3072) mesh: 1172[recv]cidx[0]3c:71:bf:94:66:9c xseqno loss, current/new:0/7, in:1, out:1, pending:0
	I (3082) [get_started, 52]: Root receive, addr: 3c:71:bf:94:66:9c, size: 15, data: (6) Hello root!
	Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
	Core 0 register dump:
	PC      : 0x400d5635  PS      : 0x00060730  A0      : 0x800d2eec  A1      : 0x3ffd46f0
	0x400d5635: mwifi_root_write at /home/robotics/esp-mdf/components/mwifi/mwifi.c:917 (discriminator 1)

	A2      : 0x00000000  A3      : 0x00000004  A4      : 0x00000000  A5      : 0x00000000
	A6      : 0x0000000f  A7      : 0x00000000  A8      : 0x800d562f  A9      : 0x3ffd46d0
	A10     : 0x00000c13  A11     : 0x000000bf  A12     : 0x00000005  A13     : 0x3ffd472c
	A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
	EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

	Backtrace: 0x400d5635:0x3ffd46f0 0x400d2ee9:0x3ffd4780 0x4008b719:0x3ffd47d0
	0x400d5635: mwifi_root_write at /home/robotics/esp-mdf/components/mwifi/mwifi.c:917 (discriminator 1)

	0x400d2ee9: root_task at /home/robotics/esp-mdf/examples/get-started/main/get_started.c:55 (discriminator 1)

	0x4008b719: vPortTaskWrapper at /home/robotics/esp-mdf/esp-idf/components/freertos/port.c:403

	Rebooting...
	ets Jun  8 2016 00:22:57

will-emmerson, 2019-04-24:
Looks like you haven't setup the router settings, could be that?:

```
I (353) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
```

zhanzhaocheng, 2019-04-24:
This [issue](https://github.com/espressif/esp-mdf/commit/67a0a5bd72ed284f7a5bf50694f1ab942319dd6b) has been fixed, please update ESP-MDF

nopnop2002, 2019-04-24:
Thank you quick responce.
I updated esp-mdf & it fixed.
Please close this issues.

	I (1370) [mwifi, 98]: MESH is started
	I (1370) [get_started, 199]: event_loop_cb, event: 0
	I (1374) [get_started, 203]: MESH is started
	I (1379) [get_started, 40]: Root is running
	I (5813) wifi: n:13 2, o:13 0, ap:13 2, sta:0 0, prof:13
	I (5815) wifi: station: 3c:71:bf:94:66:9c join, AID=1, bgn, 40D
	I (5817) [mwifi, 116]: Routing table is changed by adding newly joined children add_num: 1, total_num: 2
	I (5823) [get_started, 199]: event_loop_cb, event: 5
	I (5829) [get_started, 199]: event_loop_cb, event: 3
	I (6033) mesh: 1009[xrsp:1]the asked:140, max window:4, force to increase/decrease(up) xseqno:1 for child 3c:71:bf:94:66:9c, xrsp_seqno:137, heap:177524
	I (6041) mesh: 1163[recv]cidx[0]3c:71:bf:94:66:9c, xseqno duplicate check failure, new xseqno:140, in:1, out:1, pending:0
	I (6046) mesh: 1172[recv]cidx[0]3c:71:bf:94:66:9c xseqno loss, current/new:0/140, in:1, out:1, pending:0
	I (6057) [get_started, 52]: Root receive, addr: 3c:71:bf:94:66:9c, size: 17, data: (139) Hello root!
	I (6071) [get_started, 57]: Root send, addr: 3c:71:bf:94:66:9c, size: 15, data: (0) Hello node!
	I (7050) [get_started, 52]: Root receive, addr: 3c:71:bf:94:66:9c, size: 17, data: (140) Hello root!
	I (7054) [get_started, 57]: Root send, addr: 3c:71:bf:94:66:9c, size: 15, data: (1) Hello node!
	I (8053) [get_started, 52]: Root receive, addr: 3c:71:bf:94:66:9c, size: 17, data: (141) Hello root!
	I (8057) [get_started, 57]: Root send, addr: 3c:71:bf:94:66:9c, size: 15, data: (2) Hello node!
	I (9057) [get_started, 52]: Root receive, addr: 3c:71:bf:94:66:9c, size: 17, data: (142) Hello root!

zhanzhaocheng, 2019-04-24:
1. Modify the maximum connection of each layer by `make menuconfig` to form `root + 2 intermediate paraent Node + 1 Leaf Node`
[image]

2. device type:
                                 MESH_IDLE: The node type is selected by MESH according to the network conditions.
                                 MESH_ROOT: The Device is the root node
                                 MESH_NODE: The device is a non-root node and contains intermediate nodes and leaf nodes.
                                 MESH_LEAF: The device is a leaf node, closes the softap, and is used in a low-power solutions

nopnop2002, 2019-04-25:
Thank you for your advice.
Please close this issues.

closed 2019-04-25 by zhanzhaocheng