espressif/esp-mdf #51: 关于时钟同步问题

Deryckyin, 2019-04-26:
我希望把ESP32的MESH网络用在一种多个设备的控制系统之中，因此需要网络里每个节点能协同动作。
显然，由于MESH网络个节点直接通信延迟的不确定性，仅仅通过MESH网络进行直接动作命令的传递，实际上无法实现各个节点的精确动作协同。
因此，我们想了解，ESP32的MESH框架中，是否存在一种高精度的同步时钟技术。使得我们可以在命令下发到各个节点之后，能指定动作执行的时间，这样也就间接实现多设备之间的动作协同。

zhanzhaocheng, 2019-04-28:
目前 esp-mdf 暂不支持此功能. 请提供您的更详细的需求:
1. 时间的精度要求是多少
2. 网络规模是多大

Deryckyin, 2019-04-28:
> 目前 esp-mdf 暂不支持此功能. 请提供您的更详细的需求:
> 
> 1. 时间的精度要求是多少
> 2. 网络规模是多大

精度要求：一个MESH网内各节点之间的时间误差不超过5ms。
网络规模：不超过三层50个节点。

其实这个要求并不算很高。由于mesh通信延迟的存在，对于需要多个节点协同控制或者同步数据采集的的应用来说，精准的时间同步是个必要的功能，因此，希望ESP-MDF能够集成时间同步模块。

Deryckyin, 2019-04-28:
https://www.analog.com/cn/education/education-library/videos/5579266065001.html
这是别的MESH产品中的时间同步功能示例，可用于实现精准的时间戳和控制。ESP32的mesh能不能提供个类似功能啊。

zhanzhaocheng, 2019-04-29:
这个需求是可以满足, 之后将会增加 ESP-MDF 中, 待我们评估完具体的方案及时间安排后, 会将信息同步给你

Deryckyin, 2019-04-29:
> 这个需求是可以满足, 之后将会增加 ESP-MDF 中, 待我们评估完具体的方案及时间安排后, 会将信息同步给你

谢谢。期待ESP-MDF的完善和发展。

zhanzhaocheng, 2019-07-17:
时间同步功能已经支持, 通过 tsf 进行同步的可以达到 ms 级, 请使用 [esp_mesh_get_tsf_time()](https://github.com/espressif/esp-idf/blob/4a9f339447cd5b3143f90c2422d8e1e1da9da0a4/components/esp32/include/esp_mesh.h#L1487)