espressif/esp-mdf #52: mesh_assoc_t define

nopnop2002, 2019-05-13:
In print_system_info_timercb(), It get mesh_assoc using esp_wifi_vnd_mesh_get().
But there is no mesh_assoc_t define.
Where is mesh_assoc_t define?

	static void print_system_info_timercb(void *timer)
	{
	    uint8_t primary                 = 0;
	    wifi_second_chan_t second       = 0;
	    mesh_addr_t parent_bssid        = {0};
	    uint8_t sta_mac[MWIFI_ADDR_LEN] = {0};
	    mesh_assoc_t mesh_assoc         = {0x0};
	    wifi_sta_list_t wifi_sta_list   = {0x0};

	    esp_wifi_get_mac(ESP_IF_WIFI_STA, sta_mac);
	    esp_wifi_ap_get_sta_list(&wifi_sta_list);
	    esp_wifi_get_channel(&primary, &second);
	    esp_wifi_vnd_mesh_get(&mesh_assoc);
	    esp_mesh_get_parent_bssid(&parent_bssid);

	    MDF_LOGI("System information, channel: %d, layer: %d, self mac: " MACSTR ", parent bssid: " MACSTR
	             ", parent rssi: %d, node num: %d, free heap: %u", primary,
	             esp_mesh_get_layer(), MAC2STR(sta_mac), MAC2STR(parent_bssid.addr),
	             mesh_assoc.rssi, esp_mesh_get_total_node_num(), esp_get_free_heap_size());

nopnop2002, 2019-05-13:
i got mesh_assoc_t.
components\esp32\include\esp_mesh_internal.h

Please close this issues.

ESP-LBB, 2019-05-13:
Closing as this question appears to be answered.

closed 2019-05-13 by ESP-LBB