espressif/esp-mdf #53: esp32 wifi data frame unreadable by phone

tekpalb, 2019-05-20:
hi all,

I have to develop an esp32 wifi mesh network,but using esp32's sta+ap mode both.phone is unable to read/decode the original data or unable to connect to internet,please do give any solution.
note-esp32 wifi mesh network(in sta+ap mode)

zhanzhaocheng, 2019-05-20:
The ESP-MESH AP does not allow the mobile phone to be directly connected. The mobile phone can control the ESP-MESH device in the following two ways.
1. Connect the ESP-MESH to a router, and the data on the phone is transferred to the device through the router.
2. The mobile phone directly connects to the ESP-MESH device via Bluetooth.

tekpalb, 2019-05-21:
Thank you zhanzhaocheng for the answer,
but what i actually need is connect a mobile phone at the end of the leaf node in wifi esp32 mesh (let's say leaf node is at some distance from the router) for internet access(leaf node acting as a wifi repeter).Any solution is apprciated.

regards'

ESP-LBB, 2019-05-22:
I'm afraid this is not feasible. In ESP-MESH mode, though it works on STA+AP mode, but the AP mode is accessible only for a ESP-MESH device, that means the smartphone can't connect to the node directly via wifi to internet.
ESP-MESH was designed for IoT application.

tek145, 2019-05-22:
Thank you ESP-LBB.
can we connect esp32 wifi  mesh network to esp8266 then to phone.

ESP-LBB, 2019-05-22:
esp32 mesh network -> (with or without) esp8266 -> Phone -> Internet : Yes
Phone -> (with or without) esp8266 -> esp32 mesh network -> Internet : No
The phone can connect to the mesh network via bluetooth and get the info of the mesh network or control the devices in the network, but can't get to the Internet.

tek145, 2019-05-23:
Thank you 
this answer my question. so i am closing the issue

closed 2019-05-23 by tek145