espressif/esp-mdf #55: parent bssid is not match mac address of root node.

nopnop2002, 2019-05-30:
I build 2 node ESP-MESH.
One node is Root node, and other node is Leaf node.

I used esp-mdf-master\examples\function_demo\mwifi.
I modified print_system_info_timercb() as follow.

	static void print_system_info_timercb(void *timer)
	{
	    uint8_t primary                 = 0;
	    wifi_second_chan_t second       = 0;
	    mesh_addr_t parent_bssid        = {0};
	    uint8_t sta_mac[MWIFI_ADDR_LEN] = {0};
	    mesh_assoc_t mesh_assoc         = {0x0};
	    wifi_sta_list_t wifi_sta_list   = {0x0};

	    esp_wifi_get_mac(ESP_IF_WIFI_STA, sta_mac);
	    esp_wifi_ap_get_sta_list(&wifi_sta_list);
	    esp_wifi_get_channel(&primary, &second);
	    esp_wifi_vnd_mesh_get(&mesh_assoc);
	    esp_mesh_get_parent_bssid(&parent_bssid);

	    MDF_LOGI("System information, channel: %d, layer: %d, self mac: " MACSTR ", parent bssid: " MACSTR
	             ", parent rssi: %d, router rssi : %d, node num: %d, free heap: %u", primary,
	             esp_mesh_get_layer(), MAC2STR(sta_mac), MAC2STR(parent_bssid.addr),
	             mesh_assoc.rssi, mesh_assoc.router_rssi, esp_mesh_get_total_node_num(), esp_get_free_heap_size());

	    for (int i = 0; i < wifi_sta_list.num; i++) {
	        MDF_LOGI("Child mac: " MACSTR, MAC2STR(wifi_sta_list.sta[i].mac));
	    }

	#ifdef MEMORY_DEBUG
	    if (!heap_caps_check_integrity_all(true)) {
	        MDF_LOGE("At least one heap is corrupt");
	    }

	    mdf_mem_print_heap();
	    mdf_mem_print_record();
	#endif /**< MEMORY_DEBUG */
	}

System information of Root Node:

	I (21452) [ROOT, 378]: System information, channel: 1, layer: 1, self mac: a4:cf:12:44:62:98, parent bssid: 24:0a:c4:27:69:c9, parent rssi: -17, router rssi : -16, node num: 3, free heap: 156980
	I (21460) [ROOT, 381]: Child mac: a4:cf:12:44:47:f8
	I (21465) [ROOT, 381]: Child mac: a4:cf:12:44:4f:00

self mac address is a4:cf:12:44:62:98

System information of Leaf Node:

	I (141488) [ROOT, 378]: System information, channel: 1, layer: 2, self mac: a4:cf:12:44:4f:00, parent bssid: a4:cf:12:44:62:99, parent rssi: -26, router rssi : -50, node num: 3, free heap: 150696

parent bssid is a4:cf:12:44:62:99

parent bssid is not match mac address of root node.

zhanzhaocheng, 2019-05-30:
Parent bssid refers to the address of the parent node's `AP`.
Parent mac refers to the address of the parent node's `sta`

> NOTE: 

Each ESP32 chip has MAC addresses for Station (STA), access point (AP), Bluetooth low energy (BLE) and local area network (LAN).

Their address values are incremented by one, i.e. LAN Mac = BLE Mac + 1 = AP Mac + 2 = STA Mac + 3.

For example: 

- MAC for STA: xx:xx:xx:xx:xx:00 
- MAC for AP: xx:xx:xx:xx:xx:01 
- MAC for BLE: xx:xx:xx:xx:xx:02 
- MAC for LAN: xx:xx:xx:xx:xx:03

The device’s STA address is used For ESP-MESH communication.

nopnop2002, 2019-05-30:
Thank you.
I could acquire own MAC address of the AP using this.

esp_wifi_get_mac(ESP_IF_WIFI_AP, ap_mac);

Please close this issues.

closed 2019-05-30 by ESP-LBB