espressif/esp-mdf #56: Release Plan for MDF Framework

xpratpa, 2019-05-31:
Is there any Release Plan?

ESP-LBB, 2019-06-03:
Yes, we plan to release 1.0 by the end of June.

Ks89, 2019-06-12:
Hi!
Is there a way to import esp-mdf into either Arduino IDE or PlatformIO?

At the moment I added "https://dl.espressif.com/dl/package_esp32_dev_index.json" to the board manager URLs. I can build code with for instance "#include "esp_wifi.h"", but not "#include "mwifi.h"".

Do you have any suggestions?

thank u

busyluo, 2019-07-03:
It's July. How long will it take？

ESP-LBB, 2019-07-17:
Sorry for the delay, but we finally release the beta version. Please have test and feedback if any issue found, thanks.

bonastreyair, 2019-08-15:
What about integrating IDFv4.0? When should we expect to be integrated? Is it going to be on the MDF v1.0?
Thanks,

ESP-LBB, 2019-08-15:
No, we will release v1.0 before integrating IDF4.0. 
IDF4.0 has made many modifications and improvements so we also look forward updating to it, but it's not a stable version yet, so we may start the task after it releases, and then we can estimate how long will it take.