espressif/esp-mdf #57: Can I make node where it can't become root-node

nopnop2002, 2019-06-05:
Hello.

Can I make node where it can't become root-node irrespective of order of esp32 charging of a power supply?

(Example)
node1 is node where it can become never root-node.
node2 is node where it can become root-node.
I turn on the power of node1 first.
I turn on the power of node2 from later.
Building of Mesh - Wifi starts.
node2 will be root-node and node1 will be leaf-node.

ESP-LBB, 2019-06-05:
Yes, there are 4 types of node that you can define in your code (or menuconfig)
1. root (can only be root)
2. non-root (can only be non-root)
3. leaf (can only be leaf node)
4. idle (can become any type of node)

nopnop2002, 2019-06-06:
Thank you for answer.
I'm in holiday.
I will back to work next Wednesday.
I will try it when i back in work.

nopnop2002, 2019-06-11:
I was back to work.
I searched in my sdkconfig about mesh type.
But i can't find item about mesh type.
please tell me key-word about mesh type in sdkconfig.

ESP-LBB, 2019-06-11:
mwifi_config_t

nopnop2002, 2019-06-11:
Thank you.
I got this code.

    /**
     * @breif Start mesh
     */
    mwifi_init_config_t cfg   = MWIFI_INIT_CONFIG_DEFAULT();
    mwifi_config_t config     = {
        .router_ssid     = CONFIG_ROUTER_SSID,
        .router_password = CONFIG_ROUTER_PASSWORD,
        .mesh_id         = CONFIG_MESH_ID,
        .mesh_password   = CONFIG_MESH_PASSWORD,
        .mesh_type       = MESH_LEAF,
    };

    /**
     * @brief Set the log level for serial port printing.
     */
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    /**
     * @brief Initialize wifi mesh.
     */
    MDF_ERROR_ASSERT(mdf_event_loop_init(event_loop_cb));
    MDF_ERROR_ASSERT(wifi_init());
    MDF_ERROR_ASSERT(mwifi_init(&cfg));
    MDF_ERROR_ASSERT(mwifi_set_config(&config));
    MDF_ERROR_ASSERT(mwifi_start());

I have one question.
I use following timer call back function.
I have 2 of ESP32.
One is ROOT-NODE, and other is LEAF-NODE.

I turned on the power of LEAF-NODE ESP32.
print_system_info_timercb indicated [parent bssid] as all 0.
Then I turned on the power of ROOT-NODE ESP32.
print_system_info_timercb indicated correct [parent bssid].

Then i turned off the power of ROOT-NODE.
There is NO ROOT-NODE, but print_system_info_timercb keeps indicating [parent bssid] just as it is.
[parent bssid] don't return all 0.

	static void print_system_info_timercb(void *timer)
	{
	    uint8_t primary                 = 0;
	    wifi_second_chan_t second       = 0;
	    mesh_addr_t parent_bssid        = {0};
	    uint8_t sta_mac[MWIFI_ADDR_LEN] = {0};
	    mesh_assoc_t mesh_assoc         = {0x0};
	    wifi_sta_list_t wifi_sta_list   = {0x0};
	    static int counter = 0;

	    esp_wifi_get_mac(ESP_IF_WIFI_STA, sta_mac);
	    esp_wifi_ap_get_sta_list(&wifi_sta_list);
	    esp_wifi_get_channel(&primary, &second);
	    esp_wifi_vnd_mesh_get(&mesh_assoc);
	    esp_mesh_get_parent_bssid(&parent_bssid);

	    MDF_LOGI("System information, channel: %d, layer: %d, self mac: " MACSTR ", parent bssid: " MACSTR
	             ", parent rssi: %d, node num: %d, free heap: %u", primary,
	             esp_mesh_get_layer(), MAC2STR(sta_mac), MAC2STR(parent_bssid.addr),
	             mesh_assoc.rssi, esp_mesh_get_total_node_num(), esp_get_free_heap_size());

	    for (int i = 0; i < wifi_sta_list.num; i++) {
	        MDF_LOGI("Child mac: " MACSTR, MAC2STR(wifi_sta_list.sta[i].mac));
	    }

	#ifdef MEMORY_DEBUG
	    if (!heap_caps_check_integrity_all(true)) {
	        MDF_LOGE("At least one heap is corrupt");
	    }

	    mdf_mem_print_heap();
	    mdf_mem_print_record();
	#endif /**< MEMORY_DEBUG */
	}

ESP-LBB, 2019-06-11:
yeah, "parent bssid" just means the node choose(d) this one as its parent node, it won't fall back to all 0 if its parent node is power off, it won't change until it connects to another node.

also for "node num" in case you get confused, this info is included in the beacon from the root, it's the number of all nodes in a mesh network, so after you turn off the root, there is no more beacon from the root, and it keeps at 2.

nopnop2002, 2019-06-12:
ok
I understood.
Please close this issues.
Thank you for your support.

closed 2019-06-12 by ESP-LBB