espressif/esp-mdf #60: ota update of different device type and firmware in same mesh

Kasuhikdas, 2019-06-16:
- Development Kit:      [ESP32-Wrover-Kit]
- Core (if using chip or module): ESP-WROOM32
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB|external 5V]

I have multiple esp-32 in mesh network. There are different types of firmware across the devices. 
In mupgrade it updated all the devices with same firmware as written in document. Is there any way to update few devices in mesh with a particular ota and others with different ota??

zhanzhaocheng, 2019-06-17:
Different types of devices need to be upgraded separately, and each time only the same type of device can be upgraded.
For example, there are two types of devices in the network: Type 1 (A, B), Type (C, D), you call `mupgrade_firmware_send`, just pass in the address of (A, B), wait until A, B is upgraded and then give C, D upgrade

Kasuhikdas, 2019-06-17:
way to set one address is like below
uint8_t dest_mac_addr[] = {0x30,0xae,0xa4,0x04,0xc0,0xf4};

but there is provision of sending multiple addresses.
mupgrade_firmware_send(const uint8_t *dest_addrs, size_t dest_addrs_num, mupgrade_result_t *result)

how to send multiple addresses?

zhanzhaocheng, 2019-06-18:
uint8_t dest_mac_addr[] = {0x30,0xae,0xa4,0x04,0xc0,0xf4, 0x30,0xae,0xa4,0x04,0xc0,0xf3, 0x30,0xae,0xa4,0x04,0xc0,0xf2};

Kasuhikdas, 2019-06-19:
thanks,that worked. 
What about firmware version? I dont want to push same firmware version like the ota example in esp-idf. But mupgrade same firmware is getting updated. Any solution for that?

zhanzhaocheng, 2019-06-24:
Mupgrade does not verify the firmware version of the device
You need to add a version identifier at the application layer and compare it with the server version before the upgrade.

Kasuhikdas, 2019-06-24:
I understand that. But i have different fimwares. So,if root is different type of device then can not check roots firmware version for the upgrade and if we do the checking in node and stopes the update in node then upgrade process in root will keep sending packets and fail eventually.

zhanzhaocheng, 2019-06-24:
Mupgrade temporarily does not support the child node to actively stop the upgrade, I will add this function as soon as possible, currently using the following two ways:
1. The root node first asks for the type and version number of all devices, and then the root node sends the package to the device that needs to be upgraded.
2. After the child node is powered on, it sends the device type and version number to the server actively, and the server upgrades the device in batches.

zhanzhaocheng, 2019-06-28:
1. You can enable firmware type checking by `make menuconfig`. You can add type identifiers to different types of device firmware. After the upgrade is completed, this flag will be checked. Only the identifier will be used for version switching.
[image]

2. You can also determine the firmware name (ie version) when you receive the MDF_EVENT_MUPGRADE_STARTED event, if the firmware of the device is not called `mupgrade_stop()`

rkashi, 2019-07-23:
**MUPGRADE**

I have made some of the log strings bold to highlight them and my questions are around  it. Please help

> **0x40080000: _WindowOverflow4 at /home/ravi/wa/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1779**
 Is it a problem it? Is so, how to fix this?

> **0x400844c0: set_tx_gain_table at ??:?** 
What does this indicate?

> **D (128) [mupgrade_check, 142]: Add an identifier to the firmware: LIGHT_RGB**
It is showing the string I set during "make menuconfig" 

After the binary is downloaded, I see these printed out
> **I (492656) esp_image: segment 0: paddr=0x001f0020 vaddr=0x3f400020 size=0x48850 (297040) map**
> **I (492896) esp_image: segment 1: paddr=0x00238878 vaddr=0x3ffbdb60 size=0x03640 ( 13888) **
> **I (492906) esp_image: segment 2: paddr=0x0023bec0 vaddr=0x40080000 size=0x00400 (  1024) **
> **0x40080000: _WindowOverflow4 at /home/ravi/wa/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1779**
What does this mean - WindowOverflow4. 

> **I (493886) [ota_task, 72]: Connection closed, Firmware download completed**
Connection closed after download.

> **Recieving UpdateD (496916) [mupgrade_node, 135]: Response mupgrade status, written_size: 1259376, response_size: 557**
Why is it writing less than the received bytes?

> **D (493886) [mupgrade_root, 394]: packet_num: 1537, total_size: 1573744**
Size of the firmware reported is matching the file size on disk.

> **D (496926) [mupgrade_root, 264]: Response, src_addr: 24:6f:28:f6:fa:d8, request_num: 1, total_size: 1573744, written_size: 1259376, error_code: MDF_OK**
Why is the error_code MDF_OK, when its writing less than the received file size?

> **D (496946) [mupgrade_root, 317]: MDF_ERR_MUPGRADE_FIRMWARE_INCOMPLETE**
> **D (496956) [mupgrade_root, 405]: Mupgrade_firmware_send unfinished_num: 1**
> D (496966) [mupgrade_root, 408]: Count: 0, addr: 24:6f:28:f6:fa:d8
> D (496966) [mupgrade_root, 456]: seq: 678, size: 1024, addrs_num: 1
> abort() was called at PC 0x4015dba9 on core 1

This error is understood - MDF_ERR_MUPGRADE_FIRMWARE_INCOMPLETE, but Why is not clear. 

###########################################################################

--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6188
load:0x40078000,len:10272
load:0x40080400,len:6608
entry 0x40080764
I (29) boot: ESP-IDF v3.2.2-42-g4a9f33944 2nd stage bootloader
I (29) boot: compile time 13:43:05
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 8MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (88) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (95) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (103) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
[... 96 lines cut ...]
**D (496946) [mupgrade_root, 317]: MDF_ERR_MUPGRADE_FIRMWARE_INCOMPLETE**
**D (496956) [mupgrade_root, 405]: Mupgrade_firmware_send unfinished_num: 1**
D (496966) [mupgrade_root, 408]: Count: 0, addr: 24:6f:28:f6:fa:d8
D (496966) [mupgrade_root, 456]: seq: 678, size: 1024, addrs_num: 1
abort() was called at PC 0x4015dba9 on core 1
0x4015dba9: is_safe_write_address at /home/ravi/wa/esp-mdf/esp-idf/components/spi_flash/flash_ops.c:126

Backtrace: 0x40090200:0x3ffea700 0x4009042d:0x3ffea720 0x4015dba9:0x3ffea740 0x4008687e:0x3ffea760 0x4015da09:0x3ffea7b0 0x4013f578:0x3ffea7d0 0x4013f8a4:0x3ffea810 0x40131d90:0x3ffea840 0x4008d275:0x3ffea8c0
0x40090200: invoke_abort at /home/ravi/wa/esp-mdf/esp-idf/components/esp32/panic.c:707

0x4009042d: abort at /home/ravi/wa/esp-mdf/esp-idf/components/esp32/panic.c:707

0x4015dba9: is_safe_write_address at /home/ravi/wa/esp-mdf/esp-idf/components/spi_flash/flash_ops.c:126

0x4008687e: spi_flash_write at /home/ravi/wa/esp-mdf/esp-idf/components/spi_flash/flash_ops.c:336 (discriminator 2)

0x4015da09: esp_partition_write at /home/ravi/wa/esp-mdf/esp-idf/components/spi_flash/partition.c:274

0x4013f578: mupgrade_write at /home/ravi/wa/esp-mdf/components/mupgrade/mupgrade_node.c:190

0x4013f8a4: mupgrade_handle at /home/ravi/wa/esp-mdf/components/mupgrade/mupgrade_node.c:265

0x40131d90: node_read_task at /home/ravi/wa/projects/esp32-fw/main/nodemtsk.c:301

0x4008d275: vPortTaskWrapper at /home/ravi/wa/esp-mdf/esp-idf/components/freertos/port.c:403

Kasuhikdas, 2019-07-23:
In case root is not the device I want to upgrade then the ota is not sent to other nodes.
Root type ABC
Node type CDE
Update for CDE is not getting sent.

Kasuhikdas, 2019-07-23:
How do I find the version of incoming ota binary. I have device configuration in sdkconfig

rkashi, 2019-07-24:
@zhanzhaocheng 
As per my understanding, to allow selective upgrade of a particular device node recommended by you in the post below, device type check should be done in the **node device**.  Currently, the "device type" check is made in the **root node** as soon as the binary is downloaded from the cloud. This causes the upgrade task to terminate with error code **`"MDF_ERR_MUPGRADE_FIRMWARE_INVALID"`** in case the root is not of the set device type.

What is your recommendation on this proposed implementation?
https://github.com/espressif/esp-mdf/issues/60#issuecomment-506695476

zhanzhaocheng, 2019-07-24:
@Kasuhikdas @rkashi 

The current version is root will check CONFIG_MUPGRADE_FIRMWARE_FLAG, you can delete the root check, we will modify it in the next version. This feature is to prevent the firmware generated by this project from being upgraded to the device. For example, hello_world.bin

[image]

It is recommended to use the ota name to distinguish between device type and version checking. For example, the name can include the version number and device type. Determined in MDF_EVENT_MUPGRADE_STARTED if mupgrade_stop() is called for non-self version

[image]

rkashi, 2019-07-30:
@zhanzhaocheng I have network with more than 2 layers, meaning root -> Layer_1 nodes -> Layer_2 nodes.
How does mupgrade handle this situation?