espressif/esp-mdf #61: mqtt_client.h not found when using cmake (but works with make)

will-emmerson, 2019-06-27:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP32-Wrover-I|ESP32-Wrover|ESP32-PICO-D4|ESP32-SOLO-I|ESP32-WROOM-32U|ESP-WROOM-32D|ESP-WROOM32|ESP32]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 215e332
- Development Env:		[CLion]
- Operating System:		[Fedora]
- Power Supply:         [USB]

## Problem Description
I have a project which was compiling a few months ago. I updated mdf today and tried to compile it but got the following error:
`fatal error: mqtt_client.h: No such file or directory`

It is also showing in my IDE as not existing, and wants to include it using `#include "../../../esp/esp-mdf/esp-idf/components/mqtt/esp-mqtt/include/mqtt_client.h"` but this also doesn't work.

Compiling using `make` works fine, although it recompiles the whole of mdf/idf every time for some reason.

### Expected Behavior
It should compile using cmake.

### Actual Behavior
It doesn't compile using cmake.

### Steps to repropduce
1. Try using mqtt with `#include "mqtt_client.h"`

### Debug Logs

```
[ 66%] Built target coap
/home/willemmerson/CLionProjects/mupgrade/main/mqtt.c:13:25: fatal error: mqtt_client.h: No such file or directory
compilation terminated.
make[2]: *** [main/CMakeFiles/main.dir/build.make:80: main/CMakeFiles/main.dir/mqtt.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 67%] Built target expat
[ 67%] Building C object main/CMakeFiles/main.dir/sntp.c.obj
[ 69%] Built target freemodbus
[ 70%] Built target mqtt
[ 71%] Built target openssl
[ 71%] Built target protobuf-c
[ 86%] Built target libsodium
[ 88%] Built target protocomm
make[1]: *** [CMakeFiles/Makefile2:2931: main/CMakeFiles/main.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 89%] Built target spiffs
[ 90%] Built target ulp
[ 90%] Performing configure step for 'bootloader'
[ 90%] Built target blank_ota_data
CMake Error at /home/willemmerson/esp/esp-mdf/esp-idf/tools/cmake/project.cmake:25 (include):
  include could not find load file:

    components
Call Stack (most recent call first):
  CMakeLists.txt:19 (include)

CMake Error at /home/willemmerson/esp/esp-mdf/esp-idf/tools/cmake/project.cmake:28 (include):
  include could not find load file:

    idf_functions
Call Stack (most recent call first):
  CMakeLists.txt:19 (include)

CMake Error at /home/willemmerson/esp/esp-mdf/esp-idf/tools/cmake/project.cmake:57 (idf_set_global_variables):
  Unknown CMake command "idf_set_global_variables".
Call Stack (most recent call first):
  CMakeLists.txt:20 (project)

-- Configuring incomplete, errors occurred!

```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

zhanzhaocheng, 2019-06-28:
You need to add `set(COMPONENT_REQUIRES mqtt)` in `CMakeLists.txt`.
If not, try running [mqtt_example](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/mqtt_example),compare the `CMakeLists.txt` file

will-emmerson, 2019-06-28:
Hi, this is still not working, although the mqtt error has gone away now.
When I try to compile the `get-started` example (unchanged), this is the output:

```
[willemmerson@thinkpad get-started]$ idf.py build
Checking Python dependencies...
Python requirements from /home/willemmerson/esp/esp-idf/requirements.txt are satisfied.
Running make in directory /home/willemmerson/CLionProjects/get-started/build
Executing "make -j 6 all"...
-- Building empty aws_iot component due to configuration
-- Building empty aliyun iot sdk component due to configuration
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet lwip tcpip_adapter partition_table app_update spi_flash mbedtls micro-ecc bootloader_support nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx nghttp esp-tls tcp_transport esp_http_client json mcommon bt mespnow miniz mwifi mconfig console mdns esp_http_server mupgrade mlink main esp-aliyun maliyun_linkkit wear_levelling sdmmc fatfs mdebug asio jsmn aws_iot bootloader coap esp_adc_cal esp_https_ota esptool_py expat freemodbus idf_test libsodium mqtt openssl protobuf-c protocomm spiffs ulp wifi_provisioning
-- Component paths: /home/willemmerson/esp/esp-mdf/esp-idf/components/soc;/home/willemmerson/esp/esp-mdf/esp-idf/components/log;/home/willemmerson/esp/esp-mdf/esp-idf/components/heap;/home/willemmerson/esp/esp-mdf/esp-idf/components/xtensa-debug-module;/home/willemmerson/esp/esp-mdf/esp-idf/components/app_trace;/home/willemmerson/esp/esp-mdf/esp-idf/components/freertos;/home/willemmerson/esp/esp-mdf/esp-idf/components/vfs;/home/willemmerson/esp/esp-mdf/esp-idf/components/newlib;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp_ringbuf;/home/willemmerson/esp/esp-mdf/esp-idf/components/driver;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp_event;/home/willemmerson/esp/esp-mdf/esp-idf/components/ethernet;/home/willemmerson/esp/esp-mdf/esp-idf/components/lwip;/home/willemmerson/esp/esp-mdf/esp-idf/components/tcpip_adapter;/home/willemmerson/esp/esp-mdf/esp-idf/components/partition_table;/home/willemmerson/esp/esp-mdf/esp-idf/components/app_update;/home/willemmerson/esp/esp-mdf/esp-idf/components/spi_flash;/home/willemmerson/esp/esp-mdf/esp-idf/components/mbedtls;/home/willemmerson/esp/esp-mdf/esp-idf/components/micro-ecc;/home/willemmerson/esp/esp-mdf/esp-idf/components/bootloader_support;/home/willemmerson/esp/esp-mdf/esp-idf/components/nvs_flash;/home/willemmerson/esp/esp-mdf/esp-idf/components/pthread;/home/willemmerson/esp/esp-mdf/esp-idf/components/smartconfig_ack;/home/willemmerson/esp/esp-mdf/esp-idf/components/wpa_supplicant;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp32;/home/willemmerson/esp/esp-mdf/esp-idf/components/cxx;/home/willemmerson/esp/esp-mdf/esp-idf/components/nghttp;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp-tls;/home/willemmerson/esp/esp-mdf/esp-idf/components/tcp_transport;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp_http_client;/home/willemmerson/esp/esp-mdf/esp-idf/components/json;/home/willemmerson/esp/esp-mdf/components/mcommon;/home/willemmerson/esp/esp-mdf/esp-idf/components/bt;/home/willemmerson/esp/esp-mdf/components/mespnow;/home/willemmerson/esp/esp-mdf/components/third_party/miniz;/home/willemmerson/esp/esp-mdf/components/mwifi;/home/willemmerson/esp/esp-mdf/components/mconfig;/home/willemmerson/esp/esp-mdf/esp-idf/components/console;/home/willemmerson/esp/esp-mdf/esp-idf/components/mdns;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp_http_server;/home/willemmerson/esp/esp-mdf/components/mupgrade;/home/willemmerson/esp/esp-mdf/components/mlink;/home/willemmerson/CLionProjects/get-started/main;/home/willemmerson/esp/esp-mdf/components/third_party/esp-aliyun;/home/willemmerson/esp/esp-mdf/components/maliyun_linkkit;/home/willemmerson/esp/esp-mdf/esp-idf/components/wear_levelling;/home/willemmerson/esp/esp-mdf/esp-idf/components/sdmmc;/home/willemmerson/esp/esp-mdf/esp-idf/components/fatfs;/home/willemmerson/esp/esp-mdf/components/mdebug;/home/willemmerson/esp/esp-mdf/esp-idf/components/asio;/home/willemmerson/esp/esp-mdf/esp-idf/components/jsmn;/home/willemmerson/esp/esp-mdf/esp-idf/components/aws_iot;/home/willemmerson/esp/esp-mdf/esp-idf/components/bootloader;/home/willemmerson/esp/esp-mdf/esp-idf/components/coap;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp_adc_cal;/home/willemmerson/esp/esp-mdf/esp-idf/components/esp_https_ota;/home/willemmerson/esp/esp-mdf/esp-idf/components/esptool_py;/home/willemmerson/esp/esp-mdf/esp-idf/components/expat;/home/willemmerson/esp/esp-mdf/esp-idf/components/freemodbus;/home/willemmerson/esp/esp-mdf/esp-idf/components/idf_test;/home/willemmerson/esp/esp-mdf/esp-idf/components/libsodium;/home/willemmerson/esp/esp-mdf/esp-idf/components/mqtt;/home/willemmerson/esp/esp-mdf/esp-idf/components/openssl;/home/willemmerson/esp/esp-mdf/esp-idf/components/protobuf-c;/home/willemmerson/esp/esp-mdf/esp-idf/components/protocomm;/home/willemmerson/esp/esp-mdf/esp-idf/components/spiffs;/home/willemmerson/esp/esp-mdf/esp-idf/components/ulp;/home/willemmerson/esp/esp-mdf/esp-idf/components/wifi_provisioning
-- Adding linker script /home/willemmerson/CLionProjects/get-started/build/esp32/esp32_out.ld
-- Adding linker script /home/willemmerson/esp/esp-mdf/esp-idf/components/esp32/ld/esp32.common.ld
[... cut ...]
[ 97%] Building C object esp32/CMakeFiles/esp32.dir/system_api.c.obj
[ 98%] Building C object esp32/CMakeFiles/esp32.dir/wifi_init.c.obj
[ 98%] Building C object esp32/CMakeFiles/esp32.dir/task_wdt.c.obj
[ 98%] Building C object esp32/CMakeFiles/esp32.dir/wifi_os_adapter.c.obj
[ 98%] Building C object esp32/CMakeFiles/esp32.dir/hwcrypto/aes.c.obj
[ 98%] Building C object esp32/CMakeFiles/esp32.dir/hwcrypto/sha.c.obj
[ 98%] Linking C static library libesp32.a
[ 98%] Built target esp32
make: *** [Makefile:84: all] Error 2
make failed with exit code 2
```

will-emmerson, 2019-06-28:
Just out of interest, it works when using `idf.py app-flash` so the problem seems to be in bootloader.

bonastreyair, 2019-07-08:
Same thing happens to me... so new bootloader version? It was working before... so what has changed?

will-emmerson, 2019-07-08:
`#include "../../components/mesh_mqtt_handle/include/mesh_mqtt_handle.h"` doesn't work, you need to add `set(COMPONENT_REQUIRES mqtt)` in `CMakeLists.txt` as has been mentioned in a previous comment, then `#include "mqtt_client.h"` will work.

bonastreyair, 2019-07-08:
The problem I have is different, I can't compile the mqtt_example from the esp-mdf. Maybe I should open a new issue... it may not be related...

```
Toolchain path: /Users/Yair/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/Yair/esp/esp-mdf/esp-idf/requirements.txt are satisfied.
CC build/main/mqtt_example.o
/Users/Yair/esp/esp-mdf/examples/function_demo/mwifi/mqtt_example/main/mqtt_example.c:28:30: fatal error: mesh_mqtt_handle.h: No such file or directory
compilation terminated.
make[1]: *** [mqtt_example.o] Error 1
make: *** [component-main-build] Error 2
```
It can't find the mesh_mqtt_handle.h (it is in components/mesh_mqtt_handle) I erased all mdf, and clone it again and it doesn't work... why could that be?

will-emmerson, 2019-07-08:
Not sure, maybe make another issue. I am going to close this issue because I've realised what my problem was, I had both mdf and idf installed and they were getting mixed up, removing idf fixed the problem.
Are you saying that unchanged mqtt_example isn't compiling with either make or cmake?

bonastreyair, 2019-07-08:
I had idf also next to mdf, but I have erased them all, installed from nothing cloning the repository again, I run mqtt_example and I get this error... is so wierd and I don't understand what is happening... (you can close the issue since it is no the same, thanks for your help!)

bonastreyair, 2019-07-08:
I think I solved just deleting the "component.mk" on the components/ folder in the mqtt_example/... I don't know why is that.. but it was solved!

closed 2019-07-08 by will-emmerson