espressif/esp-mdf #62: Wireless Debug Example not working

hchaudhary1, 2019-07-05:
Wireless Debug Example not working

I tested this on Wrover 4.1 (with and without SDcard):
Commit: 370e5dc86e4f0bd369b17bc2928ad2f031bcc456

I have a small mesh with 9 devices on it, running the exact "router" example with no modifications.
Flashed the mdebug example into wrover and did a "log -a xx:xx:xx:xx:xx:xx" command, and it shows nothing. I have tried different flags such as "-t * -l VER", with no luck.

zhanzhaocheng, 2019-07-06:
You need to enable `mdebug` in the example of `router` and the following function under 'mespnow_init()`, see: [enable mdebug](https://github.com/espressif/esp-mdf/blob/master/examples/development_kit/light/main/light_example.c#L475)

```c
  /**
      * @brief Add debug function, you can use serial command and wireless debugging
      */
     /*< Waiting above log print complete, avoid bad data in terminal */
     vTaskDelay(pdMS_TO_TICKS(50));
     MDF_ERROR_ASSERT(mdebug_console_init());
     MDF_ERROR_ASSERT(mdebug_espnow_init());
     Mdebug_cmd_register_common();
```

hchaudhary1, 2019-07-06:
worked!

closed 2019-07-06 by hchaudhary1