espressif/esp-mdf #63: mesh_mqtt_handle.h not found in mqtt_example using make

eos1d3, 2019-07-14:
Example project: mqtt_example
MDF version: 64a6b60
Toolchain: xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0

After compiling using make, I got this fatal error:
```
/Users/username/Desktop/mwifi/mqtt_example/main/mqtt_example.c:27:10: fatal error: mesh_mqtt_handle.h: No such file or directory
 #include "mesh_mqtt_handle.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [mqtt_example.o] Error 1
make: *** [component-main-build] Error 2
```

What is the real problem? I followed https://github.com/espressif/esp-mdf/issues/61#issuecomment-509184865  to fix and it works. But is this the right way to correct the problem?

eos1d3, 2019-07-14:
I removed the esp folder together with esp-idf folder and reinstalled all of them. But this time I did not install CMake. And it compiles without any warning.

Does CMake tools make the conflict with Make?

zhanzhaocheng, 2019-07-15:
Make and CMake sometimes conflict, if you switch to each other, please `make clean`

eos1d3, 2019-07-15:
I remember I did use `make clean` more than once. but it seems CMake tools are not completely removed.

The MQTT example is now running perfectly. Thanks!

closed 2019-07-15 by eos1d3