espressif/esp-mdf #66: toolchain version not match

ginhton, 2019-07-19:
## problem
* I download the newest toolchain, and cloned the newest `esp-idf` and `esp-mdf` code. 
* The demo in `esp-idf` runs well, but when I run `get-started`, I got this error.
* I just want to know how to solve this without use the fallback to the old toolchain.
* Any suggestion will be appreciated.

## error message
```text
Toolchain path: /home/i/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0

```

zhanzhaocheng, 2019-07-19:
Different versions of esp-idf, the build toolchain version is different.
The latest esp-idf is v4.0, so it can be compiled with the latest toolchain.
Esp-mdf is based on esp-idf v3.2 and must use the corresponding toolchain
[image]

ginhton, 2019-07-22:
Thank you for your answer!

I just wonder if I can use the latest `esp-idf` and `esp-mdf` at the same time using the same toolchain?

And I want to combine some `esp-idf` code and `esp-mdf` code later. Is it possible to mix the ble mesh and the wifi mesh in one esp32-wroom-32 node?

zhanzhaocheng, 2019-07-22:
1. I recommend that you develop based on release/v3.2, which is a stable version.
2. ESP-MESH + BLE can be an example of the parameter [light](https://github.com/espressif/esp-mdf/tree/master/examples/development_kit/light). This is feasible, we also have customers based on this way

ginhton, 2019-07-22:
Thank you for your advice!

you all are so genius!

closed 2019-07-22 by ginhton