espressif/esp-mdf #67: mwifi_get_root_status() sometimes returns false even though root is connected

will-emmerson, 2019-07-19:
## Environment
- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- MDF version: c1cc50f
- Development Env: CLion
- Operating System: Fedora
- Power Supply:  USB

## Problem Description
I am running something based off the mqtt example.
I noticed that sometimes, after a few hours or days, a node would stop sending mqtt messages. This only happens to nodes > level 2, e.g. there is a node between them and root.
I am using the same code as the example here:
https://github.com/espressif/esp-mdf/blob/c1cc50f6107fae59fbef570567331b91b0b59819/examples/function_demo/mwifi/mqtt_example/main/mqtt_example.c#L165

This blocks the mqtt sending task (node_write_task) if either node or root isn't connected.
It turned out the mqtt message wasn't being sent because `mwifi_get_root_status()` was returning false, even though `mwifi_is_connected()` is true. The root definitely is connected though because I can send mqtt messages to the node.

zhanzhaocheng, 2019-07-25:
I haven't been able to reproduce for two days. I fixed a possible problem with mwifi_get_root_status(). I hope you can help verify if the problem has been fixed.

If the problem is not resolved, please provide your full log.

will-emmerson, 2019-08-07:
Unfortunately the problem is still there. I updated MDF to `8dfc3c0` on Monday and it ran fine until a few minutes ago, when the same problem happened.
But I suspect it's something I've added to the example that is causing the problem. I'll run the vanilla mqtt example for a few days and see if it exhibits the same problem.