espressif/esp-mdf #68: Upgrade configuration is not initialized

Kasuhikdas, 2019-07-21:
Development Kit: [ESP32-Wrover-Kit]
Core (if using chip or module): ESP-WROOM32
Development Env:	[Make]
Operating System:	[Ubuntu]
Power Supply: [USB|external 5V]

Upgrade is failing and device restarts. But error stays the same so root keep sending the update.

W (16517) [mupgrade_node, 142]: Upgrade configuration is not initialized
W (16525) [mupgrade_node, 233]: <MDF_ERR_MUPGRADE_NOT_INIT> mupgrade_handle
W (16533) [nodemtsk.c, 302]: <MDF_ERR_MUPGRADE_NOT_INIT> mupgrade_handle

D (18087) [mupgrade_node, 80]: Running partition, label: ota_0, type: 0x0, subtype: 0x10, address: 0x1f0000
D (18098) [mupgrade_node, 82]: Update partition, label: ota_0, type: 0x0, subtype: 0x10, address: 0x1f0000
W (18108) [mupgrade_node, 93]: esp_ota_begin failed
D (18113) [mupgrade_node, 118]: Response mupgrade status, written_size: 0, response_size: 45
abort() was called at PC 0x400f02d9 on core 0
0x400f02d9: root_task at /home/ravi/wa/projects/esp32-fw/main/task/rootmtsk.c:42

Backtrace: 0x40092bac:0x3ffe8460 0x40092dc9:0x3ffe8480 0x400f02d9:0x3ffe84a0 0x40082e02:0x3ffe84c0 0x400f01a5:0x3ffe8510 0x401bedfd:0x3ffe8530 0x401bf134:0x3ffe8570 0x400f0aef:0x3ffe85a0 0x4009004d:0x3ffe8620
0x40092bac: doBacktrace at /home/ravi/wa/esp-mdf/esp-idf/components/esp32/panic.c:707

0x40092dc9: esp_error_check_failed_print at /home/ravi/wa/esp-mdf/esp-idf/components/esp32/panic.c:707

0x400f02d9: root_task at /home/ravi/wa/projects/esp32-fw/main/task/rootmtsk.c:42

0x40082e02: spi_flash_erase_sector at /home/ravi/wa/esp-mdf/esp-idf/components/spi_flash/flash_ops.c:197

0x400f01a5: node_read_task at /home/ravi/wa/projects/esp32-fw/main/task/nodemtsk.c:289

0x401bedfd: mupgrade_root_handle at /home/ravi/wa/esp-mdf/components/mupgrade/mupgrade_root.c:148 (discriminator 3)

0x401bf134: mupgrade_firmware_send at /home/ravi/wa/esp-mdf/components/mupgrade/mupgrade_root.c:356 (discriminator 3)

0x400f0aef: tcpip_adapter_down at /home/ravi/wa/esp-mdf/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1092

0x4009004d: _xt_tick_divisor_init at /home/ravi/wa/esp-mdf/esp-idf/components/freertos/xtensa_init.c:45

02:26

esp32-fw.bin

What is the reason for this error and what is the solution?

zhanzhaocheng, 2019-07-22:
This is due to a bug in your partition. Please refer to the [mupgrade example partition table](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mupgrade/partitions.csv).

Kasuhikdas, 2019-07-22:
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      0x9000,   16k
otadata,  data, ota,      0xd000,   8k
phy_init, data, phy,      0xf000,   4k
factory,  app,  factory,  0x10000,  1920k
ota_0,    app,  ota_0,    ,         1920k
ota_1,    app,  ota_1,    ,         1920k
coredump, data, coredump, ,         64K
reserved, data, 0xfe,     ,         128K

This is my partition table i have otaData partition

zhanzhaocheng, 2019-07-24:
1. 更新 ESP-MDF 的版本, 提供完整理运行日志
2. 添加 esp_ota_begin 错误码的打印, 如下所示
```
MDF_ERROR_GOTO(ret != MDF_OK, EXIT, "<%s> esp_ota_begin failed", mdf_err_to_name(ret));
```
[image]

closed 2019-11-05 by Kasuhikdas