espressif/esp-mdf #69: 0x4008 errors on esp_mesh_send(), NODE to ROOT

hchaudhary1, 2019-07-21:
I am running the 1.0 beta.
I have a fixed root mesh with 75 nodes. All the devices are within 200 feet, in my test room.
Most of the nodes are unable to TX to root at all, and give the following error, when attempting to send to root 0x4008.

```
    uint8_t buf[200];
    memset(buf, 0, 200);
    memcpy(buf, data, size);

    mesh_data_t mesh_data;
    mesh_data.data = buf;
    mesh_data.size = size;
    mesh_data.proto = MESH_PROTO_BIN;

    esp_mesh_send(NULL, &mesh_data, MESH_DATA_TODS, NULL, 0);
```

I am only sending 40bytes, every 30 seconds, from all nodes.
See attached pic of the logs, as seen by the ESPNOW sniffer.

I am running
-BLE server
-ESPNOW, mdebug
-ESP MESH

Settings for mesh are default, but with the following overrides:
```
    mwifi_init_config_t mesh_config   = MWIFI_INIT_CONFIG_DEFAULT();
    mesh_config.max_layer = 20;
#ifdef ROOT
    mesh_config.max_connection = 10;
#else
    mesh_config.max_connection = 6;
#endif
    mesh_config.beacon_interval_ms = 200;
    mesh_config.passive_scan_ms = 650;
    mesh_config.select_rssi = -77;
    mesh_config.switch_rssi = -80;
    mesh_config.cnx_rssi = -68;
    mesh_config.backoff_rssi = -74;
```
	

The issue is extremely bad, when you turn on verbose logging for mesh and idf.
I have ~30k of free heap, so I am not sure what the issue is, but definitely an issue with the espressif software MDF or IDF.
[image]

zhanzhaocheng, 2019-07-22:
1. Error code 0x4008 corresponds to a parameter error. From this point of view, it has little relationship with memory. Have you ever had this problem when there are few devices? Is there a perfect log?
[image]

2. If you are developing based on esp-mdf.
- It is recommended that you use the mwifi_xxx related API to avoid conflicts. Use mwifi_write to replace esp_mesh_send.
- There is no need to modify MWIFI_INIT_CONFIG_DEFAULT(), ESP-MESH will adapt according to the environment.
- You can try to [fix the root node example](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/no_router), or modify it based on the above:
- Your free memory is very small. You can refer to the following for Bluetooth configuration, and call `esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);` in app_main to release classic Bluetooth.

```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y
CONFIG_GATTC_ENABLE=
CONFIG_BLE_SMP_ENABLE=
CONFIG_BT_ACL_CONNECTIONS=2
CONFIG_BLE_SCAN_DUPLICATE=
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_BTDM_CONTROLLER_MODEM_SLEEP=
CONFIG_BT_STACK_NO_LOG=

```

hchaudhary1, 2019-07-22:
Hi zhanzhaocheng,
1. yes, with 10 devices, everything is working perfect, no errors. I did also see the esp_mesh.h, but all my parameters were perfect.

2. mwifi_xxx is a good idea! I will try that
-  Yes, I am using esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
- Also, I check against the ROOT NODE example

hchaudhary1, 2019-07-22:
using the mwifi_xxx worked. thanks

closed 2019-07-22 by hchaudhary1

thilakshiK, 2019-07-25:
Hi hchaundhary1, 

I'm also trying to transmit data from multiple nodes to the root at once. I'm still new to this esp mesh concept. Can you please kindly explain to me how you did it.

zhanzhaocheng, 2019-07-25:
@thilakshiK  You can modify it based on the [mwifi example](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi) or [get-started](https://github.com/espressif/esp-mdf/tree/master/examples/get-started)