espressif/esp-mdf #70: WANT a more detail instruction of esp ble mesh console demo

ginhton, 2019-07-22:
I am trying to run all the esp ble mesh demo. But some of them all lack of detail instruction.

For example, the `ble_mesh_console/ble_mesh_node` demo. The instructions are:
```markdown
1. Build the ble mesh node console demo with sdkconfig.default  
2. register node and set oob info, load model to init ble mesh node  
3. enable bearer, so that it can be scaned and provisioned by provisioner  
```
I think this is not so clear for a learner to understand.
For example, how to use the **set oob info**? Here is the info about that command:
```bash
bmoob  [-s <value>] [-l <length>] [-x <size>] [-o <actions>] [-y <size>] [-i <actions>] [-p <address>]
  ble mesh: provisioner/node config OOB parameters
    -s <value>  Static OOB value
   -l <length>  Static OOB value length
     -x <size>  Maximum size of Output OOB
  -o <actions>  Supported Output OOB Actions
     -y <size>  Maximum size of Input OOB
  -i <actions>  Supported Input OOB Actions
  -p <address>  start address assigned by provisioner
```

If you can give a more detail instruction, I would be very appreciated. Thank you all!

ESP-LBB, 2019-07-22:
Sorry this repo is for wifi mesh, for your esp ble mesh request, I'll forward to my collegues to see if there is anywhere you can post this request.

WCCWCC, 2019-07-22:
Hi ginhton,
There is a reference document here:

# Introduction

This demo helps you to quickly implement the ESP BLE Mesh functionality in your applications. For example, you can implement this demo on your device working as a BLE Mesh Provisioner to provisioning unprovisioned devices, so they can join a mesh network, or working as a node to control the state of lights.

Before starting, you could enter the `help` command in your serial port tool to check all the commands that are already supported in this demo.

# Project Structure

The folder `ble_mesh_provisioner` contains the following files and subfolders:

| File Name        |Description               |
| ----------------------|------------------------- |
| `ble_mesh_adapter`      | Adapts the codes for initialization by using the console. |
| `ble_mesh_cfg_srv_model` | Stores all the structs related to the initialization of models. |
| `ble_mesh_console_lib` | Stores utilities for converting data formats. |
| `ble_mesh_console_main` | The main function that initializes the console and registers all the commands |
| `ble_mesh_console_system`  | Implements the system-related commands: `restart`, `free`, `make`. |
| `ble_mesh_reg_cfg_client_cmd`| Implements the **Configuration Client** model related command: `bmccm`. |
| `ble_mesh_reg_gen_onoff_client_cmd`| Implements the **Generic OnOff Client** model related command: `bmgocm`.|
| `ble_mesh_reg_test_perf_client_cmd` | Implements the performance test related command: `bmcperf`. |
| `ble_mesh_register_node_cmd`  | Implements the node related commands: `bmreg`, `bmoob`, `bminit`, `bmpbearer`, `bmtxpower`.  |
| `ble_mesh_register_provisioner_cmd` | Implements the provisioner related commands: `bmpreg`, `bmpdev`, `bmpbearer`, `bmpgetn`, `bmpaddn`,`bmpbind`,`bmpkey`. |
| `register_bluetooth`    | Implements the command to get the MAC address of a BLE device: `btmac` |

# What You Need

Download and flash the `ble_mesh_console/ble_mesh_provisioner` project to your ESP32 development board and then use commands below to implement this demo.

## Implement the demo as a provisioner

The commands that are related to implementing this demo as a provisioner are shown in the table below:

| Command        |Description               |
| ----------------------|------------------------- |
| `bmreg`  | ble mesh: provisioner/node register callback |
| `bmpreg` | ble mesh provisioner: register callback |
| `bmoob`  | ble mesh: provisioner/node config OOB parameters|
| `bminit` | ble mesh: provisioner/node init |
| `bmpbearer`  | ble mesh provisioner: enable/disable different bearers |
| `bmpdev`| ble mesh provisioner: add/delete unprovisioned device |

Please follow the steps below to implement this demo as a provisioner. 

1. Enter `bmreg` to register the `ble_mesh_prov_cb` and `ble_mesh_model_cb` callback functions.
    * `ble_mesh_prov_cb` handles all events triggered when your board is provisioning other devices. 
    * `ble_mesh_model_cb` handles all events triggered when any server model of your board receives or sends any message. 

2. Enter `bmpreg` to initialize your board as a provisioner, which registers the `ble_mesh_prov_adv_cb` callback function.
    * `ble_mesh_prov_adv_cb` will be triggered when your board receives an unprovisioned Device beacon broadcasted by an unprovisioned device.

3. Enter `bmoob -p 0x5` to initialize all the information needed for a device to be a provisioner. 
    * The parameter `-p` stands for provisioner and its value `0x5` indicates the starting address assigned to the node by the provisioner.

4. Enter `bminit -m 0x01` to initialize the model that is necessary for the provisioner's provisioning. 
    * The parameter `-m` stands for model and its value `0x01` indicates the model ID of the **Configuration Client** model. 
    
    After this command, the initialization now has been completed.

5. Enter `bmpbearer -b 0x3 -e 1` to enable bearers, which include the PB-ADV bearer and the PB-GATT bearer. 
    * The parameter `-b` stands for "bearer" and its value `0x3` indicates both the PB-ADV bearers and the PB-GATT bearers are enabled; 
    * The parameter `-e` stands for "enable" and its value `1` indicates to enable the bearer. 
    
    After this command, the provisioner now can send and receive messages.

6. Enter `bmpdev -z add -d MAC_address -b 0x3 -a 0 -f 1` to start provisioning an unprovisioned device.
    
    * The parameter `MAC_address` indicates the MAC address of the unprovisioned device. Here, you can use the `btmac` command to query the unprovisioned device's MAC address. 
[... 125 lines cut ...]
The main program constantly reads data from the command line. `esp_console_run` will parse the argument and call the callback function of the previously initialized command.

```c
    /* Main loop */
    while (true) {
        /* Get a line using linenoise.
         * The line is returned when ENTER is pressed.
         */
        char *line = linenoise(prompt);
        if (line == NULL) { /* Ignore empty lines */
            continue;
        }
        /* Add the command to the history */
        linenoiseHistoryAdd(line);

        /* Try to run the command */
        int ret;
        esp_err_t err = esp_console_run(line, &ret);
        ...
        /* linenoise allocates line buffer on the heap, so need to free it */
        linenoiseFree(line);
    }
    return;
}
```

ginhton, 2019-07-22:
Thank you all!

I will follow your instructions and continue my experiment.

you are genius!

ginhton, 2019-07-22:
Thank for your great instructions! I got "Provisioning:Success"

# Further Instruction?
Is this the end of this demo? Are there some further steps? For example:
```text
add and bind NetKey, Appkey for provisioner and nodes.
use GenOnOffClient model in provisioner to trigger GenOnOffServer in node.
```

If there are, how can I do those? Are they implemented in this demo?
 (Before asking this question, I have tried `bmpbind`, `bmpkey`, `bmgocm` without parameters. And I browsed the source code, but still have no idea what is the right parameter)

# Instruction Typo
By the way, there is an typo in your instruction in Section `2.2 Implement the demo as a node` step 4.
```
Enter bmpbearer -b 0x3 -e 1 to enable bearers, which include the PB-ADV bearer and the PB-GATT bearer.
```
`bmpbearer` should be `bmnbearer`.

# Acknowledgement
Thank for all your great work! Have a good day!

jai-kannan1184, 2020-04-15:
As mentioned by ginhton i would also like to know how to bind appkeys to a model for on off server and on off client based on this example