espressif/esp-mdf #71: mesh -> with router example doesn't connect to tcp server

cgfoed, 2019-07-24:
After some debugging I'm quite confident that there is a bug in 
https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/router

I always get the error
W (67050) [router_example, 59]: socket connect, ret: -1, ip: 192.168.4.1, port: 8070

What is working:
- Connection between different TCP-apps on iOS using the ESP32-softAP Wifi which runs on a seperate WROOM Module using softAP example
- Connection between TCP-apps to a TCP Socket running besides the softAP Wifi using Neil Kolbans snippet: https://github.com/nkolban/esp32-snippets/tree/master/sockets
- Connection between a WROOM running https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/tcp_client and an iOS TCP Server

I got every samplecode working so fart - except function_demo/mwifi/router

Tried hard coding IP and Server to socket_tcp_client_create and different IP / Port configs i've got no clue to where to look for the error now

zhanzhaocheng, 2019-07-24:
1. I hope you provide a complete log
2. You can create a TCP server on your mobile APP or computer to eliminate interference from ESP32 softAP.

cgfoed, 2019-07-24:
2. You can create a TCP server on your mobile APP or computer to eliminate interference from ESP32 softAP.
<< I did that. I can connect to a TCP server on my mobile APP with other devices or other WROOM running https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/tcp_client 

1. I hope you provide a complete log

First log from 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/tcp_client 

```
        rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
	configsip: 0, SPIWP:0xee
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:DIO, clock div:2
	load:0x3fff0018,len:4
	load:0x3fff001c,len:6472
	load:0x40078000,len:11664
	ho 0 tail 12 room 4
	load:0x40080400,len:6664
	entry 0x40080764
	I (30) boot: ESP-IDF v4.0-dev-1287-gd7e659df2 2nd stage bootloader
	I (30) boot: compile time 11:36:25
	I (30) boot: Enabling RNG early entropy source...
	I (36) boot: SPI Speed      : 40MHz
	I (40) boot: SPI Mode       : DIO
	I (44) boot: SPI Flash Size : 4MB
	I (48) boot: Partition Table:
	I (52) boot: ## Label            Usage          Type ST Offset   Length
	I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
	I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
	I (74) boot:  2 factory          factory app      00 00 00010000 00100000
	I (82) boot: End of partition table
	I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x19e64 (106084) map
	I (132) esp_image: segment 1: paddr=0x00029e8c vaddr=0x3ffb0000 size=0x0323c ( 12860) load
	I (137) esp_image: segment 2: paddr=0x0002d0d0 vaddr=0x40080000 size=0x00400 (  100x40080000: _WindowOverflow4 at /home/grube/esp/esp-idf/components/freertos/xtensa_vectors.S:177824) load
	I (139) esp_image: segment 3: paddr=0x0002d4d8 vaddr=0x40080400 size=0x02b38 ( 11064) load
	I (152) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x69074 (430196) map
	0x400d0018: _stext at ??:?
	I (308) esp_image: segment 5: paddr=0x00099094 vaddr=0x40082f38 size=0x0d154 ( 53588) load
	0x40082f38: esp_flash_get_size at /home/grube/esp/esp-idf/components/spi_flash/esp_flash_api.c:609
	I (340) boot: Loaded app from partition at offset 0x10000
	I (340) boot: Disabling RNG early entropy source...
	I (340) cpu_start: Pro cpu up.
	I (344) cpu_start: Application information:
	I (349) cpu_start: Project name:     tcp_client
	I (354) cpu_start: App version:      v4.0-dev-1287-gd7e659df2
	I (360) cpu_start: Compile time:     Jul 24 2019 11:36:23
	I (366) cpu_start: ELF file SHA256:  378f9738b8b92605...
	I (372) cpu_start: ESP-IDF:          v4.0-dev-1287-gd7e659df2
	I (379) cpu_start: Starting app cpu, entry point is 0x400810a4
	0x400810a4: call_start_cpu1 at /home/grube/esp/esp-idf/components/esp32/cpu_start.c:283
	I (0) cpu_start: App cpu up.
	I (389) heap_init: Initializing. RAM available for dynamic allocation:
	I (396) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (402) heap_init: At 3FFB91A8 len 00026E58 (155 KiB): DRAM
	I (408) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (415) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (421) heap_init: At 4009008C len 0000FF74 (63 KiB): IRAM
	I (427) cpu_start: Pro cpu start user code
	I (446) spi_flash: detected chip: generic
	I (446) spi_flash: flash io: dio
	I (446) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (549) wifi: wifi driver task: 3ffc0a80, prio:23, stack:3584, core=0
	I (549) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (549) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (579) wifi: wifi firmware version: 3c39e82
	I (579) wifi: config NVS flash: enabled
	I (579) wifi: config nano formating: disabled
	I (579) wifi: Init dynamic tx buffer num: 32
	I (579) wifi: Init data frame dynamic rx buffer num: 32
[... 886 lines cut ...]
I (13250) event: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
I (13251) [router_example, 332]: event_loop_cb, event: 17
I (13253) [router_example, 353]: Root obtains the IP address. It is posted by LwIP stack automatically
I (13263) [router_example, 84]: TCP client read task is running
I (13264) [router_example, 169]: TCP client write task is running
I (13270) [router_example, 43]: Create a tcp client, ip: 192.168.4.4, port: 8070
I (13285) [router_example, 56]: socket created, sockfd: 54
D (15351) [router_example, 246]: Node send, size: 72, data: {"src_addr": "24:0a:c4:1d:4f:b0","data": "Hello TCP Server!","count": 1}
D (18354) [router_example, 246]: Node send, size: 72, data: {"src_addr": "24:0a:c4:1d:4f:b0","data": "Hello TCP Server!","count": 2}
I (22531) [router_example, 281]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -14, node num: 1, free heap: 158176
W (31544) [router_example, 59]: socket connect, ret: -1, ip: 192.168.4.4, port: 8070
I (32048) [router_example, 43]: Create a tcp client, ip: 192.168.4.4, port: 8070
I (32049) [router_example, 56]: socket created, sockfd: 55
I (32531) [router_example, 281]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -14, node num: 1, free heap: 158080
[... cut ...]
I (52531) [router_example, 281]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -14, node num: 1, free heap: 157988
I (62531) [router_example, 281]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -14, node num: 1, free heap: 158192
W (69564) [router_example, 59]: socket connect, ret: -1, ip: 192.168.4.4, port: 8070

```

While the app doesn't see any new client connection

zhanzhaocheng, 2019-07-24:
1. Your esp-mdf version is dirty (esp-mdf version: v1.0-beta1-2-gc1cc50f-dirty), do you have any changes, (git diff)
2. Can you use the router for a test? I can't reproduce your question.
3. You add an error code to print when the connection fails.
```c
     Ret = connect(sockfd, (struct sockaddr *)&server_addr, sizeof(struct sockaddr_in));
     MDF_ERROR_GOTO(ret < 0, ERR_EXIT, "<%s> socket connect, ret: %d, ip: %s, port: %d",
                    Strerror(errno), ret, ip, port);
```
[image]

cgfoed, 2019-07-24:
```
I (13270) [router_example, 43]: Create a tcp client, ip: 192.168.4.50, port: 8070
D (15346) [router_example, 245]: Node send, size: 72, data: {"src_addr": "24:0a:c4:1d:4f:b0","data": "Hello TCP Server!","count": 1}
D (18349) [router_example, 245]: Node send, size: 72, data: {"src_addr": "24:0a:c4:1d:4f:b0","data": "Hello TCP Server!","count": 2}
I (22531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 158168
W (31544) [router_example, 58]: <Software caused connection abort> socket connect, ret: -1, ip: 192.168.4.50, port: 8070
I (32045) [router_example, 43]: Create a tcp client, ip: 192.168.4.50, port: 8070
I (32531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 158080
I (42531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 158272
W (50549) [router_example, 58]: <Software caused connection abort> socket connect, ret: -1, ip: 192.168.4.50, port: 8070
I (51050) [router_example, 43]: Create a tcp client, ip: 192.168.4.50, port: 8070
I (52531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 157988
I (62531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 158180
W (69554) [router_example, 58]: <Software caused connection abort> socket connect, ret: -1, ip: 192.168.4.50, port: 8070
I (70055) [router_example, 43]: Create a tcp client, ip: 192.168.4.50, port: 8070
I (72531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 157896
I (82531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 157996
W (88562) [router_example, 58]: <Software caused connection abort> socket connect, ret: -1, ip: 192.168.4.50, port: 8070
```

```
git show
commit c1cc50f6107fae59fbef570567331b91b0b59819 (HEAD -> master, origin/master, origin/HEAD)
Merge: 5aa32d3 6622a98
Author: Zhan Zhao Chen <zhanzhaochen@espressif.com>
Date:   Wed Jul 17 15:44:20 2019 +0800

    Merge branch 'docs/README.md' into 'master'
    
    docs(README): Modify the hyperlink in the README
    
    See merge request application/esp-mdf!298
```

```
~/esp/esp-mdf/examples/function_demo/mwifi/router$ git diff
diff --git a/examples/function_demo/mwifi/router/main/router_example.c b/examples/function_demo/mwifi/router/main/router_example.c
index cfc0846..9193bdc 100755
--- a/examples/function_demo/mwifi/router/main/router_example.c
+++ b/examples/function_demo/mwifi/router/main/router_example.c
@@ -54,8 +54,8 @@ static int socket_tcp_client_create(const char *ip, uint16_t port)
     MDF_ERROR_GOTO(sockfd < 0, ERR_EXIT, "socket create, sockfd: %d", sockfd);
 
     ret = connect(sockfd, (struct sockaddr *)&server_addr, sizeof(struct sockaddr_in));
-    MDF_ERROR_GOTO(ret < 0, ERR_EXIT, "socket connect, ret: %d, ip: %s, port: %d",
-                   ret, ip, port);
+    MDF_ERROR_GOTO(ret < 0, ERR_EXIT, "<%s> socket connect, ret: %d, ip: %s, port: %d",
+                   strerror(errno), ret, ip, port);
     return sockfd;
 
 ERR_EXIT:

```

I'll try dedicated router tomorrow.

cgfoed, 2019-07-25:
✔ iPad TCP Server.......................<Hardware AP>........WROOM ESP-MDF router sample

The question remains why I can't use the espressif softAP running on a WROOM with **this specific example** - when all other combinations of TCP communication work.

As I **need** to use an ESP32 as "router" in my specific project I have no other choice as to investigate this issue further.

Any help would be great!

cgfoed, 2019-07-25:
maybe interesting

I get an other error, when I try:
✖ WROOM ESP-MDF router sample.....<-softAP->........WROOM TCP Server

```
I (168769) [router_example, 43]: Create a tcp client, ip: 192.168.4.1, port: 5001
W (168771) [router_example, 58]: <Connection reset by peer> socket connect, ret: -1, ip: 192.168.4.1, port: 5001
```
**This** error occurs **immediately**, while in the other config:

✖ WROOM ESP-MDF router sample.....<-softAP->........iPhone TCP Server
```
I (51050) [router_example, 43]: Create a tcp client, ip: 192.168.4.50, port: 8070
I (52531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 157988
I (62531) [router_example, 280]: System information, channel: 1, layer: 1, self mac: 24:0a:c4:1d:4f:b0, parent bssid: 24:0a:c4:c1:07:41, parent rssi: -7, node num: 1, free heap: 158180
W (69554) [router_example, 58]: <Software caused connection abort> socket connect, ret: -1, ip: 192.168.4.50, port: 8070
```
the other error occurs after 30s or so.

Still this works fine:
✔ iPhone TCP Client ....................................<-softAP->........WROOM TCP Server

zhanzhaocheng, 2019-07-25:
@cgfoed  We will fix ESP32 as a softap issue as soon as possible. Once we fix it, we will notify you immediately. You are temporarily using a router for testing.

zhanzhaocheng, 2019-07-25:
Because our ESP-MDF uses the 192.168.4.1 network segment by default, if softap also uses this network segment, there will be conflicts. Please modify the softap network segment as follows:
[image]

```c
    tcpip_adapter_ip_info_t ip_info = {
        .ip.addr      = ipaddr_addr("192.168.1.1"),
        .netmask.addr = ipaddr_addr("255.255.255.0"),
        .gw.addr      = ipaddr_addr("192.168.1.1"),
    };

    tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP);
    tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &ip_info);
    tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP);

```

cgfoed, 2019-07-25:
Thanks! 
That did the trick.

closed 2019-07-25 by cgfoed

Cabral2017, 2021-04-23:
Hello @zhanzhaocheng , I have the same problem using esp-mdf router example. Can you explan me how I can fix this erro:

`I (16945) mesh: <nvs>write layer:0
I (16946) mesh: [DONE]connect to router:VERO-FELIPE, channel:9, rssi:-65, b0:08:75:34:e7:34[layer:0, assoc:0], my_vote_num:1/voter_num:1, rc[24:0a:c4:fc:72:4d/-65/1]
I (17083) wifi:new:<9,1>, old:<9,1>, ap:<9,1>, sta:<9,0>, prof:9
I (17084) wifi:state: init -> auth (b0)
I (17087) wifi:state: auth -> assoc (0)
I (17094) wifi:state: assoc -> run (10)
I (17095) wifi:state: run -> init (4a0)
I (17095) wifi:new:<9,0>, old:<9,1>, ap:<9,1>, sta:<9,0>, prof:9
I (17099) wifi:new:<9,0>, old:<9,0>, ap:<9,1>, sta:<9,0>, prof:9
I (17107) [mwifi, 187]: Parent is disconnected, reason: 4
I (17110) [router_example, 332]: event_loop_cb, event: 8
I (17111) mesh: [wifi]disconnected reason:4(assoc expire), continuous:1/max:12, root, vote(,stopped)<><>
I (17117) [router_example, 349]: Parent is disconnected on station interface
I (19455) [mwifi, 187]: Parent is disconnected, reason: 205
I (19456) [router_example, 332]: event_loop_cb, event: 8
I (19457) mesh: [wifi]disconnected reason:205(), continuous:2/max:12, root, vote(,stopped)<><>
I (19458) [router_example, 349]: Parent is disconnected on station interface
I (19590) wifi:new:<9,1>, old:<9,0>, ap:<9,1>, sta:<9,0>, prof:9
I (19592) wifi:state: init -> auth (b0)
I (19595) wifi:state: auth -> assoc (0)
I (19603) wifi:state: assoc -> run (10)
I (19618) wifi:connected with VERO-FELIPE, aid = 108, channel 9, BW20, bssid = b0:08:75:34:e7:34
I (19619) wifi:security: WPA2-PSK, phy: bgn, rssi: -64
I (19622) wifi:pm start, type: 0

I (19626) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (19706) mesh: <nvs>write layer:1
I (19707) [mwifi, 174]: Parent is connected
I (19708) [router_example, 332]: event_loop_cb, event: 7
I (19709) [mwifi, 257]: State represents: 0
I (19715) [router_example, 340]: Parent is connected on station interface
I (19716) mesh: <flush>upstream packets, connections(max):6, waiting:0, upQ:0
I (19725) [router_example, 332]: event_loop_cb, event: 11
I (19735) [router_example, 332]: event_loop_cb, event: 14
I (19741) mesh: <flush>root
I (19744) mesh: [TXQ]<max:32>up(0, be:0), down(0, be:0), mgmt:0, xon(req:0, rsp:0), bcast:0, wnd(0, parent:00:00:00:00:00:00)
I (19755) mesh: [RXQ]<max:32 = cfg:32 + extra:0>self:0, <max:32 = cfg:32 + extra:0>tods:0
I (19880) wifi:AP's beacon interval = 1024000 us, DTIM period = 1
I (19884) mesh: [scan]new scanning time:1500ms, beacon interval:1000ms
D (20066) [router_example, 246]: Node send, size: 72, data: {"src_addr": "24:0a:c4:fc:72:4c","data": "Hello TCP Server!","count": 0}
I (21529) [router_example, 279]: System information, channel: 9, layer: 1, self mac: 24:0a:c4:fc:72:4c, parent bssid: b0:08:75:34:e7:34, parent rssi: -61, node num: 1, free heap: 178524
I (22693) esp_netif_handlers: sta ip: 192.168.4.140, mask: 255.255.255.0, gw: 192.168.4.1
I (22694) [router_example, 332]: event_loop_cb, event: 65
I (22698) [router_example, 358]: Root obtains the IP address. It is posted by LwIP stack automatically
I (22708) [router_example, 169]: TCP client write task is running
I (22709) [router_example, 84]: TCP client read task is running
I (22721) [router_example, 44]: Create a tcp client, ip: 192.168.4.145, port: 2222
D (23069) [router_example, 246]: Node send, size: 72, data: {"src_addr": "24:0a:c4:fc:72:4c","data": "Hello TCP Server!","count": 1}
D (26072) [router_example, 246]: Node send, size: 72, data: {"src_addr": "24:0a:c4:fc:72:4c","data": "Hello TCP Server!","count": 2}
I (31529) [router_example, 279]: System information, channel: 9, layer: 1, self mac: 24:0a:c4:fc:72:4c, parent bssid: b0:08:75:34:e7:34, parent rssi: -62, node num: 1, free heap: 165384
I (41529) [router_example, 279]: System information, channel: 9, layer: 1, self mac: 24:0a:c4:fc:72:4c, parent bssid: b0:08:75:34:e7:34, parent rssi: -63, node num: 1, free heap: 165384
I (51529) [router_example, 279]: System information, channel: 9, layer: 1, self mac: 24:0a:c4:fc:72:4c, parent bssid: b0:08:75:34:e7:34, parent rssi: -63, node num: 1, free heap: 165384
I (61529) [router_example, 279]: System information, channel: 9, layer: 1, self mac: 24:0a:c4:fc:72:4c, parent bssid: b0:08:75:34:e7:34, parent rssi: -62, node num: 1, free heap: 165384
I (71529) [router_example, 279]: System information, channel: 9, layer: 1, self mac: 24:0a:c4:fc:72:4c, parent bssid: b0:08:75:34:e7:34, parent rssi: -62, node num: 1, free heap: 165384
I (81529) [router_example, 279]: System information, channel: 9, layer: 1, self mac: 24:0a:c4:fc:72:4c, parent bssid: b0:08:75:34:e7:34, parent rssi: -61, node num: 1, free heap: 165384
W (83494) [router_example, 59]: socket connect, ret: -1, ip: 192.168.4.145, port: 2222
I (83995) [router_example, 44]: Create a tcp client, ip: 192.168.4.145, port: 2222`