espressif/esp-mdf #72: mwifi_root_write return MDF_OK instead of error code

BartoszKubiak, 2019-07-25:
MDF to v1.0-beta1-2-gc1cc50f
If I use mwifi_root_write() with MAC that doesn't exist and MWIFI_COMMUNICATE_MULTICAST, it still return MDF_OK instead of ESP_ERR_MESH_NO_ROUTE_FOUND

source code:
```
	...
        mwifi_data.group = false;
        mwifi_data.communicate = MWIFI_COMMUNICATE_MULTICAST;
      	...	
        MDF_LOGI("root write: %.*s to " MACSTR, recv_length, data, MAC2STR(addr_list));
        ret = mwifi_root_write(addr_list, addrs_num, &mwifi_data, data, recv_length, true);
        if (ret != MDF_OK)
        {
            MDF_LOGE("<%s> mwifi_root_write", mdf_err_to_name(ret));
        } else
	{
            MDF_LOGI("<%s> mwifi_root_write", mdf_err_to_name(ret));
	}
```

result:
I (8060) [UART2MESH, 466]: root write: {"request":"introduce"} to 30:ae:a4:cc:4d:40
W (8061) [mwifi, 552]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Node failed to send packets, dest_addr: 30:ae:a4:cc:4d:40, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3ffba524, size: 23
W (8076) [mwifi, 681]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Root node failed to send packets, dest_mac: 30:ae:a4:cc:4d:40
I (8087) [UART2MESH, 473]: <MDF_OK> mwifi_root_write

reason:
mwifi_transmit_write() at the end returns always MDF_OK instead of ret, so mwifi_root_write() also returns MDF_OK

after fixing mwifi_transmit_write():
I (10735) [UART2MESH, 466]: root write: {"request":"introduce"} to 30:ae:a4:cc:4d:45
W (10737) [mwifi, 552]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Node failed to send packets, dest_addr: 30:ae:a4:cc:4d:45, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3ffba524, size: 23
W (10751) [mwifi, 681]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Root node failed to send packets, dest_mac: 30:ae:a4:cc:4d:45
W (10762) [mwifi, 1086]: Mwifi_transmit_write
E (10767) [UART2MESH, 470]: <ESP_ERR_MESH_NO_ROUTE_FOUND> mwifi_root_write

zhanzhaocheng, 2019-07-25:
Thank you very much for your feedback, we will fix it as soon as possible.

zhanzhaocheng, 2019-07-25:
I have fixed this problem please update esp-mdf verification