espressif/esp-mdf #73: error: 'MESH_EVENT_ROOT_GOT_IP' undeclared

microponts, 2019-07-25:
编译例子的时候报了 MESH_EVENT_ROOT_GOT_IP 未定义的错误

microponts, 2019-07-25:
继续往下编译还出现了找不到 miniz 的错误
 mconfig_chain.c:26:10: fatal error: miniz.h: No such file or directory

zhanzhaocheng, 2019-07-26:
1. 请提供一下 ESP-MDF 的版本, 开发环境, 完整的编译日志.以便于更快的定位和解决问题, [问题反馈](https://docs.espressif.com/projects/esp-mdf/zh_CN/latest/api-guides/general-notes.html#id10)
2. 你是否配置了 ESP-MDF 的路径, 请参考 [快速开始](https://github.com/espressif/esp-mdf/blob/master/README_cn.md#%E5%BF%AB%E9%80%9F%E5%BC%80%E5%A7%8Bmake)

david-glover, 2019-09-20:
MESH_EVENT_ROOT_GOT_IP and MESH_EVENT_ROOT_LOST_IP were removed in commit a272e7204d7004e0bb95e508496206652b0aa422 on April 16.  

You can checkout the release v1.0-beta1 in esp-mdf to avoid this issue.  Also, make sure you have the toolchain version 5.2.0 (see #81).
 
 https://github.com/espressif/esp-idf/commit/a272e7204d7004e0bb95e508496206652b0aa422#diff-5a20fead5c29f9549e4ef3d73ccaabfb

closed 2021-07-08 by microponts