espressif/esp-mdf #74: MDF bundled miniz causes crash with low level API tinfl_decompress

Dee-Fuse, 2019-08-01:
## Environment

- Development Kit: none
- Core (if using chip or module): ESP-WROOM32
- MDF version (``git rev-parse --short HEAD`` to get the commit id.):  867212cfbaa55a3da344b652e3508fa0a7ea4f71
- Development Env: PlatformIO
- Operating System: Windows
- Power Supply:    USB

## Problem Description

Using the MDF component "miniz" causes crash when calling `tinfl_decompress`
Using the `rom/miniz.h` instead works as expected. (Not linking the mdf miniz component)

I cant use the MDF without the miniz bundled in `components/third_party/miniz` because mwifi uses higher leven APIs than the rom version.

Am I missing something here?

### Expected Behavior

Both miniz components should behave (almost) the same with low level API calls.

### Actual Behavior

MDF miniz throws "StoreProhibited" exception whe trying to decompress with `tinfl_decompress`

### Steps to repropduce

pull test repository https://github.com/Dee-Fuse/testMiniz
Compile and upload
See ESP crash

### Code to reproduce this issue

See https://github.com/Dee-Fuse/testMiniz

To have a working code, switch comments on line 33 and 34 in `src/main.cpp` 

## Debug Logs

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5336
load:0x40078000,len:8816
ho 0 tail 12 room 4
load:0x40080400,len:6412
entry 0x40080748
I (30) boot: ESP-IDF 3.30202.190627 2nd stage bootloader
I (31) boot: compile time 12:06:09
I (31) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (36) boot: SPI Mode       : DIO
I (39) boot: SPI Flash Size : 4MB
I (43) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (58) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (64) boot:  2 factory          factory app      00 00 00010000 00100000
I (71) boot: End of partition table
I (74) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0707c ( 28796) map
I (92) esp_image: segment 1: paddr=0x000170a4 vaddr=0x3ffbdb60 size=0x021c0 (  8640) load
I (96) esp_image: segment 2: paddr=0x0001926c vaddr=0x40080000 size=0x00400 (  1024) load
I (98) esp_image: segment 3: paddr=0x00019674 vaddr=0x40080400 size=0x0699c ( 27036) load
I (116) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x15668 ( 87656) map
I (147) esp_image: segment 5: paddr=0x00035688 vaddr=0x40086d9c size=0x0491c ( 18716) load
I (161) boot: Loaded app from partition at offset 0x10000
I (162) boot: Disabling RNG early entropy source...
I (162) cpu_start: Pro cpu up.
I (163) cpu_start: Starting app cpu, entry point is 0x40081090
I (153) cpu_start: App cpu up.
I (171) heap_init: Initializing. RAM available for dynamic allocation:
I (177) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (182) heap_init: At 3FFC1118 len 0001EEE8 (123 KiB): DRAM
I (188) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (193) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (199) heap_init: At 4008B6B8 len 00014948 (82 KiB): IRAM
I (204) cpu_start: Pro cpu start user code
I (221) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (0) get_started: Decompressing! 54 bytes
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c433  PS      : 0x00060f30  A0      : 0x800d10ae  A1      : 0x3ffb9110
A2      : 0xa5a5a5a5  A3      : 0x00000000  A4      : 0x00000120  A5      : 0xa5a5a5a5
A6      : 0x3ffc1134  A7      : 0x00000000  A8      : 0x800e1012  A9      : 0x3ffb91d0  
A10     : 0x3ffc1134  A11     : 0x40085390  A12     : 0x00000001  A13     : 0x3ffb8ea0
A14     : 0x3ffb8ea0  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xa5a5a5a5  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x4000c433:0x3ffb9110 0x400d10ab:0x3ffb9120 0x400d0a9b:0x3ffb9230 0x400867e9:0x3ffbbd90

Rebooting...
```
Expected output:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5336
load:0x40078000,len:8816
ho 0 tail 12 room 4
load:0x40080400,len:6412
entry 0x40080748
I (31) boot: ESP-IDF 3.30202.190627 2nd stage bootloader
I (31) boot: compile time 12:26:48
I (31) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (36) boot: SPI Mode       : DIO
I (39) boot: SPI Flash Size : 4MB
I (42) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (58) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (64) boot:  2 factory          factory app      00 00 00010000 00100000
I (71) boot: End of partition table
I (74) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06d08 ( 27912) map
I (92) esp_image: segment 1: paddr=0x00016d30 vaddr=0x3ffbdb60 size=0x021c0 (  8640) load
I (95) esp_image: segment 2: paddr=0x00018ef8 vaddr=0x40080000 size=0x00400 (  1024) load
I (98) esp_image: segment 3: paddr=0x00019300 vaddr=0x40080400 size=0x06d10 ( 27920) load
I (117) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x14538 ( 83256) map
I (146) esp_image: segment 5: paddr=0x00034558 vaddr=0x40087110 size=0x045a8 ( 17832) load
I (160) boot: Loaded app from partition at offset 0x10000
I (160) boot: Disabling RNG early entropy source...
I (160) cpu_start: Pro cpu up.
I (161) cpu_start: Starting app cpu, entry point is 0x40081090
I (0) cpu_start: App cpu up.
I (169) heap_init: Initializing. RAM available for dynamic allocation:
I (175) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (181) heap_init: At 3FFC1118 len 0001EEE8 (123 KiB): DRAM
I (186) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (191) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (197) heap_init: At 4008B6B8 len 00014948 (82 KiB): IRAM
I (202) cpu_start: Pro cpu start user code
I (219) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (0) get_started: Decompressing! 54 bytes
I (0) get_started: 0x3ffc1134   48 65 6c 6c 6f 20 57 6f  72 6c 64 21 0d 0a 48 65  |Hello World!..He|
I (10) get_started: 0x3ffc1144   6c 6c 6f 20 57 6f 72 6c  64 21 0d 0a 48 65 6c 6c  |llo World!..Hell|
I (10) get_started: 0x3ffc1154   6f 20 57 6f 72 6c 64 21  0d 0a 48 65 6c 6c 6f 20  |o World!..Hello |
I (20) get_started: 0x3ffc1164   57 6f 72 6c 64 21                                 |World!|
Decompressed 54 / 54 bytes

```

## Other items if possible
Build log (crashing)
```
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30202.190627 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...    
Dependency Graph
|-- <miniz> 1.0
```
[firmware_crash.zip](https://github.com/espressif/esp-mdf/files/3456218/firmware_crash.zip)

Build log (working)
```
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-espidf 3.30202.190627 (3.2.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
No dependencies
```
[firmware_working.zip](https://github.com/espressif/esp-mdf/files/3456224/firmware_working.zip)

EspHuifeng, 2019-08-02:
The problem has been identified and we are dealing with it.

EspHuifeng, 2019-08-13:
We have solved this problem. You can try is again. Before you do it, you could read readme.md in miniz directory, because I put the information about how we modified into it.

Dee-Fuse, 2019-08-19:
After some testing I wasn't able to get the new version of miniz working correctly.
I have to use a wrapping buffer to be able to decompress multi megabyte datastreams on the fly (chunked).
Even without setting the `CONFIG_MINIZ_MINIMIZE_STACK_CONSUME` flag I got an error -1 after decompressing the first chunk of data.
I've updated my repository to reflect my codeabse with chunked / wrapping buffers and the current patch you've pushed.

With the two defines at the top of `main.cpp` you can switch my implementation with the one you contributed (with changes to be able to handle wrapping buffers). And also which miniz header should be used.

diplfranzhoepfinger, 2025-03-14:
https://github.com/lbernstone/miniz-esp32

tonyshuang, 2025-03-14:
您好，信件已收到。

diplfranzhoepfinger, 2025-03-14:
https://github.com/joltwallet/esp_full_miniz

diplfranzhoepfinger, 2025-03-14:
@DeeFuse this is offline:  https://github.com/Dee-Fuse/testMiniz