espressif/esp-mdf #75: Wifi Mesh automatically generate a root node without router

lixiantai, 2019-08-03:
I've tried the Mwifi demo code to build a mesh network without router, it works. 

Can I build a mesh network without configure whether the root node or the non-root node? If there is no root node in the network, one of the nodes can become the root node. I read the guide [Root Node Failure](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/mesh.html#mesh-managing-a-network), but it need the router.

zhanzhaocheng, 2019-08-05:
In the no-router scheme, because there is no router as a reference, the device cannot select which device should be the root node, so you need to specify the root node.
If you don't want to specify the node type in your code, you need to have a trigger that becomes the root node: such as via buttons, Bluetooth, etc.

lixiantai, 2019-08-05:
Thanks for reply, I understood what you said. I want to build such mesh network: all nodes' mesh id and channel are configured via Bluetooth, and power on some of these will also build a mesh network.
I would like to know if it is feasible, If a node cannot find the root node, it can become the root node with random delay (that may avoid generating two or more root node when power on at the same time).
Thanks.

githubforsank, 2020-01-07:
Each node can list --> the RSSI of the child nodes .. thereby keeping a record of the number and RSSI of its nearby nodes.  This can be converted into a weight number and broadcasted as the voting happens for root. Thus one of them can become root. Re-root selection is also possible. Is there any expected work ..? Really need this feature.

Thank you,
Sasank Panda

syed-haider5, 2020-05-15:
@githubforsank  
How can I forcefully make a node ROOT NODE, because in my case couple of nodes are in same RSSI threshold of WiFi router, so it keeps shifting the root node. I need to make 1st node ROOT, all other will be the child. (5 Nodes are in a straight line, in same room 2m distance between each node, and wifi is about 5m away from first node).
My expectation is that the first node should be the root always, but in my case RSSI of wifi will be nearly equal to each node so they get playing with it.