espressif/esp-mdf #77: ESP-MDF (MWIFI) Packet Fragmentation not working

corlovsky, 2019-08-14:
## Environment
MinGW32 on Windows 10

- Core (if using chip or module): ESP-EYE

- MDF version = aa79a9d
- Development Env:		Notepad++ with MINGW32
- Operating System:		Windows 100
- Power Supply:         USB

## Problem Description

I am trying to use the Router MWIFI example. Everything works as expected (mesh network dynamically adds nodes and sends string to TCP server) until I try and send a string bigger than 1455 bytes. I don't think the packet fragmentation is working properly. Or maybe I didn't enable it correctly. I tried to add it to the menuconfig file and have it enabled. The output messages are below:

```
E (11390) [mwifi, 685]: <MDF_ERR_INVALID_ARG> !(size > 0 && size < 8096)
I (11795) [router_example, 392]: System information, channel: 4, layer: 1, self mac:
 c4:4f:33:18:af:bd, parent bssid: cc:40:d0:0b:16:e9, parent rssi: -31, node num: 1,
free heap: 112104
I (12117) event: sta ip: 192.168.1.29, mask: 255.255.255.0, gw: 192.168.1.1
I (12118) [router_example, 443]: event_loop_cb, event: 17
I (12120) [router_example, 464]: Root obtains the IP address. It is posted by LwIP s
tack automatically
I (12130) [router_example, 97]: TCP client read task is running
I (12137) [router_example, 47]: Create a tcp client, ip: 192.168.1.15, port: 6969
I (12131) [router_example, 182]: TCP client write task is running
**W (12661) [mwifi, 1203]: Buffer is too small, size: 1455, the expected size is: 5715**
```

### Expected Behavior

The main thing I am focused on is the message that says "Buffer is too small, size: 1455, the expected size is 5715" I know my file is bigger than the max packet size, but I expect it to be able to automatically split up the file into smaller packets and transmit them automatically.

### Actual Behavior

The mesh connects to the network and the TCP server, and attempts to send a packet, but fails because the file is bigger than the max packet size. The TCP server receives correctly when using the stock example code, but changing it to a bigger string makes it fail.

### Steps to reproduce

1. Use make menuconfig to set the IP address, port, router SSID and password, and COM port if Windows, USBtty0 if linux.
2. Use "make -j5 flash monitor" to flash code to ESP32 device (doesn't have to be ESP-EYE, it just wont get a new image.
3. Set up TCP server on receiving end... Python Code is below: 

<script src="https://gist.github.com/corlovsky/d564ffc5616fee0f331fc3361ccd3417.js"></script>

### Code to reproduce this issue

<script src="https://gist.github.com/corlovsky/20df6a5450b22615aa65175c75bc338a.js"></script>

## Debug Logs

```
E (11390) [mwifi, 685]: <MDF_ERR_INVALID_ARG> !(size > 0 && size < 8096)
I (11795) [router_example, 392]: System information, channel: 4, layer: 1, self mac:
 c4:4f:33:18:af:bd, parent bssid: cc:40:d0:0b:16:e9, parent rssi: -31, node num: 1,
free heap: 112104
I (12117) event: sta ip: 192.168.1.29, mask: 255.255.255.0, gw: 192.168.1.1
I (12118) [router_example, 443]: event_loop_cb, event: 17
I (12120) [router_example, 464]: Root obtains the IP address. It is posted by LwIP s
tack automatically
I (12130) [router_example, 97]: TCP client read task is running
I (12137) [router_example, 47]: Create a tcp client, ip: 192.168.1.15, port: 6969
I (12131) [router_example, 182]: TCP client write task is running
**W (12661) [mwifi, 1203]: Buffer is too small, size: 1455, the expected size is: 5715**
```

## Other items if possible

Below is my menuconfig (Kconfig.projbuild) also:

<script src="https://gist.github.com/corlovsky/72bdfaa7a5b6faafbc2c3025224ca274.js"></script>

zhanzhaocheng, 2019-08-21:
1. mwifi supports a maximum size of 8kB, you can use [mwifi_test](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/console_test) to test him.
2. As your log says, the buffer you gave is small, `Buffer is too small, size: 1455, the expected size is: 5715**`, you can fix it in two ways:
- Modify the size of the buffer
```c
       char *data    = MDF_MALLOC(MWIFI_PAYLOAD_LEN * 8);
       int  size = MWIFI_PAYLOAD_LEN * 8;
        mwifi_read(src_addr, &data_type, data, &size, portMAX_DELAY);
        MDF_FRRE(data);
```

- Incoming secondary pointers by mwifi internal application space
```c
       char *data  = NULL;
       int  size       = 0;
       mwifi_read(src_addr, &mwifi_type, &data, &size, portMAX_DELAY);
       MDF_FRRE(data);
```

corlovsky, 2019-08-22:
Thanks! It turns out I only changed it in one place and needed to change it in multiple places.

closed 2019-08-22 by corlovsky

moefear85, 2022-02-04:
where exactly are these "incoming secondary pointers by mwifi internal application space"? I'm getting the same error. Simply modifying the size of the buffer as suggested in the first way, isn't enough in my case.