espressif/esp-mdf #81: ESP-idf examples gets compiled but not esp-mdf examples

Piruthivi, 2019-08-31:
Am using ubuntu for my esp32 project. Working with both esp-idf and mdf programs. The thing is i could  compile esp-idf programs at ease with both make and idf.py commands, but esp-mdf programs(for example get-started) not getting compiled through make and idf.py both. The following is make command response in command prompt. I have tried updating both esp-idf and mdf git files but still issue continues.

_$make
Toolchain path: /home/praveen/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.6
Python requirements from /home/praveen/esp/esp-mdf/esp-idf/requirements.txt are satisfied.
Building partitions from /home/praveen/work/get-started/partitions.csv...
CC build/app_trace/app_trace_util.o
CC build/app_trace/host_file_io.o
CC build/app_trace/app_trace.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CXX build/asio/asio/asio/src/asio.o
In file included from /home/praveen/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /home/praveen/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,
                 from /home/praveen/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,
                 from /home/praveen/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,
                 from /home/praveen/esp/esp-mdf/esp-idf/components/asio/asio/asio/include/asio/impl/error.ipp:19,
                 from /home/praveen/esp/esp-mdf/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:22,
                 from /home/praveen/esp/esp-mdf/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/home/praveen/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/praveen/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
/home/praveen/esp/esp-mdf/esp-idf/make/component_wrapper.mk:285: recipe for target 'asio/asio/src/asio.o' failed
make[1]: *** [asio/asio/src/asio.o] Error 1
/home/praveen/esp/esp-mdf/esp-idf/make/project.mk:530: recipe for target 'component-asio-build' failed
make: *** [component-asio-build] Error 2_

will-emmerson, 2019-08-31:
I had a similar problem, and just ended up removing IDF and $IDF_PATH. But I expect the problem is that you are using IDF v4 which uses version 8.2.0 of the toolchain, whereas MDF uses it's own version of IDF (`/home/praveen/esp/esp-mdf/esp-idf`), based on v3, which is using version 5.2.0 of the toolchain. Hence the `WARNING: Toolchain version is not supported: esp32-2019r1` message above.
You can't upgrade the version of IDF in MDF to v4 (yet), so to use both together you'll have to downgrade the standalone IDF to the same version as in MDF and downgrade the toolchain to 5.2.0.
But I could be wrong about all of that.

Piruthivi, 2019-09-01:
So if i remove IDF_PATH which i assigned before and adding this esp-idf within mdf will be working right?

david-glover, 2019-09-20:
I agree with will-emmerson.  I also had the same issue when compiling with the version 8.2.0 of the toolchain.  As I still had the version 5.2.0, I changed my path to point to that one and there were no issues with compilation.
I had already checked out the version 1.0 MDF release due to another compilation error.  (as in #73)

I did not change my IDF_PATH environment variable, and it is still pointing to the IDF v4 outside of MDF.

Suraj-singh999898, 2020-03-11:
I have a similar issue, when or how can it be solved?

version
> esp idf v4.0
> esp mdf latest

idf installation process same shown [https://docs.espressif.com/projects/esp-idf/en/v4.0/get-started/index.html]
mdf installation > https://docs.espressif.com/projects/esp-idf/en/v4.0/get-started/index.html

and compiler error for mdf "getting started" project
 error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)

if I downgrade my xtensa toolchain version to 5.2.0 then it works but I don't wont to downgrade idf version.

Thanks for your effort.

devrim-oguz, 2020-05-26:
Any updates on this?