espressif/esp-mdf #84: esp-mdf example get_started not compiling

dhaval2017, 2019-09-17:
esp-mdf get_started example giving error like this

D:/Espressif/esp-mdf/components/mwifi/mwifi.c:62:31: error: unknown type name 'mesh_event_t'; did you mean 'mesh_event_id_t'?
 static void esp_mesh_event_cb(mesh_event_t event)
                               ^~~~~~~~~~~~
                               mesh_event_id_t
D:/Espressif/esp-mdf/components/mwifi/mwifi.c: In function 'mwifi_start':
D:/Espressif/esp-mdf/components/mwifi/mwifi.c:301:16: error: 'mesh_cfg_t' {aka 'struct <anonymous>'} has no member named 'event_cb'
     mesh_config.event_cb = esp_mesh_event_cb;
                ^
D:/Espressif/esp-mdf/components/mwifi/mwifi.c:301:28: error: 'esp_mesh_event_cb' undeclared (first use in this function); did you mean 'esp_mesh_get_id'?
     mesh_config.event_cb = esp_mesh_event_cb;
                            ^~~~~~~~~~~~~~~~~
                            esp_mesh_get_id

EspHuifeng, 2019-09-19:
It shouldn't occur if you clone the esp-mdf repository and set the MDF_PATH variable correctly , Please print the MDF_PATH variable and check it before you build the project again

closed 2019-09-28 by dhaval2017

ub3132003, 2020-11-18:
so as I,and I'd check the MDF_PATH 
`In file included from /mnt/d/sj/code/esp-mdf/components/mcommon/include/mdf_common.h:23,
                 from /mnt/d/sj/code/esp-mdf/components/mwifi/include/mwifi.h:18,
                 from /mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c:15:
/mnt/d/sj/code/esp-mdf/esp-idf/components/esp32/include/rom/rtc.h:1:2: warning: #warning rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead [-Wcpp]
 #warning rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead
  ^~~~~~~
In file included from /mnt/d/sj/code/esp-mdf/components/mcommon/include/mdf_common.h:24,
                 from /mnt/d/sj/code/esp-mdf/components/mwifi/include/mwifi.h:18,
                 from /mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c:15:
/mnt/d/sj/code/esp-mdf/esp-idf/components/esp32/include/rom/crc.h:1:2: warning: #warning rom/crc.h is deprecated, please use esp32/rom/crc.h instead [-Wcpp]
 #warning rom/crc.h is deprecated, please use esp32/rom/crc.h instead
  ^~~~~~~
In file included from /mnt/d/sj/code/esp-mdf/components/mcommon/include/mdf_common.h:34,
                 from /mnt/d/sj/code/esp-mdf/components/mwifi/include/mwifi.h:18,
                 from /mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c:15:
/mnt/d/sj/code/esp-mdf/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
/mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c:133:31: error: unknown type name 'mesh_event_t'; did you mean 'eth_event_t'?
 static void esp_mesh_event_cb(mesh_event_t event)
                               ^~~~~~~~~~~~
                               eth_event_t
/mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c: In function 'mwifi_start':
/mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c:402:16: error: 'mesh_cfg_t' {aka 'struct <anonymous>'} has no member named 'event_cb'
     mesh_config.event_cb = esp_mesh_event_cb;
                ^
/mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c:402:28: error: 'esp_mesh_event_cb' undeclared (first use in this function); did you mean 'esp_mesh_get_id'?
     mesh_config.event_cb = esp_mesh_event_cb;
                            ^~~~~~~~~~~~~~~~~
                            esp_mesh_get_id
/mnt/d/sj/code/esp-mdf/components/mwifi/mwifi.c:402:28: note: each undeclared identifier is reported only once for each function it appears in`