espressif/esp-mdf #87: wrong data receiving on characteristics

dhaval2017, 2019-09-28:
I am using mconfig example to make my android app for configuration

I have enable notify on read but I am getting wrong public key as sent by esp-mdf

I am receiving

-----BEGIN PUBLIC KEY-----\nMIGfMA0GCSqGSIb3DQEBAQUAA4GNADCBiQKBgQDTpPeoX5k0ivqYOJkY+Z/RNIEw\n7fQidMrBl6zR46bxwOWmmWBNVMibNYVB6Ty7R2wxnqItMNfYrRUSuwggELToLrBs\nfll10ecYfM+oGKptz0qvfsA8zpIlD6CAs004tHv4tRkYuIWZT9eCshXSUkWTqvDp\nPsX+af9u+Wp9DeeFowIDAQAB\n-----END PUBLIC K-----\n

on espmdf I have printed its 
-----BEGIN PUBLIC KEY-----
MIGfMA0GCSqGSIb3DQEBAQUAA4GNADCBiQKBgQCcAV7JoSWxC5rSCil1WHMVJCya
f+n/eGZOtEwPRuaRMQCRPlxCdEx3bO6SFsssU5lQJPMILSrWioXz4i4AMzGsJIOx
4W1/KTy3dPCAlLs90wRd2l/HDAheSUBksNrVZB6g0lYaNG3+BvT2R/Ryc2sS5Qf2
HXinIniB1f1om6HJ6QIDAQAB
-----END PUBLIC KEY-----

EspHuifeng, 2019-09-29:
It is hard for me to find out why it is deferent between receiving public key and printed public key according to the message that you give. Please let me know how to get error results.

dhaval2017, 2019-09-29:
I feel like its problem of xamarin development 
As I checked in nrf connect I am receiving correct public key

closed 2019-09-29 by dhaval2017